tzapu/WiFiManager #1452: No way to tell if Exit has been pressed when using (non-)blocking startWebPortal()

nullstalgia, 2022-07-13:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP32

Hardware:			ESP32-WROOM

Core Version:      	ESP32 PlatformIO Core V5.0.0

### Description

- This is also an issue stating that if you use startWebPortal() with blocking enabled, the sketch continues at that point and the web server is unreachable. So I turned blocking off and use process() in the loop()

After the user clicks Exit on the UI, `wifiManager.getWebPortalActive()` stays true, and `wifiManager.getConfigPortalActive()` stays false. 

`abort` seems to be the only variable that does change in line, but it is private. I threw in a getAbort() so I could get my project working the way I intend, but hopefully these issues can be addressed in a smarter way. :)

tablatronix, 2022-07-14:
Yeah makes sense, exit for webportal is undefined

#1353

tablatronix, 2022-07-14:
I added a WebPortalActive=false fix, as it seemed to be missing, and should get stopped

Another workaround is to override the `server->on(WM_G(R_exit),` callback.

tablatronix, 2022-07-14:
Another solution would be to add enum returns to process() instead of a bool.. for more useful states

nullstalgia, 2022-07-14:
Great, this seems to work! I think once I had it stuck on 0 across reboots before this patch, but a quick test did not reproduce that issue... Could be user error there.

I'll let you close the issue if you wish to use it as a marker for the process() enums. 

But I had a quick question: Is there a reason the parameters can not be set in the web portal? It does not affect my use case, so I don't mind, but I was just wondering.

tablatronix, 2022-07-14:
Not sure I understand, can you elaborate what is not working with parameters?

nullstalgia, 2022-07-15:
When connecting to the AP from autoConnect(), the parameters appear in the Setup WiFi page.

When connecting via the local IP when using startWebPortal(), the parameters do not appear in the Setup WiFi page. I assumed this was intentional.

tablatronix, 2022-07-15:
Are you sure its not just because your params are out of scope?

nullstalgia, 2022-07-15:
You are 100% right. That totally went over my head.

Thanks again for the quick help and great library!