tzapu/WiFiManager #1453: Build error for ESP32 in VSCode/PIO

adbrimhall, 2022-07-15:
I get the following build error in VSCode using PlatformIO.

**_In file included from .pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:99:14: fatal error: Update.h: No such file or directory_**

my platformio.ini is using the git library:
lib_deps =
	https://github.com/tzapu/WiFiManager.git

If I change my platformio.ini to this instead:
lib_deps =
	WiFiManager

I get a different build error:

_**In file included from .pio/libdeps/esp32dev/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/esp32dev/WiFiManager/WiFiManager.h:16:10: fatal error: ESP8266WiFi.h: No such file or directory**_

I am new to VSCode and PlatformIO.  Not sure how to proceed.  Any ideas?

ahmedtariqh, 2022-07-16:
i had the same issue , tried with arduino ide and got it built!
i think it might be a platformio issue
looking forward to get it solved

veitk, 2022-07-21:
ouch, i'm not alone. is downgrading platformio a solution for this?

tablatronix, 2022-07-21:
Not sure, will it build for esp8266?

veitk, 2022-07-21:
sorry in my case it was a different library which caused unexpected trouble.

Kuuchuu, 2022-07-31:
I'm also having this issue, I haven't been able to work out any solutions yet. I'm unable to use the arduino ide for this project

stsvilik, 2022-08-06:
Same issue. Looks like including just `WiFiManager` in lib_deps downloads outdated version of `WiFiManager.h`. If i specify full git repo path `https://github.com/tzapu/WiFiManager.git`, it loads the lib fine without editor error, but fails horribly during the build with bunch of errors in PIO toolchain shared files.

CharlesPavlov, 2022-08-08:
It would seem I'm having the same issue:

`In file included from .pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.cpp:13:
.pio/libdeps/esp32doit-devkit-v1/WiFiManager/WiFiManager.h:99:14: fatal error: Update.h: No such file or directory`

sauvant, 2022-08-22:
Same here. Any advice?

sauvant, 2022-08-22:
I adviced myself ;-) 

lib_ldf_mode = deep+

in platformio.ini solved the issue for me.

tablatronix, 2022-08-22:
Yeah i have no idea why it does that, it seems like a bug in pio or with certain branches

kinsi55, 2022-08-25:
I guess people (Like me) install it from the PlatformIo UI, and that would explain why its not working

[image]

tablatronix, 2022-08-25:
Hmm maybe pio manager doesn't support dev versions

ahmedreghini, 2022-11-17:
I solve it by adding this in the main.cpp : 
#include <Update.h>
#include <WebServer.h>
#include <DNSServer.h>

rossog93, 2022-12-29:
I had in `platformio.ini ` this: `lib_ldf_mode = deep` causing errors. I added `lib_ldf_mode = deep**+**' and worked fine.

rmhomecouk, 2023-02-02:
> I guess people (Like me) install it from the PlatformIo UI, and that would explain why its not working
> 
> [image]

Thank you for the hint, I reference the github link instead and it worked straight away.

tablatronix, 2023-02-03:
Yeah you have to specify in ini file, platformio lib does not support non release versions or at least I never figured out how to do it. I am going to make this release soon and stop using prereleases

closed 2023-02-03 by tablatronix

mahboud, 2023-04-03:
I see this issue was closed, but I just grabbed 2.0.15-rc.1 using PlatformIO and still get a 

`.pio/libdeps/esp32-c3-devkitm-1/WiFiManager/WiFiManager.h:87:14: fatal error: Update.h: No such file or directory`

jfujitani, 2023-07-08:
> I solve it by adding this in the main.cpp : #include <Update.h> #include <WebServer.h> #include <DNSServer.h>

I know this is an old comment on a closed issue but, I found that I also got those kinds of errors ( related to `update.h` etc). The errors went away, without having to directly include the headers mentioned in the error message, when I added `#include <WiFiManager.h> to any of my source files.

fg1998, 2023-10-21:
> `https://github.com/tzapu/WiFiManager.git`

Using complete URL instead of 'taps/WiFiManager'in platformio.ini works like a charm.
Thanks

PhilippeLRQ, 2023-11-08:
It is the same issue here. 
It seems that VS code is looking to esp8266wifi.h ????

In file included from src\main.cpp:2:0:
.pio\libdeps\esp32dev\WiFiManager/WiFimanager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory

Help needed please.