tzapu/WiFiManager #1454: Connect to known SSID after a power failure? dont start a new access point.

jackkitley, 2022-07-16:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			D1 mini

### Description

Im trying to reconnect to my known SSID that was successful but i cannot reconnect to it after a power outage. All the device does is boot up another AP which i dont want. I have tried multiple ways to keep restarting the device, wifi begin etc etc but i always starts a new AP on SSID not found boot up.

What i want to achieve:
After a power outage for the ESP device to keep retrying the known SSID from the config file. 
I will handle a reset for AP manually if i wish to but all i want is on boot for the device to keep trying to connect to the known SSID as the routers take a while to boot.

### Settings in IDE

Module:  		Wemos D1

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {
  Serial.begin(9600);

  bool spiffsSetup = loadConfigFile();
  if (!spiffsSetup)
  {
    Serial.println(F("Forcing config mode as there is no saved config"));
    forceConfig = true;
  }

  setupWifiManager();
}

void loop() {
   //Continue to connect to MQTT and reconnect if dropped. Connect to known SSID on boot and dont start a new AP.
  //Listen for button press to start AP if i wish to.
}

void setupWifiManager()
{
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 20);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 10);
  WiFiManagerParameter custom_mqtt_user("user", "mqtt user", mqtt_user, 50);
  WiFiManagerParameter custom_mqtt_pass("pass", "mqtt pass", mqtt_pass, 50);

  // WiFiManager
  // Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_mqtt_user);
  wifiManager.addParameter(&custom_mqtt_pass);

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  wifiManager.setConfigPortalTimeout(180);
  if(forceConfig) {
    wifiManager.autoConnect("AP");  
  }
}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2022-07-16:
You will have to wait longer to reconnect since your ap is not up yet.. or add your own counter to retry.

wm does have a connectiontimeout and a retry count setting you can try increasing those.

jackkitley, 2022-07-16:
@tablatronix Thank you. Do you possibly have an example? 

If i pull the power from wifi i can see serial trying to connect but if i pull power from device and then power up again it will go straight to AP and not try reconnect which is annoying. I want to avoid the AP at all costs.

tablatronix, 2022-07-16:
```
    //sets timeout for which to attempt connecting, useful if you get a lot of failed connects
    void          setConnectTimeout(unsigned long seconds);

    // sets number of retries for autoconnect, force retry after wait failure exit
    void          setConnectRetries(uint8_t numRetries); // default 1
````

jackkitley, 2022-07-16:
The wifi could take in excess of 5min to come up again.

jackkitley, 2022-07-16:
@tablatronix ah, i have not seen setConnectRetries befone. I will make that max out haha

tablatronix, 2022-07-16:
depending on if you want your sketch to continue without wifi, you might want to goto sleep or add your own timer in loop to retry every minute or so if not connected

jackkitley, 2022-07-16:
@tablatronix ive tried my own loop but doesnt seem to work. In South africa we have power out times. The whole house will be without electricity and when it comes back i want the ESP to just connect back again so i can get data into my home automation. 

Its annoying if it sets up an AP.

tablatronix, 2022-07-16:
and to use autoconnect but not start cp ever ( ony on demand etc ) use `wm.setEnableConfigPortal(false)`

jackkitley, 2022-07-16:
@tablatronix i need WIFI for MQTT to publish my hot water times.

jackkitley, 2022-07-16:
> depending on if you want your sketch to continue without wifi, you might want to goto sleep or add your own timer in loop to retry every minute or so if not connected.
> 
> I have to test out why the esp automatic reconnect has not been working, I have noticed this as well for the last several versions, i have yet to test it or see if wm is breaking it

Yes, its not great that it doesnt try last connected. It just boots into creating AP.

jackkitley, 2022-07-16:
@tablatronix 

Look ok? 

`  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 20);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 10);
  WiFiManagerParameter custom_mqtt_user("user", "mqtt user", mqtt_user, 50);
  WiFiManagerParameter custom_mqtt_pass("pass", "mqtt pass", mqtt_pass, 50);

  // WiFiManager
  // Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_mqtt_user);
  wifiManager.addParameter(&custom_mqtt_pass);

  wifiManager.setConnectTimeout(180);
  wifiManager.setConnectRetries(50);
  wifiManager.setEnableConfigPortal(false);
  wifiManager.autoConnect("AP");
  //erases the wifi settings from chip. Going to add this to mqtt command and button press.
  //    wifiManager.resetSettings();
`

tablatronix, 2022-07-16:
If you have alot of power failures I would suggest you do not use configportal at all, and provide a manual reset or ondemand cp start.. Or add a counter like tasmota does, after x restarts start ap etc.

jackkitley, 2022-07-16:
@tablatronix 

Yes, this is what i disabled for tasmota and what i wish to do for WM.

jackkitley, 2022-07-16:
i disabled for x retries, start AP because of power failures so it just connects to SSID when its back up

jackkitley, 2022-07-16:
@tablatronix How would you recommend i tackle this? im very new to this space of manual coding. i use Tasmota but develiped my own C++ for a custom device.

jackkitley, 2022-07-16:
@tablatronix 

>   wifiManager.setConnectTimeout(180);
  wifiManager.setConnectRetries(50);
  if(forceConfig) {
    wifiManager.autoConnect("AP");
  }else {
    wifiManager.setEnableConfigPortal(false);
  }

jackkitley, 2022-07-16:
@tablatronix i flashed device. First start up it spun up an AP.

> ip:192.168.88.21,mask:255.255.255.0,gw:192.168.88.1
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.88.21 
*wm:[2] [CB] _savewificallback calling 
Should save config
*wm:[2] shutdownConfigPortal 
station: 76:da:ca:ea:8b:12 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*wm:[2] restoring usermode STA
bcn 0
del if1
pm open,type:2 0
mode : sta(38:2b:78:04:51:5b)
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting

jackkitley, 2022-07-16:
going to disconnect WIFI now.

jackkitley, 2022-07-16:
> @tablatronix
> 
> > wifiManager.setConnectTimeout(180);
> > wifiManager.setConnectRetries(50);
> > if(forceConfig) {
> > wifiManager.autoConnect("AP");
> > }else {
> > wifiManager.setEnableConfigPortal(false);
> > }

I powered up device but doesnt even try to connect to last SSID.

jackkitley, 2022-07-16:
> setEnableConfigPortal
esp is supposed to auto-reconnect in background, but it seems to be not working lately.

Yes. Its not working at all on power failure.

tablatronix, 2022-07-16:
you still have to call autoconnect

```
if(!forceConfig) wifiManager.setEnableConfigPortal(false);
wifiManager.autoConnect("AP");

```

jackkitley, 2022-07-16:
hmm

I see getWiFiIsSaved in docs. Doesnt explain much. Ok, ill check on Wifi saved then and not force config. forceConfig is from my saved creds;

> void setup() {
  Serial.begin(9600);

  bool spiffsSetup = loadConfigFile();
  if (!spiffsSetup)
  {
    Serial.println(F("Forcing config mode as there is no saved config"));
    forceConfig = true;
  }

  setupWifiManager();
}

jackkitley, 2022-07-16:
if (!wifiManager.getWiFiIsSaved()) wifiManager.setEnableConfigPortal(false);
  wifiManager.autoConnect("AP");

jackkitley, 2022-07-16:
[image]
@tablatronix on the right track! thanks bud

SboleWork, 2022-08-24:
@jackkitley were you able to get this sorted for your application?

My device seems to deal with loadshedding without problems.

When the power comes back on it tries to connect to the known network.
If it fails to connect to the known network -  i.e. the router isn't up yet, it reverts to AP mode and allows me to connect to it and search for networks.
If I don't connect to it, it times out after 30 seconds, restarts the ESP32 and tries to connect to the known network.
This loops until either the router comes up and it connects or I connect to a new network.

`void setupNetwork() {
  //Connect to WiFi
  WiFi.mode(WIFI_STA);
  // wifiManager.resetSettings();
  wifiManager.setConfigPortalTimeout(30);
  if (!wifiManager.autoConnect(DEVICE_SSID, DEVICE_PASSWORD)){
    ESP.restart();
    delay(1000);
  };
}`

jackkitley, 2022-08-24:
@SboleWork Hi. Got it sorted.

`If it fails to connect to the known network - i.e. the router isn't up yet, it reverts to AP mode and allows me to connect to it and search for networks.` - this was my problem. the router didnt boot fast enough and the tries were only a handful before it reverted to AP mode which i didnt want.

Now it tries for a long time and eventually connects. I have a button which i use to revert back to AP mode.

I used the manager in my geyserwise project.

https://github.com/jackkitley/geyserwise_max_iot

tablatronix, 2022-08-24:
you just change the settings or add a wait loop in your code, the option is setconnecttimeout and numretries