tzapu/WiFiManager #1458: Not Connecting to internet (This problem have with only routers) (for mobile hotspots working code 100%)

sanjulanipun, 2022-07-21:
Esp8266

Hardware:			ESP-12f

Core Version:      	2.6.0

### Description

### i make code with esp8266 firebase library with code. after config router wifi 1st time working 100%. after 2nd-time auto connecting wifi connecting properly. but no internet acess. router has internet I check it using a mobile phone and another nodemcu it working there have internet. maybe some bug have with wifimanger.

if i connect this for mobile hospot it working properly no any error. 

anyone got this problem ? 
 
### 

Additional libraries :  #include "FirebaseESP8266.h" #include <ESP8266WiFi.h> 

### Debug Messages

```
*WM:[2] Connecting as wifi client... 
*WM:[2] setSTAConfig static ip not set, skipping 
*WM:[1] Connecting to SAVED AP: Sanju 4G
*WM:[1] connectTimeout not set, ESP waitForConnectResult... 
*WM:[2] Connection result: WL_CONNECTED
*WM:[1] AutoConnect: SUCCESS 
*WM:[2] Connected in 565 ms
*WM:[1] STA IP Address: 192.168.1.4
connected! Device Starting snyc...
Error in set, connection lost
Error in set, connection lost

![image](https://user-images.githubusercontent.com/32708235/180149876-e95a35b6-856a-48b0-b3f6-1e4251e23535.png)

this is when i program esp12f  erase flash with all contents, after config wifi, 1st time it working. after power off and connet it only connet to wifi but no internet connection. same problem.
![image](https://user-images.githubusercontent.com/32708235/180151645-4f415dcc-b685-4a3a-bf4b-b4b4e0d74c57.png)

after restart 
![image](https://user-images.githubusercontent.com/32708235/180151786-e2f0eca7-9890-4bd4-b6ef-8300be826905.png)

```

sanjulanipun, 2022-07-21:
i just try this but no good results i get. #871  i have no idea what is that error.

tablatronix, 2022-07-21:
Sounds like AP dhcp is not working

closed 2022-07-21 by tablatronix