tzapu/WiFiManager #1459: WiFiManager::waitForConnectResult > while(millis() < timeoutmillis)

Ramim256, 2022-07-24:
I want to execute some line of code from my sketch when it's trying to connect, I mean when it is in the while loop printing the  ".", at "WiFiManager.cpp > WiFiManager::waitForConnectResult > while(millis() < timeoutmillis)";
is there any possible way to do this ??

tablatronix, 2022-07-24:
I can add a callback here and see if it helps, you have to be careful though as wifi is very busy during this time and you can cause crashes, certainly no interrupts for eg.

another option I am considering 
#1460

Ramim256, 2022-07-28:
i want to access "ssid" variable from the main sketch, and  execute some line of code from my sketch when it's trying to connect,
my code for 0.91 inch OLED display, 

demo code:
void connectWiFi(void)
{
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  display.clearDisplay();
  WiFi.begin(ssid, password);
  int i = 10;
  int j = 0;
  if (WiFi.status() != WL_CONNECTED) {
    while (currentMillis - previousMillis > reconnectInterval * 1000)
    {
      delay(500);
      Serial.print(currentMillis - previousMillis);
      Serial.print("\n");
      if (i < 120) {
        robojaxText("Connecting to ", 28, 0, 1, false);
        robojaxText(ssid, 13, 14, 1, false);
        robojaxText(".\n", i, 22, 1, false);
        i = i + 5;
        display.display();
        if (WiFi.status() == WL_CONNECTED) {
          Serial.println("");
          Serial.println("WiFi connected.");
          Serial.println("IP address: ");
          Serial.println(WiFi.localIP());
          String ipa = WiFi.localIP().toString();
          display.clearDisplay();
          robojaxText("Connected: ", 32, 0, 1, true);
          robojaxText(ipa, 20, 12, 1, true);
          delay(5000);
          return;
        }
      }
      else {
        i = 10;
        display.clearDisplay();
        j = j + 1;
      }
      if (j > 1) {
        display.clearDisplay();
        reconnectInterval = 50;
        robojaxText("Failed to Connect ", 12, 8, 1, false);
        robojaxText("Retring in ", 6, 23, 1, false);
        robojaxText(String(reconnectInterval), 72, 23, 1, false);
        robojaxText("second", 92, 23, 1, false); //second
        display.display();
        delay(5000);
        previousMillis = currentMillis;
        j = 0;
      }
    }
  }
}