tzapu/WiFiManager #1461: SSID and password are not saved after configuration In The ESP32

pandahvaz, 2022-07-26:
#### Hardware
WiFimanager Branch/Release:  2.0.10-beta

Esp32: 

Hardware:			ESP32-Wroom 32D

Arduino IDE 1.8.12

### Description

My problem is that the SSID and password are not saved after configuration. I upload the sample code in the EPS32 and configuration SSID And Password and Connect to Network. Every things is great. , but after power failure or restart, all information related to ID and password is lost and I have to re-configure. that is not good at all.

This problem only exists for ESP32 board.
"There is no problem when I run the same code on ESP8266. 

### Settings in IDE

Module:  			ESP32-Wroom 32D

Additional libraries: Version 2.0.10-Beta

### Sketch

```cpp
#BEGIN
/**
 * OnDemandConfigPortal.ino
 * example of running the configPortal AP manually, independantly from the captiveportal
 * trigger pin will start a configPortal AP for 120 seconds then turn it off.
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

// select which pin will trigger the configuration portal when set to LOW
#define TRIGGER_PIN 0

int timeout = 120; // seconds to run for

void setup() {
  WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP  
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("\n Starting");
  pinMode(TRIGGER_PIN, INPUT_PULLUP);
}

void loop() {
  // is configuration portal requested?
  if ( digitalRead(TRIGGER_PIN) == LOW) {
    WiFiManager wm;    
 
    // set configportal timeout
    wm.setConfigPortalTimeout(timeout);

    if (!wm.startConfigPortal("OnDemandAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      //reset and try again, or maybe put it to deep sleep
      ESP.restart();
      delay(5000);
    }

    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

  }

  // put your main code here, to run repeatedly:
}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2022-07-26:
Only time i have seen this is when flash is corrupt, a full nvs erase might fix it

pandahvaz, 2022-07-27:
I have used several types of ESP32 and still had the same problem.
At first, I used the  ESP32 module and programmed it with [ESP32 WROOM PGA Programmer](https://www.robotics.org.za/ESP32-WROOM-PGA).

Then I did the same thing on the development board like as [DOIT ESP32 DEVKIT V1](https://makeradvisor.com/tools/esp32-dev-board-wi-fi-bluetooth/).
The same problem persisted in both Board.

tablatronix, 2022-07-27:
What does your serial log look like?
what esp lib verison are you on, try to update it maybe ?

nelsonii, 2022-08-12:
I run an Adafruit QT Py ESP32-C3 and had the same problem. Changing the Flash Mode (Arduino > Tools > Flash Mode) solved the problem for me. Drove me crazy!

Flash Mode: QIO  : IMPORTANT -- THE C3 CHIP MUST USE QIO (not the default of DOUT).

tablatronix, 2022-08-12:
oh wow interesting, maybe I can check that flash gets saved, I would expect esp lib to do this already, maybe I will raise an issue there also

nelsonii, 2022-08-12:
Oh, lordy. It was driving me nuts. I finally found something that mentioned QIO for the -C3. If you need anyone to test against the  Adafruit QT Py ESP32-C3, I'm your guy. I'm using this board for a number of things.

And, let me say: Thank You. This WiFi Portal is wonderful. I appreciate the effort you've put into it. I work on Assistive/Adaptive Technology, and this is being used to help folks control their hospital beds by voice. This is making a difference.

tablatronix, 2022-08-17:
I have the same issue on an S2, wont save no matter what.. wtaf

tablatronix, 2022-08-17:
Its a bug, works in stable, does not work in 
*wm:[1] ESP SDK version:  v4.4-beta1-308-gf3e0c8bc41

tablatronix, 2022-08-24:
Its bad bootloader flash mode, have to clear pio cache and frameworks and reupload

closed 2022-08-24 by tablatronix

crasu, 2023-01-14:
pio system prune solved the issue for me