tzapu/WiFiManager #1462: Possible to keep over the air uploads when connected to wifi?

jackkitley, 2022-07-27:
### Basic Infos

#### Hardware
WiFimanager Latest

Esp8266: ESP 12F

Hardware:			ESP-12f

Core Version:      	2.4.0, staging

### Description

I wanted to know when connected to WIFI that i could access the assigned IP address to still be able to upload .bin files. Is this possible?

Thanks

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here
```

tablatronix, 2022-07-27:
Yes, just you normally would

closed 2022-07-27 by tablatronix

jackkitley, 2022-07-27:
Hmm, when the AP is up its 192.168.4.1 but when its connected to router then if i access the IP i get nothing. So it doesnt spin up an interface to configure settings. Im wanting to upload a new .bin file.

tablatronix, 2022-07-27:
oh you mean httpupload? not arduinoota?

use startwebportal() ondemand example

jackkitley, 2022-07-27:
Yes sir.

I did try startwebportal() from your example but didnt seem to work. ill try again with my D1 mini.

So startwebportal() serves the exact same AP page?

Thank you.

tablatronix, 2022-07-27:
Yes just without starting the softap configportal
configportal is softap+httpd+captiveportal
webportal is httpdserver only

https://github.com/tzapu/WiFiManager/blob/master/examples/OnDemand/OnDemandWebPortal/onDemandWebPortal.ino

jackkitley, 2022-07-27:
> WiFiClient espClient;

> void setup() {
  Serial.begin(9600);

  pinMode(TRIGGER_PIN, INPUT);
//  pinMode(LED_BUILTIN, OUTPUT);

  loadConfigFile();
  setupWifiManager();
}

void loop() {
//  checkButton();

  reconnect_mqtt();
  client.loop();

//  wifiManager.startWebPortal();
}

> setupWifiManager

jackkitley, 2022-07-27:
Thank you.
Huge help
[image]

jackkitley, 2022-07-28:
@tablatronix Just one more question. I had params that i setup (in code above) but when i spin up the portal and go to /p or setup then there are no params.

The params i use are for MQTT settings? How can i get them to the portal as well as initial AP setup.

Thanks

tablatronix, 2022-07-29:
Sounds like a scope issue, are you setting your params in setup ?
You will need to make them global or static somehow