tzapu/WiFiManager #1463: WiFiManagerParameters not shown in non-blocking mode

paopre, 2022-07-31:
### Basic Infos

Additional WiFiManagerParameters are not shown with  setConfigPortalBlocking(false).
See the attached code. It's the same code of [AutoConnectNonBlocking.ino](https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/AutoConnectNonBlocking/AutoConnectNonBlocking.ino), with two added lines.

#### Hardware
WiFimanager Branch/Release:  2.0.11-beta

Esp8266/Esp32: Esp8266

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu

```cpp
#BEGIN
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP    
    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //reset settings - wipe credentials for testing
    //wm.resetSettings();
    
    WiFiManagerParameter custom_text("<p>This is just a text paragraph</p>");
    wm.addParameter(&custom_text);
    
    wm.setConfigPortalBlocking(false);
    wm.setConfigPortalTimeout(60);
    //automatically connect using saved credentials if they exist
    //If connection fails it starts an access point with the specified name
    if(wm.autoConnect("AutoConnectAP")){
        Serial.println("connected...yeey :)");
    }
    else {
        Serial.println("Configportal running");
    }
}

void loop() {
    wm.process();
    // put your main code here, to run repeatedly:
}
#END
```

paopre, 2022-07-31:
Just checked there's an example that shows how to fix this:
https://github.com/tzapu/WiFiManager/tree/master/examples/NonBlocking/AutoConnectNonBlockingwParams

closed 2022-07-31 by paopre

tablatronix, 2022-07-31:
Yeah your params only exist in setup()