tzapu/WiFiManager #1467: WifiManager library on ESP32-S3. Form mobile phone cant see esp32s3 Access point

ains98, 2022-08-05:
WifiManager library on ESP32-S3. Form mobile phone or another computer cant see esp32s3 Access point.

ains98, 2022-08-07:
if using lattes [release/v5.1.0'](https://github.com/platformio/platform-espressif32/commit/a28daed18bcea98708bc2bd207da28034fe7e0fd) 
WiFi manager softAP does`t working i go back to 4.4.0 then works everything

tablatronix, 2022-08-08:
hmm

ains98, 2022-08-10:
With PlatformIO core V5.1.0 ESP32-s2-solana dev board also not  working with WiFiManager lib.
Cant find SSID AutoConnectAP 
I used 4 different mobile phones and 2 different lapto PC to find esp32 in WiFi list

platform.ini:
[env:esp32-s2-saola-1]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
monitor_speed = 115200

lib_deps =
    https://github.com/tzapu/WiFiManager.git

Terminal output:
SPIFFS mounted
*wm:[1] AutoConnect
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client...
*wm:[2] setSTAConfig static ip not set, skipping
*wm:[1] Connecting to SAVED AP: espressif
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED
*wm:[2] Starting Config Portal
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[2] AP has anonymous access!
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started
*wm:[2] Config Portal Running, blocking, waiting for clients...
*wm:[2] NUM CLIENTS:  0

Code i`m using for this test:

```c++
#include <Arduino.h>
#include <SPIFFS.h>
#include <FS.h>
#include <WiFiManager.h> 

void setup() {
  Serial.begin(115200);
  if (!SPIFFS.begin(true))
  {
    Serial.println("SPIFFS failure");
  }
    Serial.println("SPIFFS mounted");

    WiFiManager wifiManager;
   
    wifiManager.autoConnect("AutoConnectAP");
    
    Serial.println("connected...yeey :)");
}

void loop() {
  // put your main code here, to run repeatedly:
}
```

tablatronix, 2022-08-10:
try updating your esp library, there was a bug recently if i recall about no aps found...

*wm:[1] Connecting to SAVED AP: espressif
*wm:[1] connectTimeout not set, ESP waitForConnectResult...
*wm:[2] [EVENT] WIFI_REASON: 201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND
*wm:[2] Connection result: WL_NO_SSID_AVAIL

ains98, 2022-08-10:
I tested with esp32-S3 Dev and ESP32-S2 dev both have same issue.
Only works with old ESP32 dev board.

if i try in main.cpp file 

  WiFi.mode(WIFI_AP);
  WiFi.softAP("ESP32-Dev");

i can find SSID and connect to it.
using lattes platformIO Core @ 6.1.3

tablatronix, 2022-08-10:
Try enabling esp debug and see if there is anything useful, I will test my S2 later, but it worked last time I tried it.
Also what exact esp lib/arduino version are you on?

tablatronix, 2022-08-10:
```
[env:esp32-s2-saola-1]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino

; change microcontroller
board_build.mcu = esp32s2

; change MCU frequency
board_build.f_cpu = 240000000L
```

```
*wm:[1] Free heap:        188476
*wm:[1] ESP SDK version:  v4.4.1-1-gb8050b365e
```

[image]

tablatronix, 2022-08-10:
[image]
[image]

tablatronix, 2022-08-10:
Shrug works for me

letis009, 2022-08-21:
awesome library but ... same issue with my c3 
works 100% fine with esp32-wrover-iB but no visible AP on esp32-c3-wroom-02
using latest version available today

using arduino IDE 1.8 and tested c3 with other wifi libraries to scan , connect and host AP with successful results

```
16:05:45.560 -> entry 0x403ce000
16:05:45.702 -> [   193][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
16:05:45.750 -> [   224][V][WiFiGeneric.cpp*wm:[1] resetSettings 
16:05:46.265 -> [   769][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
16:05:46.265 -> [   770][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
16:05:46.265 -> *wm:[1] SETTINGS ERASED 
16:05:46.265 -> *wm:[1] AutoConnect 
16:05:46.313 -> [   789][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
16:05:46.313 -> [   795][V][WiFiGeneric.cpp:338] _arduino_event_cb(): STA Started
16:05:46.313 -> [   795][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
16:05:46.313 -> *wm:[1] No Credentials are Saved, skipping connect 
16:05:46.313 -> *wm:[2] Starting Config Portal 
16:05:46.313 -> *wm:[2] AccessPoint set password is VALID 
16:05:46.313 -> [   820][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
16:05:46.313 -> [   820][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
16:05:46.313 -> *wm:[2] Disabling STA 
16:05:46.359 -> *wm:[2] Enabling AP 
16:05:46.359 -> *wm:[1] StartAP with SSID:  AutoConnectAP
16:05:46.359 -> [   840][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
16:05:46.359 -> [   850][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
16:05:46.359 -> [   850][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
16:05:46.833 -> *wm:[1] AP IP address: 192.168.4.1
16:05:46.881 -> *wm:[1] Starting Web Portal 
16:05:46.881 -> [  1352][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
16:05:46.881 -> [  1353][V][WebServer.cpp:85] WebServer(): WebServer::Webserver(port=80)
16:05:46.881 -> *wm:[2] HTTP server started 
16:05:46.881 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
16:06:15.526 -> *wm:[2] NUM CLIENTS:  0
```

tablatronix, 2022-08-22:
hmm, I have a C3 I will check it, and you are sure its not crashing or anything? Just no softap, does softap start throw an error with esp debug on ?

letis009, 2022-08-22:
thank you for coming back 
I opened an issue with a lot more details after a day of testing 
[#1479](https://github.com/tzapu/WiFiManager/issues/1479) 
i don't know if this is related issue

cniedzi, 2022-09-03:
Board: Lolin S2 mini
ESP32 S2
WiFiManager ver. 2.0.12-beta
esp32 core ver. 2.0.4

CASE 1 - AP "ZegarAP" not visible!!!

```
WiFiManager wm;
wm.resetSettings();
wm.autoConnect("ZegarAP");
```

Debug outcome:
```
*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  ZegarAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
```

CASE 2 - AP "ZegarAP" visible!!!

```
WiFiManager wm;
wm.startConfigPortal("ZegarAP");

```

Debug outcome:
```
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  ZegarAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
```

CASE 3 - AP "ZegarAP" not visible!!!

```
WiFi.mode(WIFI_STA);
WiFiManager wm;
wm.startConfigPortal("ZegarAP")

```

Debug outcome:
```
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  ZegarAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
```

tablatronix, 2022-09-04:
#1482

tablatronix, 2022-09-04:
Investigating and working on a fix

closed 2022-09-04 by tablatronix