tzapu/WiFiManager #1468: What does WIFI_REASON:  4 Mean

GeorgeIoak, 2022-08-06:
I'm trying to debug a minor problem that happens when I change to a new network. Everything works fine but after it restarts I need to manually restart the board once or twice before the application runs properly. This only happens happen I run WM. When I check the log I find this:

```
parsed json
*wm:[2] Added Parameter: key_custom
*wm:[2] Added Parameter:
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: TheIoaks
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] [EVENT] WIFI_REASON:  4
```
and at this point I need to restart the board. I searched but I cannot seem to find the meaning of "4"

tablatronix, 2022-08-09:
WIFI_REASON_ASSOC_EXPIRE

GeorgeIoak, 2022-08-10:
So that seems to be not expected after you've entered the new WiFi AP information, correct?

tablatronix, 2022-08-10:
Is it every other reboot?

GeorgeIoak, 2022-08-10:
You know it might be. I need to do some additional debugging before I really bother you but that's why I was trying to understand the normal flow. In general though it seems that once I change the WiFi AP I have to manually reboot twice before the board runs the code normally. After that it always runs fine.

tablatronix, 2022-08-10:
Yeah its a known esp issue #1137

GeorgeIoak, 2022-08-11:
Reading through the older issues and replies am I correct that a workaround for this issue is to add a `setConnectRetries(3)` ?

tablatronix, 2022-08-11:
You can try that, yes, and add longer connect timout, this issue only presents with some routers, you can also try 

    bool          _aggresiveReconn        = true; // use an agrressive reconnect strategy, WILL delay conxs
                                                   // on some conn failure modes will add delays and many retries to work around esp and ap bugs, ie, anti de-auth protections
                                                   // https://github.com/tzapu/WiFiManager/issues/1067

tablatronix, 2022-08-11:
#1067

GeorgeIoak, 2022-08-11:
I'm developing in VSCode with PIO and it's showing 5.1.0 is installed which was recently updated I see (11 days ago). I'll review #1067 and run some tests to see if anything changes for the better.

closed 2022-09-12 by tablatronix