tzapu/WiFiManager #1477: Help request: Use WifiManager as a web configuration portal

BUSHA, 2022-08-16:
### Basic Infos
I need your help with a very simple feature - use WifiManager as a web configuration portal.
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266

### Description

I read almost all corresponding issues about this feature and I pretty sure that it's possible. But I'm newbie in the arduino world, so I have no clue how to implement this.

I need next:

- Keep all functionality that WiFiManager provides right now, I mean initial setup through the AP.
- Keep current ability to automatically create the AP if saved wifi network isn't available and setup a new network.
- Add a custom input field to the setup page with a pre-filled link and ability to modify it and save it in the SPIFFS or EEPROM.
- Keep the web server running after the initial setup (Without AP, in STA mode of course)
- Have an ability anytime access the web portal through the device ip in the browser and change this link in the custom field.
- This link will be used as a source of JSON data for the project, so I need a way to use it in the code.

Appreciate you help, thank you!

### Settings in IDE

Module: Wemos D1

walles12, 2022-11-12:
You should have a look at https://github.com/maakbaas/esp8266-iot-framework

tablatronix, 2022-11-12:
Yeah this can all be done already, read the code, its all commented. and check the super example

closed 2022-11-12 by tablatronix