tzapu/WiFiManager #1478: ESP32-S2 crashes instantly on autoConnect() ?

nullstalgia, 2022-08-20:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP32-S2

Hardware: ESP32-S2 Mini (Wemos/Lolin board)

Core Version:      	5.0.0

### Description

Chip instantly crashes upon using autoConnect().

### Sketch

```cpp
#include <Arduino.h>
#include <WiFiManager.h>
WiFiManager wifiManager;
#include "USB.h"

#if ARDUINO_USB_CDC_ON_BOOT
#define HWSerial Serial0
#define USBSerial Serial
#else
#define HWSerial Serial
USBCDC USBSerial;
#endif

void setup() {
    Serial.begin(115200);
    delay(5000);

    wifiManager.setDebugOutput(true);
    wifiManager.autoConnect();
 
}

void loop() {}
```

Platformio.ini:

```
[env:esp32-s2-saola-1]
platform = espressif32
board = esp32-s2-saola-1
framework = arduino
monitor_speed = 115200
build_flags = 
	-D ARDUINO_USB_CDC_ON_BOOT
lib_deps = 
	https://github.com/tzapu/WiFiManager.git
```
### USB Serial Output:

Output from above sketch:

```
--- Terminal on COM9 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
*wm:[1] AutoConnect 
```

Since I'm using the CDC, I don't get a stack trace, and I'm actually not sure if the proper UART GPIOs are broken out on this board.

tablatronix, 2022-08-20:
Enable esp debug, what is cdc? Does it work when you remove it?

nullstalgia, 2022-08-20:
USB CDC == USB communications device class, basically the technical terms for the USB <-> UART interface.

If I remove it and *all* other Serial mentions, I don't get any Serial output over USB, and it still crashes. 

Adding ` -DCORE_DEBUG_LEVEL=5` to the build flags gives no extra output over the USB Serial.

I may just have to pray and hope that the proper UART pins *are* broken out and hook up to them to *see* that extra ESP output + the stack trace.

tablatronix, 2022-08-20:
hmm, never used it ill test out your basic sketch, I assumed you tried IDF stable and master?

nullstalgia, 2022-08-20:
Not sure how to switch between those on PlatformIO. I'm just using the supplied ESP32 Arduino core.

I'll gladly test whatever I need to in order to get this working.

Unfortunately, it looks like I may be boned in terms of IO?

The ESP32-S2's UX0 is on GPIOs 44 and 43 

[image]

[image]

But this board lacks these pins. Oh boy.

nullstalgia, 2022-08-20:
Huh... I had success with the tiny sketch once when using `WiFi.mode(WIFI_STA);` at the very start. But this success did not carry over to any future attempts.

tablatronix, 2022-08-20:
I thought the soala did not break our usb unless you move the jumpers? That is why they added 2 usb on the s3

nullstalgia, 2022-08-20:
Sadly, I don't have a Saola board. I have the one below it, the LOLIN ESP32-S2 Mini. I was just using the picture as a reference for the ESP32-S2's pins.

But looking closely at it, it looks like the Saola doesn't even have jumpers for switching the USB port, just the USB pins broken out on the side.

nullstalgia, 2022-08-21:
So, after doing some tests and research, I have some stuff to share.

When testing the board's WiFi with a normal `Wifi.begin()`, it would also crash!

It turns out that the ESP32-S2 board I have is a clone of the Lolin S2 Mini, and has a flaw in the copy: the capacitor on the reset line is missing its connection to GND. Scraping the soldermask of the GND plane directly near it and using a solder blob to connect it to the capacitor works with fixing *A* crash when Wifi is being used.

I was able to write the rest of my sketch (send a payload to my MQTT server and then go into deep sleep) to make sure it works with the capacitor fix, and it does.

*However,* this does not fix my issue with the library.

~~If I try to use autoConnect() still, it does just crash.~~ ok or not anymore

i swear im going mad

Now it does work?? But autoConnect() consistently fails until I put `WiFi.mode(WIFI_AP);` before autoConnect()

tablatronix, 2022-08-22:
I heard of another issue maybe with this board maybe, not sure, you have to decrease the wifi power or it wont initialize wifi, as there is not enough power or something?

nullstalgia, 2022-08-24:
Well, part of the issue is that I already have a PCB made for this footprint in mind 😅 

Was working fine in preliminary tests with RGB and deep sleep, and I guess the clone's engineers were also lulled into the false sense of security.

I'll be sure to do more robust testing in the future :P

But, it seems that the capacitor bodge+lowering WiFi power might be my best best to avoid needing to spend more money.

Thank you for bearing with me!

tablatronix, 2022-08-31:
Have you tried clearing and reloading the esp lib or platform in pio?

NightSkySK, 2022-09-02:
I have ESP32-S2-Soala board, and also have problems with WiFi autoConnect(). In my case it doesn't crash, but even if logs show that AccessPoint is created I can't detect it by any device:
```
*wm:[1] AutoConnect 
*wm:[2] ESP32 event handler enabled 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to SAVED AP: espressif

*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[2] AP has anonymous access! 
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0
```

tablatronix, 2022-09-02:
#1482

closed 2022-09-02 by tablatronix

Vangreen, 2023-04-16:
> So, after doing some tests and research, I have some stuff to share.
> 
> When testing the board's WiFi with a normal `Wifi.begin()`, it would also crash!
> 
> It turns out that the ESP32-S2 board I have is a clone of the Lolin S2 Mini, and has a flaw in the copy: the capacitor on the reset line is missing its connection to GND. Scraping the soldermask of the GND plane directly near it and using a solder blob to connect it to the capacitor works with fixing _A_ crash when Wifi is being used.
> 
> [image]
> 
> [Source of image and fix.](https://esp32.com/viewtopic.php?f=19&t=28506&sid=a5e7a6fff370eebb744a5e758b6cd18f&start=10)
> 
> I was able to write the rest of my sketch (send a payload to my MQTT server and then go into deep sleep) to make sure it works with the capacitor fix, and it does.
> 
> _However,_ this does not fix my issue with the library.
> 
> ~If I try to use autoConnect() still, it does just crash.~ ok or not anymore
> 
> i swear im going mad
> 
> Now it does work?? But autoConnect() consistently fails until I put `WiFi.mode(WIFI_AP);` before autoConnect()
> 
> By fails, I mean the AP it makes when a connection fails does not appear on my phone. And if it has credentials saved, connection attempts _will_ fail.

Thanks, works for me

Aggeloz, 2023-11-16:
> So, after doing some tests and research, I have some stuff to share.
>
> When testing the board's WiFi with a normal `Wifi.begin()`, it would also crash!
>
> It turns out that the ESP32-S2 board I have is [...]

Funny thing is, my clone board has all its connections correct. I tested with a multimeter and even did the fix and it still gives me the same problem, i even added a capacitor between the reset button and GND and it still has the same issues. I dont understand what else could be causing this.

tablatronix, 2023-11-16:
What is the exact crash? Could still be bad power, or memory issue, try full erase and power with an actual power supply

Aggeloz, 2023-11-16:
> What is the exact crash? Could still be bad power, or memory issue, try full erase and power with an actual power supply. There can also be crashes if your flash mode is set wrong

I get no messages from the monitor cause im using CDC for console output but its none of those things you mentioned. My boards crash as soon as i get my fingers close to the antenna or i put it on a breadboard, also if its already sitting on a breadboard it either crashes the CDC output or the board basically never boots.

tablatronix, 2023-11-17:
You will have to use a serial adapter to get logs probably

Aggeloz, 2023-11-17:
I'll set it up to keep the core dump on flash memory and read it after, it's kind of impossible to connect a serial adapter to it since the rx/tx pins are not broken out.

Aggeloz, 2023-11-17:
Im getting no errors at all, the chip just hangs and then restarts. Its really weird behavior. Im gonna attempt to attach an external serial to the chip but i doubt im going to manage it because the solder points for the cpu are minuscule.

tablatronix, 2023-11-17:
Yeah you can output serial on any pins

Aggeloz, 2023-11-18:
So i did more testing and it seems like when im using an external serial connection instead of CDC the board doesnt crash at all, so im willing to bet its the CDC instead of the wifi connection thats causing the issue. Im gonna do more testing but for now that all i have.

VaAndCob, 2023-11-28:
From my experience,  ESP32-S2 Lolin have trouble with WiFi,  I have more than 10 modules and they have the same behaviour.
I found that you must throttle wifi transmit power down after the first time wifi connected.  
default tx power is max, that makes wifi unstable and hot.
Connecting other sensors or modules directly to the Vbus (5V) pin could also make wifi unstable.

Aggeloz, 2023-11-28:
I know, ive tried setting the tx power alot lower but it didnt help at all.

PotatoFi, 2025-04-15:
Just adding that in 2025, on some clone S2 minis that I unknowingly ordered from Aliexpress, I ran into this problem and the solder blob fixed it on both units.