tzapu/WiFiManager #1479: Loading example OnDemand.ino does not result in a portal when credentials are not already saved

letis009, 2022-08-22:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp32:           ESP32-C3

Hardware:			ESP32-C3-Wroom-02

Core Version:      	ESP32 Arduino 2.0.4 based on ESP-IDF 4.4.2

### Description

Config Portal is not visible on any device, only after flashing example sketch with correct SSID and password does the OnDemand button work to enable Portal 

after credentials saved the portal is accessible via button press

tried to reduce tx power, set hostname, country, channel and removed password on the autoconnect AP but no difference on any of these changes 

### Settings in IDE

Module:  			ESP32 C3 Devkit

Additional libraries:  WifiManager.h version 2.0.12 beta

### Sketch with no AP

```
/**
 * This is a kind of unit test for DEV for now
 * It contains many of the public methods
 * 
 */
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <time.h>
#include <stdio.h>

#define USEOTA
// enable OTA
#ifdef USEOTA
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#endif

const char* modes[] = { "NULL", "STA", "AP", "STA+AP" };

unsigned long mtime = 0;

WiFiManager wm;

// TEST OPTION FLAGS
bool TEST_CP         = false; // always start the configportal, even if ap found
int  TESP_CP_TIMEOUT = 90; // test cp timeout

bool TEST_NET        = true; // do a network test after connect, (gets ntp time)
bool ALLOWONDEMAND   = true; // enable on demand
int  ONDDEMANDPIN    = 0; // gpio for button
bool WMISBLOCKING    = true; // use blocking or non blocking mode, non global params wont work in non blocking

char ssid[] = "za zda-ro-vye";  //  your network SSID (name)
char pass[] = "d15_muN9@13";       // your network password

//callbacks
  // called after AP mode and config portal has started
  //  setAPCallback( std::function<void(WiFiManager*)> func );
  // called after webserver has started
  //  setWebServerCallback( std::function<void()> func );
  // called when settings reset have been triggered
  //  setConfigResetCallback( std::function<void()> func );
  // called when wifi settings have been changed and connection was successful ( or setBreakAfterConfig(true) )
[... 776 lines cut ...]
15:21:40.188 -> [ 86195][V][Parsing.cpp:228] _parseRequest(): headerValue: Microsoft NCSI
15:21:40.188 -> [ 86201][V][Parsing.cpp:227] _parseRequest(): headerName: Host
15:21:40.188 -> [ 86207][V][Parsing.cpp:228] _parseRequest(): headerValue: www.msftconnecttest.com
15:21:40.234 -> [ 86214][V][Parsing.cpp:255] _parseArguments(): args: 
15:21:40.234 -> [ 86219][V][Parsing.cpp:238] _parseRequest(): Request: /connecttest.txt
15:21:40.234 -> [ 86225][V][Parsing.cpp:239] _parseRequest():  Arguments: 
15:21:40.234 -> [ 86231][E][WebServer.cpp:647] _handleRequest(): request handler not found
15:21:40.234 -> *wm:[2] <- Request redirected to captive portal 
15:21:40.234 -> [ 86248][W][WebServer.cpp:434] send(): content length is zero
15:21:40.513 -> [ 86513][V][WebServer.cpp:294] handleClient(): New client: client.localIP()=192.168.4.1
15:21:40.513 -> [ 86513][V][Parsing.cpp:121] _parseRequest(): method: GET url: /connecttest.txt search: 
15:21:40.513 -> [ 86517][V][Parsing.cpp:227] _parseRequest(): headerName: Connection
15:21:40.513 -> [ 86523][V][Parsing.cpp:228] _parseRequest(): headerValue: Close
15:21:40.513 -> [ 86529][V][Parsing.cpp:227] _parseRequest(): headerName: User-Agent
15:21:40.513 -> [ 86535][V][Parsing.cpp:228] _parseRequest(): headerValue: Microsoft NCSI
15:21:40.558 -> [ 86542][V][Parsing.cpp:227] _parseRequest(): headerName: Host
15:21:40.558 -> [ 86547][V][Parsing.cpp:228] _parseRequest(): headerValue: www.msftconnecttest.com
15:21:40.558 -> [ 86554][V][Parsing.cpp:255] _parseArguments(): args: 
15:21:40.558 -> [ 86559][V][Parsing.cpp:238] _parseRequest(): Request: /connecttest.txt
15:21:40.558 -> [ 86566][V][Parsing.cpp:239] _parseRequest():  Arguments: 
15:21:40.558 -> [ 86571][E][WebServer.cpp:647] _handleRequest(): request handler not found
15:21:40.558 -> *wm:[2] <- Request redirected to captive portal 
15:21:40.558 -> [ 86588][W][WebServer.cpp:434] send(): content length is zero

```

tablatronix, 2022-08-22:
This example seems old, are you sure you are using the latest git or beta and master branch?

letis009, 2022-08-22:
sorry for the delay . tested again 
I downloaded the 
https://github.com/tzapu/WiFiManager.git
version now and retested 
if I do  wm.resetSettings(); & wm.erase();  then after reboot i can not see the AP even those Debug clearly states it is running 
I also tried not calling wifi.mode(STA); to see if maybe it would make a difference but only flashing a sketch with SSID and password set then flashing OnDemandConfigPortal.ino from /examples/super/ do I see a AP after pressing button

tablatronix, 2022-08-22:
Maybe make sure the ap is not timing out, sometimes It takes me over 2 minutes to see an ap, just to be sure you are running it for longer that 120s etc.

What board do you have? I think I have a c3 mini dev board, ill test it

letis009, 2022-08-23:
I am using the custom board with bare module. power and uart broken out
2.5a 5v usb power 
3.3v ldo reg with in and output filter caps
Enable timing circuit
All as per Espressif hardware design specs
Max3232 ltt -> rs232 level converter 

Same exact build works 100% on ESP32-wrover-ie but i can only Assume the c3 is risc-v and the wrover is freertos could be a route issue or something in the timing from flash arrangement differences 

Instruction sets a bit out of my pay grade ATM 🤣

Tested time outs on 300s (5min) 
The ap shows up almost immediately if ssid:password present before hand 
It seems if the first attempt to connect to existing ap is successful then the ondemand works fine 

I cant get the setSSID and password setPreload to work
Is that still functional in latest beta ?

letis009, 2022-08-23:
I ran some more tests and set the portal timeout to 3000s (50 minutes) and same MO really 
no visible AP

tablatronix, 2022-08-23:
hmm I cannot get my C3 to boot, stuck in loop. Tried all kinds of pio ini options

letis009, 2022-08-23:
the primary use case I was testing was to use the config portal onDemand 
I have bodged a stable work around but it isnt as designed 

will flash the c3 with sketch containing credentials then flash the actual sketch 
this seems to work fine for now 

I will try get my hands on an S3 or similar to test

tablatronix, 2022-08-23:
ok i have the bootloader bug fixed, ill test now, I swear suck a waste of time, everytime I try to build esp tgheres some new bs.

tablatronix, 2022-08-23:
```
[WIFI] HOSTNAME: esp32c3-3FFA6C
*wm:[1] AutoConnect 
[   255][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.
```

oh well i give up for now

letis009, 2022-08-23:
where ... how does wifimanager store the credentials 
where would be a good place to get this type of info

tablatronix, 2022-08-23:
It uses the esp begin() native flash config

tablatronix, 2022-08-23:
Ok i have a working setup, but I am having some other strange issue. No ap because my code just stops executing?

```
[  3122][D][WiFiGeneric.cpp:966] _eventCallback(): WiFi Reconnect Running
[  3130][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
*wm:[2] [EVENT] WIFI_REASON:  201
*wm:[2] [EVENT] WIFI_REASON: NO_AP_FOUND 
*wm:[2] Connection result: WL_NO_SSID_AVAIL
*wm:[1] AutoConnect: FAILED 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
[  3162][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[  3165][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
...??????
```

tablatronix, 2022-08-23:
Ok I bypassed some stuff and now it starts the AP but I see no AP!!!!!

So yeah this is buggered

letis009, 2022-08-23:
It is weird . Thank you for looking into the issue

tablatronix, 2022-08-23:
Can you try this change?

`    bool          _disableSTA             = true; // disable sta when starting ap, always`

tablatronix, 2022-08-23:
well nm, i just removed all my testing and it still works.. I wonder if it is flash corruption..

letis009, 2022-08-23:
Will test and let you know results

tablatronix, 2022-08-24:
I had to use Jasons fork, I cannot get platform IO to work on 3.0.2 without crashing or idf 4.4.1-beta

```
[env:esp32-c3-devkitm-1_WORKING]
platform  = https://github.com/Jason2866/platform-espressif32.git
board_build.partitions = minimal.csv
board_build.flash_mode = qio
board = esp32-c3-devkitm-1
framework = arduino
```

What version esp framework are you using and if pio, whats your ini?

letis009, 2022-08-24:
spinning up a vm with clean vsc and pio 
will come back shortly with test results

tablatronix, 2022-08-24:
Its hardcoded in the .h file

letis009, 2022-08-24:
got you , Updating to espressif framework v5 and will test further

letis009, 2022-08-24:
Oops , clicked the wrong comment 

VScode with pio and esp-idf on version 5 seems to work 100%
what ever the issue is seems resolved

can close issue 
thanks for assistance

closed 2022-08-26 by letis009