tzapu/WiFiManager #1480: [FR] add option to change config url after setup

TomW1605, 2022-08-23:
i am using this for a project where i need to have a web ui with a few pages, i would like that to be the first thing shown when going to the IP of the device (eg it is bouned to `/`). but i would also like to be able to reconfigure the wifi settings without needing to mess arround with buttons or haivng a timeout. so idealy i could just have a button on the main page that takes you to the WiFiManager config page. but for that to work WiFiManager would need to be listening on a diffrent url.

could there be an option to change this?

tablatronix, 2022-08-30:
You can add your own endpoint handlers and override the defaults there is also a variable somewhere with the root

TomW1605, 2022-08-30:
thanks for the reply. i know i can define my own endpoints for my other pages and i assumed i could override the default ones. i also found where the root path is defined, however, i didnt want to edit the library because then if i need to compile this code on my other computer it wont work. i always try and avoid editing librarys unless it is 100% nesasery to maintain portablitaly.\

so i was hoping to add it as a function, i dont think it would be that hard and if it is something the project would be open too i would be happy to give it a shot and open a PR with the changes

tablatronix, 2022-08-30:
I wonder should I make the root configurable but ALSO the root path for all WM pages?

eg.

const char R_root[]               PROGMEM = "wm/";
const char R_wifi[]               PROGMEM = "R_root/wifi";

Just a thought.. hmm

TomW1605, 2022-08-30:
that is probably a good idea, it minimises the risk of having other clashes if you want to have your own page for other things like status or info