tzapu/WiFiManager #1481: The device does not generate the wifi network in Access Point Mode

MattiaCC93, 2022-08-26:
Hi,
I have successfully flashed my ESP32-C3 esp32-c3-devkitc-02, the system boots but the wifi network is not generated. Yet from the IoT device logs, everything seems to work. I cannot see the wifi network neither from my PC nor from my mobile.
I thank you in advance whoever helps me.

This is the code used:
`#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

}

void loop() {
    // put your main code here, to run repeatedly:   
   Serial.println("+++");
   delay(1000);
}`

These are the  Arduino IDE logs:
`
Creating esp32c3 image...
Merged 2 ELF sections
Successfully created esp32c3 image.
"C:\\Users\\dev\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.4/tools/gen_esp32part.exe" -q "C:\\Users\\dev\\AppData\\Local\\Temp\\arduino_build_927744/partitions.csv" "C:\\Users\\dev\\AppData\\Local\\Temp\\arduino_build_927744/app-client.ino.partitions.bin"
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
Using library WiFiManager at version 2.0.12-beta in folder: C:\Users\dev\Documents\Arduino\libraries\WiFiManager 
Using library WiFi at version 2.0.0 in folder: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WiFi 
Using library Update at version 2.0.0 in folder: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\Update 
Using library WebServer at version 2.0.0 in folder: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\WebServer 
Using library DNSServer at version 2.0.0 in folder: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\DNSServer 
Using library FS at version 2.0.0 in folder: C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4\libraries\FS 
"C:\\Users\\dev\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\riscv32-esp-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/riscv32-esp-elf-size" -A "C:\\Users\\dev\\AppData\\Local\\Temp\\arduino_build_927744/app-client.ino.elf"
Sketch uses 782790 bytes (59%) of program storage space. Maximum is 1310720 bytes.
Global variables use 34652 bytes (10%) of dynamic memory, leaving 293028 bytes for local variables. Maximum is 327680 bytes.
C:\Users\dev\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.3.0/esptool.exe --chip esp32c3 --port COM3 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x0 C:\Users\dev\AppData\Local\Temp\arduino_build_927744/app-client.ino.bootloader.bin 0x8000 C:\Users\dev\AppData\Local\Temp\arduino_build_927744/app-client.ino.partitions.bin 0xe000 C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.4/tools/partitions/boot_app0.bin 0x10000 C:\Users\dev\AppData\Local\Temp\arduino_build_927744/app-client.ino.bin 
esptool.py v3.3
Serial port COM3
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 7c:df:a1:ba:51:10
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000d7fff...
Flash params set to 0x022f
Compressed 12912 bytes to 9261...
Writing at 0x00000000... (100 %)
Wrote 12912 bytes (9261 compressed) at 0x00000000 in 0.4 seconds (effective 290.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 128...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (128 compressed) at 0x00008000 in 0.1 seconds (effective 399.0 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 576.3 kbit/s)...
Hash of data verified.
Compressed 819056 bytes to 483942...
Writing at 0x00010000... (3 %)
Writing at 0x0001a77a... (6 %)
Writing at 0x00024b30... (10 %)
Writing at 0x0003032d... (13 %)
Writing at 0x00038fa3... (16 %)
Writing at 0x0003ed67... (20 %)
Writing at 0x00045e7f... (23 %)
Writing at 0x0004c1a2... (26 %)
Writing at 0x000526e0... (30 %)
Writing at 0x00059a09... (33 %)
Writing at 0x0005faee... (36 %)
Writing at 0x00065e14... (40 %)
Writing at 0x0006bb12... (43 %)
Writing at 0x00071664... (46 %)
Writing at 0x00077854... (50 %)
Writing at 0x0007d866... (53 %)
Writing at 0x000834e8... (56 %)
Writing at 0x00089632... (60 %)
Writing at 0x0008ec9c... (63 %)
Writing at 0x00094c74... (66 %)
Writing at 0x0009b0a0... (70 %)
Writing at 0x000a0cab... (73 %)
Writing at 0x000a6e52... (76 %)
Writing at 0x000ad389... (80 %)
Writing at 0x000b389e... (83 %)
Writing at 0x000b9cc7... (86 %)
Writing at 0x000bfee8... (90 %)
Writing at 0x000c5d7c... (93 %)
Writing at 0x000cd6e4... (96 %)
Writing at 0x000d4671... (100 %)
Wrote 819056 bytes (483942 compressed) at 0x00010000 in 13.4 seconds (effective 488.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

These are the logs of the device:

`...
16:04:05.852 -> *wm:[1] SETTINGS ERASED 
16:04:05.852 -> *wm:[1] AutoConnect 
16:04:05.852 -> *wm:[1] No Credentials are Saved, skipping connect 
16:04:05.852 -> *wm:[2] Starting Config Portal 
16:04:05.852 -> *wm:[2] AccessPoint set password is VALID 
16:04:05.852 -> *wm:[2] Disabling STA 
16:04:05.852 -> *wm:[2] Enabling AP 
16:04:05.852 -> *wm:[1] StartAP with SSID:  AutoConnectAP
16:04:06.366 -> *wm:[1] AP IP address: 192.168.4.1
16:04:06.366 -> *wm:[1] Starting Web Portal 
16:04:06.366 -> *wm:[2] HTTP server started 
16:04:06.366 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
16:04:35.123 -> *wm:[2] NUM CLIENTS:  0
16:05:05.093 -> *wm:[2] NUM CLIENTS:  0
16:05:35.083 -> *wm:[2] NUM CLIENTS:  0
16:06:05.100 -> *wm:[2] NUM CLIENTS:  0
16:06:35.118 -> *wm:[2] NUM CLIENTS:  0
16:07:05.105 -> *wm:[2] NUM CLIENTS:  0
16:07:35.104 -> *wm:[2] NUM CLIENTS:  0
16:08:05.099 -> *wm:[2] NUM CLIENTS:  0
16:08:35.094 -> *wm:[2] NUM CLIENTS:  0
16:09:05.098 -> *wm:[2] NUM CLIENTS:  0
16:09:35.104 -> *wm:[2] NUM CLIENTS:  0
16:10:05.120 -> *wm:[2] NUM CLIENTS:  0
16:10:35.128 -> *wm:[2] NUM CLIENTS:  0
16:11:05.124 -> *wm:[2] NUM CLIENTS:  0
16:11:35.095 -> *wm:[2] NUM CLIENTS:  0
`

tablatronix, 2022-08-26:
#1422

tablatronix, 2022-08-26:
#1479

closed 2022-08-26 by tablatronix

tablatronix, 2022-08-31:
#1482