tzapu/WiFiManager #1482: Esp32 s3/c3 softap not working 2.3.0

tablatronix, 2022-08-26:
There are several open issues that all seem related to wifi being broken on esp32 s3 and c3, seems to work on the latest esp lib. Not sure if its a combination on wifimanager starting and stopping sta mode the ln starting softap making this hard to debug.

MattiaCC93, 2022-08-31:
Hi @tablatronix , 5 days ago i opened this issue about my esp32-c3 (#1481)
I just (5 minutes ago) tried with the latest version (**2.0.11-beta**) but the problem persists, that is the Access Point does not "generate" the wifi network.

Thanks

tablatronix, 2022-08-31:
Thanks yeah I am trying to gather the 3-4 issues into this one

FHanus, 2022-08-31:
Same issue with ESP32-S2. When I flash the ESP32 Wifi AP example it creates an access point, but the "basic" example of the WifiManager does not start an access point.

FHanus, 2022-08-31:
@tablatronix I know you already know this, but I want to confirm that downgrading to espcore version 2.3 in the board manager solves this.

tablatronix, 2022-08-31:
I can only get this to work on 2.0.4 sorry , fixing title

FHanus, 2022-08-31:
This is strange because for me the issue was on 2.0.4 and 2.0.3 fixed it

Ben-Rheinland, 2022-09-01:
Arduino IDE, Adafruit QT Py ESP32-C3, WiFiManager 2.0.12-beta:
Issue is present with Arduino-ESP32 2.0.4
No issues with Arduino-ESP32 2.0.3
Tested with WiFiManager "Basic" example

MattiaCC93, 2022-09-01:
Sorry, I have a perhaps trivial question.

When you write "version 2.0.3" or "version 2.0.4" what are you referring to?

Are you referring to [ **2.0.3-alpha**](https://github.com/tzapu/WiFiManager/releases/tag/2.0.3-alpha) and [v2.0.4-beta](https://github.com/tzapu/WiFiManager/releases/tag/2.0.4-beta)?

Thanks.

@Ben-Rheinland @FHanus

mazgch, 2022-09-01:
Hi @tablatronix, here is a hint:

Following fix in WiFiManager.cpp:717 will solve the problem on my 2.0.4 version and a ESP32S3, not sure about any side effects this might have.

```
// HANDLE issues with STA connections, shutdown sta if not connected, or else this will hang channel scanning and softap will not respond
  if(_disableSTA || (!WiFi.isConnected() && _disableSTAConn)){
    // this fixes most ap problems, however, simply doing mode(WIFI_AP) does not work if sta connection is hanging, must `wifi_station_disconnect` 
    WiFi_Disconnect();
    /// WiFi_enableSTA(false); <<< disable this 
    #ifdef WM_DEBUG_LEVEL
    DEBUG_WM(DEBUG_VERBOSE,F("Disabling STA"));
    #endif
  }
  else {
    WiFi_enableSTA(true);
  }
```

Also the WM_NOTEMP only works for the ESP32S3_DEV board any other ESP32S3 board will fail to compile this library. We should find another way of disabling the temperature for all ESP32S3 in the market.

Ben-Rheinland, 2022-09-01:
> When you write "version 2.0.3" or "version 2.0.4" what are you referring to?

This is referring to the esp32 platform by espressiv, installed in Arduino IDE using the boards manager: [Link to tutorial](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html)
> Are you referring to [ **2.0.3-alpha**](https://github.com/tzapu/WiFiManager/releases/tag/2.0.3-alpha) and [v2.0.4-beta](https://github.com/tzapu/WiFiManager/releases/tag/2.0.4-beta)?

No, I have the WiFiManager Library installed on version 2.0.12-beta.

FHanus, 2022-09-01:
> Sorry, I have a perhaps trivial question.
>
> When you write "version 2.0.3" or "version 2.0.4" what are you referring to?
>
> Are you referring to [ [...]

Confirming what @Ben-Rheinland said

FHanus, 2022-09-01:
@tablatronix  I keep encountering one more issue. When the ESP runs the basic example from the WifiManager, and gets unplugged before the WiFi manager AP turns off (not reset using button, but by unplugging the cable) it is impossible to connect to the AP when it turns back on.

tablatronix, 2022-09-01:
That might be something else, is it crashing?

I cannot even use 2.0.3 for c3 on platformio, it just boot loops

FHanus, 2022-09-01:
> That might be something else, is it crashing?
> 
> I cannot even use 2.0.3 for c3 on platformio, it just boot loops

That's the thing, after this forced reboot it goes to the same old problem of being stuck on the ESP printing out that zero clients are connected.

tablatronix, 2022-09-01:
hmmmmmmm lol

I have a patch to fix a wifi not init issue , let me see if that can help.

tablatronix, 2022-09-01:
ok so 2.0.3 esp c3 does not work afaik, and there is no patch until 2.0.4 I could be wrong.

https://github.com/espressif/arduino-esp32/issues/6862#issuecomment-1175926313

mazgch, 2022-09-01:
> @mazgch thanks, i have disable temp sensor on all for now
> 
> [201ad2e](https://github.com/tzapu/WiFiManager/commit/201ad2e838b2fe39295bf3819a46d02c49a37279)

Were you able to confirm that commenting line WiFiManager.cpp:717 solved this issue, is this a safe patch or do you see major side effects?

tablatronix, 2022-09-01:
Yeah same change, its not needed for anything, and i have no idea how to cleanly detect board

tablatronix, 2022-09-01:
I give up for now, I cannot get a damn thing to work on this c3...

```
ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381976
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x438
load:0x403ce000,len:0x918
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: 080c5cb68a075ced55f248b97bca965e3e5bd5da80a64e34e6a1638f89d6f64e
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000
*wm:[1] AutoConnect 
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40001be4  RA      : 0x42071f60  SP      : 0x3fca3e00  GP      : 0x3fc8ea00  
TP      : 0x3fc863f4  T0      : 0x40057fa6  T1      : 0x40383628  T2      : 0xffffffff  
S0/FP   : 0x3fc95a98  S1      : 0x40383b00  A0      : 0x3fc8fc34  A1      : 0x40383a02  
A2      : 0x40385288  A3      : 0x40383744  A4      : 0x4038377c  A5      : 0x3fcdf4f8  
A6      : 0x40383812  A7      : 0x403837e6  S2      : 0x3c0abc1c  S3      : 0x00000000  
````

tablatronix, 2022-09-01:
```
*wm:[3] WiFi_enableSTA enable
Guru Meditation Error: Core  0 panic'ed (Illegal instruction). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40001be4  RA      : 0x42071e7c  SP      : 0x3fca2a90  GP      : 0x3fc8ea00  
TP      : 0x3fc85130  T0      : 0x40057fa6  T1      : 0x40383628  T2      : 0xffffffff  
S0/FP   : 0x3fc95a98  S1      : 0x40383b00  A0      : 0x3fc8fc34  A1      : 0x40383a02  
A2      : 0x40385288  A3      : 0x40383744  A4      : 0x4038377c  A5      : 0x3fcdf4f8  
```

tablatronix, 2022-09-01:
Good news, my c3 devkit is garbage, I have another c3 board and it works, and i can reproduce this AND my previous suggestion does indeed work.. Still not sure why though...

`    bool          _disableSTAConn         = false;  // disable sta when starting ap, if sta is not connected ( stability )`

[image]

tablatronix, 2022-09-05:
I have not fixed it yet but you can do the .h mod i mentioned above a d see if that works

Miceuz, 2022-09-05:
I have tried the ` _disableSTAConn=false` mod, but still no AP. It's really late here, I will try a clean experiment tomorrow.

tablatronix, 2022-09-05:
Ok thanks for letting me know

Miceuz, 2022-09-06:
I have done a clean install via platformio. This is my platformio.ini:
```
[env:local]

platform=espressif32
framework = arduino

board = esp32-c3-devkitm-1
board_build.f_cpu = 160000000L

lib_deps = https://github.com/tzapu/WiFiManager.git

```
I am sure platformio pulls in esp32 code 2.0.4
I have added `bool _disableSTAConn = false;` hack into `WiFiManager.h`

Now if I have `WiFi.mode(WIFI_STA);` in my `setup()` function, I can't get AP to run, but if I change it to `WiFi.mode(WIFI_AP)` the AP starts up.

tablatronix, 2022-09-06:
Yeah there is something very wrong with changing modes, and even stranger no error on startap.. I have no idea whats going on. afaik its only the c3.. I will keep testing my guess is we are going to need something like an intermediate wifi_off or wifi_stop raw calls, or there is a missing event in the arduino esp handler maybe..

JohnWickham, 2022-09-07:
> Now if I have `WiFi.mode(WIFI_STA);` in my `setup()` function, I can't get AP to run, but if I change it to `WiFi.mode(WIFI_AP)` the AP starts up.

Thanks Miceuz, I wasn't able to get the config portal to run on ESP32-S2 for ages — using `WIFI_AP` instead of `WIFI_STA` got me there!

cniedzi, 2022-09-07:
> Now if I have `WiFi.mode(WIFI_STA);` in my `setup()` function, I can't get AP to run, but if I change it to `WiFi.mode(WIFI_AP)` the AP starts up.

I had the same issue with S2. I tried `bool _dbool _disableSTAConn = false; ` and so far so good.

tablatronix, 2022-09-07:
I opened an issue, and created a test sketch, now I can work on a fix/workaround.

https://github.com/espressif/arduino-esp32/issues/7232

shashi5841, 2022-09-10:
> There are several open issues that all seem related to wifi being broken on esp32 s3 and c3, seems to work on the latest esp lib. Not sure if its a combination on wifimanager starting and stopping sta mode the ln starting softap making this hard to debug.
> 
> Solution: USE 2.0.4

My issue solved now working esp 2.0.3 used with latest wifimanager
Below is the serial port data obtained:
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 
*wm:[2] WiFi Scan completed in 4701 ms
*wm:[1] 1 networks found
*wm:[2] AP:  -39 Shashi BSNL
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP WiFi save  
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to NEW AP: Shashi BSNL
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.1.4 
*wm:[2] shutdownConfigPortal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
connected...yeey :)

Thanks a lot ADMIN. Love your efforts keep it up..

tablatronix, 2022-09-10:
I have been trying to fix this and nothing I do fixes it, no workaround yet other that the one mentioned before
bool _disableSTAConn = false; // disable sta when starting ap, if sta is not connected ( stability )
seems to help avoid whatever gets stuck in the IDF

tablatronix, 2022-09-10:
Issue still exists in 2.0.4

tablatronix, 2022-09-12:
S2 will not show AP, if I disable disconsta, it shows up, but then refuses to let me join it...
This is a big problem, i have tried deinit wifi to no avail. Still testing

tablatronix, 2022-09-12:
Testing workaround for this

essentially turn on AP and STA then turn off STA, I have no idea what this is doing, startap is NOT working right, My guess is that wifi is getting turned off or losing cal before starting AP if we call espwifistop, and startap is not recovering. I tried wifi_off between states with no improvement

WiFi.mode(WIFI_AP_STA);
WiFi.mode(WIFI_AP);

Miceuz, 2022-09-12:
> WiFi.mode(WIFI_AP_STA);
> WiFi.mode(WIFI_AP);

This workaround works for me on esp32-c3 with esp 2.0.4 and master branch of WiFiManager.

cniedzi, 2022-09-13:
> If someone wants to help , maybe roll back esp core and see when it broke ( using the example min sketch in the arduino issue NOT WM )

ESP32 S2
Lolin S2 mini

2.0.0 ESP32S2 NativeUSB - OK
2.0.1 ESP32S2 NativeUSB- OK
2.0.2 ESP32S2 NativeUSB- OK
2.0.3 ESP32S2 NativeUSB/Lolin S2 mini - OK
2.0.4 ESP32S2 NativeUSB/Lolin S2 mini - FAILED!

cniedzi, 2022-09-13:
I think 2.0.4 core must be broken, because my S2 gets stuck after several recurring resets. In such cases I have to power off the device and wait few/dozen seconds before powering on...  I've rollbacked to 2.0.3. No such issues and no problems with Soft AP.

tablatronix, 2022-09-20:
2.0.5 is out, supposed to be fixed, I have to remove the workaround and test it.
I have no idea how to pull in 2.0.5 into platformio though. Let me install arduino

Miceuz, 2022-09-21:
@tablatronix you can pull 2.0.5 into platformio this way:

```
[env:local]

platform=espressif32
framework = arduino

platform_packages = framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.5

board = esp32-c3-devkitm-1
board_build.f_cpu = 160000000L

lib_deps = https://github.com/tzapu/WiFiManager.git
```

tobozo, 2022-11-15:
hi, 

I'm using esp32-arduino 2.0.5 (for Arduino IDE) along with the master branch of this repository and I noticed that WiFi.scanNetworks() does not use the second argument `show_hidden`.

is WiFiManager ignoring hidden AP by design? 

it doesn't seem very difficult to add the extra parameter so I assume there's a valid reason but can't wrap my head around it :thinking:  

using `wifiManager.setCleanConnect(true);` was a life changer anyway, thanks for the quality work!

tablatronix, 2022-11-15:
Isnt hidden the default?

Lalufu, 2023-04-15:
I've found this issue through searching for a solution to WM not working on my ESP32-C3, and from reading through it I'm a bit confused as to what the current state of this is. I'm on `framework-arduinoespressif32 @ 3.20007.0 (2.0.7)` and the current master checkout of WifiManager, and I don't see the hotspot SSID when the system starts without a config.
Is this supposed to work, or are some of the workarounds from above still necessary?

kunpoolsion, 2023-10-13:
I know is an old issue but maybe is worth it to revive it.
I was trying everything I found about it because I need to make an SoftAP in my project at certain point, but I was complete unable with the ESP32-C3.
Finally I was able to do it inserting:
`WiFi.setTxPower(WIFI_POWER_8_5dBm);`
right after any wifi.softap command in my project. I don´t mind the reduction of power in TX because the machine will be used very close to de user, but this depends on the project itself.
Maybe is usefull to integrate it in the library.

tablatronix, 2023-10-13:
What board are you using , this is a well known bug in wemos and other rev 1 boards as well as bad power supplies

kunpoolsion, 2023-10-13:
https://www.electrodragon.com/product/esp-led-strip-board/

Yeah, must be related with power, but probably with the power-down converter in the board, it doesn´t if I power the board by a psu at 5v directly, using the dc input in the board or by the usb-serial converter.

That means that this issue is fixed already in "normal" quality boards? :)

tablatronix, 2023-10-13:
Yeah I think alot of kind of copy the same schematic and hence the "bugs", have you tried whacking a cap on there?

kunpoolsion, 2023-10-13:
In 3.3v line?

bwjohns4, 2023-11-02:
I have had some modules where this is not an issue (on C3 modules), then other exact same modules from Espressif where the AP will not come on even though it "thinks" it's on. WM will be doing the portal countdown and everything likes it's working just fine but NO wifi SSID shows up to connect to. I'm not sure why some modules work without issue and others have an issue. I'll be testing some more but this seems to have solved the problem for me. Doing this on initial boot of the chip seems to solve this 
```      
WiFi.mode(WIFI_AP_STA);
WiFi.persistent(true);
WiFi.disconnect(true,true);
WiFi.persistent(false);
```

It seems some modules have something "stuck" in the SDK flash region that is fixed by wiping it like this.

Lalufu, 2023-11-09:
This sounds basically like what `WiFiManager::WiFi_eraseConfig()` does
on an ESP32?

Do you only have to do this once, or on every boot?