tzapu/WiFiManager #1493: error: 'base64' has not been declared

LouisLee985, 2022-09-14:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8285

Hardware:			ESP-01f

Core Version:      	2.4.0, staging

### Description

The same code compile in arduino is ok. 
But in platformio,there error.
Thanks. 

### Settings in IDE

Module:  			NodeMcu

### Debug Messages

```
In file included from C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:30,
                 from C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
                 from C:\Users\ABC\Documents\Arduino\libraries\WiFiManager\WiFiManager.h:86,
Compiling .pio\build\esp8285\FrameworkArduino\Esp-version.cpp.o
                 from C:\Users\ABC\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:
C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h: In member function 'bool esp8266webserver::StaticFileRequestHandler<ServerType>::handle(esp8266webserver::StaticFileRequestHandler<ServerType>::WebServerType&, HTTPMethod, const String&)':
C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/detail/RequestHandlersImpl.h:200:36: error: 'base64' has not been declared
  200 |         const String etag = "\"" + base64::encode(_ETag_md5, 16, false) + "\"";
      |                                    ^~~~~~
In file included from C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer.h:340,
Compiling .pio\build\esp8285\FrameworkArduino\Esp.cpp.o
                 from C:\Users\ABC\Documents\Arduino\libraries\WiFiManager\WiFiManager.h:86,
                 from C:\Users\ABC\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:
C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h: In member function 'bool esp8266webserver::ESP8266WebServerTemplate<ServerType>::authenticate(const char*, const char*)':
C:\Users\ABC\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WebServer\src/ESP8266WebServer-impl.h:105:24: error: 'base64' has not been declared
Compiling .pio\build\esp8285\FrameworkArduino\FS.cpp.o
  105 |       String encoded = base64::encode((uint8_t *)toencode, toencodeLen, false);
      |                        ^~~~~~
*** [.pio\build\esp8285\lib5ea\WiFiManager\WiFiManager.cpp.o] Error 1

```

tablatronix, 2022-09-14:
This seems to be a platform bug, or removed for size for the 8285? Either way not sure how I can fix other than to add a workaround. Maybe try to set
wm.setScanDispPerc(true); to disable images

LouisLee985, 2022-09-15:
Change board 8285 ESP-01f to nodemcuv2, and add  ```wm.setScanDispPerc(true);```  The error remains.
It seems to be a platform bug like you say.
So  edit code in PlatformIO IDE and compile it in  Arduino IDE.
Thanks.

tablatronix, 2022-09-15:
Yeah thats odd, or maybe you have duplicate library for webserver that is old, or need to wipe esp8266 platform

LouisLee985, 2022-11-02:
https://github.com/Mixiaoxiao/Arduino-HomeKit-ESP8266/pull/183/commits/ae4780e05cc1faa424baf9beb4aa96a0d181f896

tablatronix, 2022-11-02:
cool, so its a naming conflict

closed 2022-11-02 by tablatronix