tzapu/WiFiManager #1499: Custom menu with redirection?

eried, 2022-09-16:
Hi, I am using this custom menu: 

```
  std::vector<const char *> menu = {"wifi", "info", "param", "sep", "restart", "exit"};
  wm.setMenu(menu);
```

which works very good, but for example if you are in /param you dont have "Back" button, or if you press "Save" you get to a screen that says "Settings saved" but no button or redirection. 

I am missing something?

tablatronix, 2022-09-16:
Yeah, no there is no nav, I figured people would rather inject their own headers or something

there is an experimental flag hardcoded in the .h file you can try

`    boolean       _showBack               = false; // show back button`

eried, 2022-09-19:
oh, I see. Yeah, does not break anything, you can use the browser's back, but it just felt missing :)

eried, 2022-09-19:
> boolean _showBack = false;

this works well, thanks. The paramsave however does not show back

[image]

I guess here it would be nice if I just skip the portal timeout

eried, 2022-09-19:
Nice thanks! With the esp32 "Preferences", and lot of options this library is a lifesaver 😁