tzapu/WiFiManager #1500: Wifi Credentials not saved if disconnect from wifi after successful connect.

2000jago, 2022-09-20:
#### Hardware
WiFimanager Branch/Release:  Master

Esp8266:
Hardware:  ESP-12f, (wemos D1 Mini clone)
Core Version:  3.0.2

Description:
Wifi settings not saved if disconnect from wifi after making a connection.

Settings in IDE
Module:  NodeMcu, Wemos D1
Additional libraries:  none

Sketch:

```
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    WiFi.mode(WIFI_STA); // explicitly set mode, esp defaults to STA+AP
    // it is a good practice to make sure your code sets wifi mode how you want it.

    // put your setup code here, to run once:
    Serial.begin(115200);
    
    //WiFiManager, Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wm;

    // reset settings - wipe stored credentials for testing
    // these are stored by the esp library
    // wm.resetSettings();

    // Automatically connect using saved credentials,
    // if connection fails, it starts an access point with the specified name ( "AutoConnectAP"),
    // if empty will auto generate SSID, if password is blank it will be anonymous AP (wm.autoConnect())
    // then goes into a blocking loop awaiting configuration and will return success result

    bool res;
    // res = wm.autoConnect(); // auto generated AP name from chipid
    // res = wm.autoConnect("AutoConnectAP"); // anonymous ap
    res = wm.autoConnect("AutoConnectAP","password"); // password protected ap

    if(!res) {
        Serial.println("Failed to connect");
        // ESP.restart();
    } 
    else {
        //if you get here you have connected to the WiFi    
        Serial.println("connected...yeey :)");
    }

//do some stuff
  Serial.print("Disconnecting...");
  WiFi.disconnect();  // Disconnect from current wifi network
  delay (2000);
  Serial.println("...Done!");

}

void loop() {
    // put your main code here, to run repeatedly:   
}
```

```

### Debug Messages:
*wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP Wifi 
*wm:[2] WiFi Scan SYNC started 
*wm:[2] WiFi Scan completed in 2189 ms
*wm:[1] 2 networks found
*wm:[2] AP:  -65 ASUS
*wm:[2] AP:  -86 3BB-CHALERY
*wm:[2] <- HTTP Root 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- HTTP WiFi save  
*wm:[2] processing save 
*wm:[2] Connecting as wifi client... 
*wm:[2] setSTAConfig static ip not set, skipping 
*wm:[1] Connecting to NEW AP: ASUS
*wm:[1] connectTimeout not set, ESP waitForConnectResult... 
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] Connect to new AP [SUCCESS] 
*wm:[1] Got IP Address: 
*wm:[1] 192.168.2.142 
*wm:[2] shutdownConfigPortal 
*wm:[2] restoring usermode STA
*wm:[2] wifi status: WL_CONNECTED
*wm:[2] wifi mode: STA
*wm:[2] configportal closed 
*wm:[1] config portal exiting 
connected...yeey :)
Disconnecting......Done!

wm:[1] AutoConnect 
*wm:[1] No Credentials are Saved, skipping connect 
*wm:[2] Starting Config Portal 
*wm:[2] AccessPoint set password is VALID 
*wm:[2] Disabling STA 
*wm:[2] Enabling AP 
*wm:[1] StartAP with SSID:  AutoConnectAP
*wm:[1] AP IP address: 192.168.4.1
*wm:[1] Starting Web Portal 
*wm:[2] HTTP server started 
*wm:[2] Config Portal Running, blocking, waiting for clients... 
*wm:[2] NUM CLIENTS:  0
*wm:[2] NUM CLIENTS:  0

tablatronix, 2022-09-20:
Have you pulled master recently?

2000jago, 2022-09-20:
> Have you pulled master recently?

Am using V2.0.12-beta which the Arduino IDE Library Manager reports as the latest version and the releases section here on github appears to agree.  Should I download and import manually?

tablatronix, 2022-09-20:
ok this was fixed in git not released yet

closed 2022-09-21 by tablatronix