tzapu/WiFiManager #1501: Any reason why setConnectRetries has a constraint between 1 and 10?

sommerper, 2022-09-20:
Hi!

I have a device that's on a network that can have a network disconnection for longer periods of time.
Once the network is back I need the device to come back asap.
Retrying 10 times is not enough and ideally I'd like to retry an infinite amount of times.

What is the best way to achieve this without modifying the library directly in this piece of code?

```
/**
 * [setConnectRetries description]
 * @access public
 * @param {[type]} uint8_t numRetries [description]
 */
void WiFiManager::setConnectRetries(uint8_t numRetries){
  _connectRetries = constrain(numRetries,1,10);
}
```

Thank you and thanks for a great project!

tablatronix, 2022-09-20:
That is the connect retry for autoconnect, you do not want your coding sitting stuck in autoconnect like that nor should it be used like that.

You probably want to make your own loop for checking if connected or not, or call autoconnect itself in a loop

closed 2022-09-22 by sommerper

tablatronix, 2022-09-22:
Yeah technically we could do that but it's blocking and probably not what you want. It mostly to fix some aps that always reject the first conx