tzapu/WiFiManager #1506: QUESTION: How to get post data

ByronAP, 2022-09-28:
#### Hardware
WiFimanager Branch/Release:  Master
Hardware:			ESP32

### Description

Does anyone have an example on how to get post data? 

I use 
`wm.server->on("/saveschedule", HTTP_POST, handleSaveScheduleRoute);`

and

```
void handleSaveScheduleRoute()
    {
#if defined(DEBUG)
        Serial.println("[HTTP] handle save schedule");
#endif
        wm.server->send(200, "text/html", "hello you are at save schedule");
    }
```

which for some reason catches both get and post... but anyway I do not see a way to access the post data, anyone have an example on how to do this?

Thank you :-)

ByronAP, 2022-09-29:
OK so looking at the code the answer is quite simple since WM is using the default WebServer.h thus checking for a 'plain' argument to see if raw data was sent then getting the contents of the 'plain' arg.

Example handler:
```
void handlePostDataRoute()
    {
        if (wm.server->hasArg("plain") == false)
        {
            // no raw data posted
            wm.server->send(400);
            return;
        }

        // this is the data
        String body = wm.server->arg("plain");

        Serial.println(body);

        // return success
        wm.server->send(204);
    }

```

closed 2022-09-29 by ByronAP

tablatronix, 2022-09-29:
This is really an esp webserver question