tzapu/WiFiManager #1509: fatal error: esp_arduino_version.h: No such file or directory

SyukurCode, 2022-10-13:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			DOIT ESP32 DEVKIT V1.

Additional libraries: 

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None, Default"

Multiple libraries were found for "WiFi.h"

 Used: C:\Users\User\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi

 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.57.0_x86__mdqgnx93n4wtt\libraries\WiFi

In file included from C:\Users\User\AppData\Local\Temp\arduino_modified_sketch_684509\Basic.ino:1:0:

C:\Users\User\Documents\Arduino\libraries\WiFiManager-2.0.13-beta/WiFiManager.h:127:37: fatal error: esp_arduino_version.h: No such file or directory

compilation terminated.

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

```

tablatronix, 2022-10-13:
I am assuming this goes away in arduino 2.x

emersongcardona, 2022-10-14:
I have the same trouble and yes, it is in arduino 2.x

tablatronix, 2022-10-14:
Include arduino.h?

What enviroment is this?

commit dbefaed1 (2022-10-14): fix #1509
closed 2022-10-14 by tablatronix