tzapu/WiFiManager #1510: Checkbox startup values

AJTaylor79, 2022-10-13:
I seem to be having an issue with Checkbox startup values.

The file loads ok with the true/false value for the value of the checkbox however the variable that reference these always returns a 1.

As example emoncms_enable loads as false, goes through this routine as always returns a 1, this only happens on startup rather than if I go into the ondemand portal and change the values, they then seem to work ok. 

Any idea what the issue could be?

Thanks for your help

Andrew

### Sketch

```cpp
#BEGIN
  char customHtml_checkbox[] = "type=\"checkbox\""; //Can also be changed to "type=\"checkbox\"checked"; 
    if (emoncms_enable) {
      strcat(customHtml_checkbox, " checked");
    }
   WiFiManagerParameter custom_emoncms_enable_checkbox("emoncms_enable", "Enable Emoncms", "T", 2, customHtml_checkbox, WFM_LABEL_AFTER);
   char customHtml_checkbox_2[] = "type=\"checkbox\""; //Can also be changed to "type=\"checkbox\"checked"; 
     if (mqtt_enable) {
      strcat(customHtml_checkbox_2, " checked");
    }
  WiFiManagerParameter custom_mqtt_enable_checkbox("mqtt_enable", "Enable MQTT", "T", 2, customHtml_checkbox_2, WFM_LABEL_AFTER);

  emoncms_enable = (strncmp(custom_emoncms_enable_checkbox.getValue(), "T", 1) == 0);
  mqtt_enable = (strncmp(custom_mqtt_enable_checkbox.getValue(), "T", 1) == 0);

#END
```

### Debug Messages

```
messages here
```

AJTaylor79, 2022-11-26:
Has anyone got any thoughts on this as I would love to get it working? @tablatronix thanks

mylms, 2025-04-28:
I have the same issue. I have same code - ussualy works good, bud sometimes some checkbox (or all chbxs) are checked. I have to uncheck all in config portal and save.
Did you solve it?