tzapu/WiFiManager #1513: Is there a way to make the WiFiManager work with the corporate WiFi networks?

ateker01, 2022-10-24:
Sorry if already asked - I spent quite a bit of time but could not find any related. 

Is there a way to make the WiFiManager work with the corporate WiFi networks?

For example in the hotels, when you need to connect to the WiFi; there is no password but a redirected page where you enter your room number and last name... etc
Similarly with the corporate network, you enter a userid and password instead of the WiFi's password.

Is there such functionally with the WiFi Manager?

Thanks

tablatronix, 2022-10-24:
No, most are non standard captive portals

mahboud, 2023-04-01:
Here's an idea for getting around this issue. I don't know if it's possible.

After the ESP32 successfully gets on the WiFi, it should try to access a very simple webpage on the developer's site or any simple website. A good example is [Apple's](http://captive.apple.com). That page would only return a single easy to check value, such as "success".

If when accessing that site, you get a "success", then the ESP32 is done.

If you get a network error, then the WiFi network is bad. You might have a WiFi network that isn't connected to the Internet, or other issues (but that is not the topic of this discussion).

If, however, you get an HTTP response back that contains HTML, then what would be cool is if the WiFiManager's web server either puts that HTML in a frame or replaces the contents of its main page, and basically becomes a relay for the web page that is being used to get further info from the user.

That way, the user sees whatever that challenge is, whether to enter an email or a hotel room number and a last name, or just pressing some "Connect" button. Hopefully the HTML could be kept in a state where the user's actions will go directly to the original webpage. Or the WifiManager can forward that input.

If this all happened over HTTPS, then this wouldn't work, but because of the nature of some of these pages, where they try to redirect you to their webpage, the simple ones are plain old HTTP and not HTTPS.

At this point I expect to hear "we've already thought of that and here's why it doesn't work!" I would think that this issue has been discussed countless times, given how these networks prevent us from deploying our projects on the huge number of free public networks that just have that one or two extra steps you need to go through to get a connection.

adamsstephen, 2023-04-01:
I don't know why I get dragged in to this.

mahboud, 2023-04-05:
> I don't know why I get dragged in to this.

Sorry @adamsstephen. You had suggested this: https://github.com/tzapu/WiFiManager/issues/481#issuecomment-485091696