tzapu/WiFiManager #1514: Terminating WiFiManager's web portal

ambarusa, 2022-10-28:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266: 

Hardware:			ESP-12e,

Core Version:      	2.4.0, staging

### Description

Hi all. I'm having troubles with terminating WiFiManager's web portal, and starting my own. I know... there were issues opened before regarding this, but this case is different, because it's used in non-blocking mode, and a reset() would change the behavior in the system.

Briefly, after the software connects to the Wi-Fi, I want to change the webservers in the `WiFiEventStationModeGotIP` callback function. First I got exceptions, and it turned out because `stopConfigPortal()` was called by me, and later on by `wm.process()`. I solved this by setting `wm.setDisableConfigPortal(false);` which stops the `process()` function to care about closing the webserver.

But still, it seems like after calling the `wm.stopWebPortal();` the webserver doesn't close, however the `wm.getWebPortalActive()` tells the opposite. I could put in a loop my AsyncWebServer's `begin()` function, but I'm not sure if it's handled by the library, and there's no API go get it's status.

So, what do you suggest I should terminate the WiFiManager's web portal?

### Settings in IDE

Module:  			Wemos D1 Mini

Additional libraries:  SPAsyncWebServer

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

WiFiEventHandler wifiConnectHandler;

void onWifiConnect(const WiFiEventStationModeGotIP &event)
{
   wifiDisconnectHandler = WiFi.onStationModeDisconnected(onWifiDisconnect);
   MDNS.begin(hostname);
   OTA_init();
   wm.stopWebPortal();
   Webserver_start();
   Mqtt_init();
   Mqtt_connect();
}

void setup() {
   Webserver_init();
   Clock_init();
   wm.setDebugOutput(false);
   wm.setConfigPortalBlocking(false);
   WiFi.setAutoReconnect(true);
   wm.setHostname(hostname);
   wifiConnectHandler = WiFi.onStationModeGotIP(onWifiConnect);
   if (!wm.autoConnect(hostname))
   {
      Set_clock_state(CLOCK_STATE_AP);
   }
}

void 100_ms_task() {
   wm.process();
}
#END
```

### Debug Messages

```
messages here
```

inoffice, 2022-12-12:
Hi and good day
Sorry by my newie in WifiManager, and sorry if item comment before, but not found any. ambarusa say "terminating WiFiManager's web portal, and starting my own" How?... i try this, redirect to my own portal after Wifimanager conect to some Wifi, and redirecto my own Webserver create in my ESP32 D1 mini. I agree some info!! THANKS

tablatronix, 2023-09-08:
Found this today, hints at the same issue

https://github.com/me-no-dev/ESPAsyncWebServer/issues/777