tzapu/WiFiManager #1515: power outage

stef-ladefense, 2022-10-28:
a question, 
how to handle a power outage and a long reconnection to local wifi modem.
let me explain:
when the electric current returns after a cut, the esp's will start faster than the internet modem.
therefore, they will switch to portal mode after setConnectTimeout(45)and exiting without a connection after a setConfigPortalTimeout(180).
the way i found is to test the output of wifiManager.autoConnect and do an esp.restart to restart the connection, the modem will restart and the esp will connect to it.

is there an internal solution that would automatically restart a connection after the timeout without rebooting?

Thanks a lot

mkrasselt1, 2022-11-29:
You could use this to give the device up to 5 min to conenct to the accesspoint on boot and up to 5 reconnects to the same station :)
```
//Give more time to connect to access point
  WiFiManager.setConnectTimeout(300);
  WiFiManager.setConnectRetries(5);
```

Tomerv32, 2022-12-26:
Hey
**same question here, same issue with power outages:**

```
  wm.setConnectTimeout(30);
  wm.setConnectRetries(10);
  wm.setConfigPortalBlocking(false);
  bool res = wm.autoConnect(ap, "password");
```

I have some cases where my ESP32 moves to a different place,
and needs to be re-configured to a new WiFi network, even though it already has one saved.
(I can't add a button for reset/portal open etc..)
As I don't want to wait 300*5 seconds until the Portal opens in such a case,
I need a way to try reconnecting while the portal is open.

couldn't find such a thing in the library. any ideas? :)

stef-ladefense, 2022-12-26:
> You could use this to give the device up to 5 min to conenct to the accesspoint on boot and up to 5 reconnects to the same station :)
> 
> ```
> //Give more time to connect to access point
>   WiFiManager.setConnectTimeout(300);
>   WiFiManager.setConnectRetries(5);
> ```

ok thank
does this mean that it will wait 300s before trying to connect and this 5 times?
over a total duration of 5X300?

mkrasselt1, 2023-02-16:
> ok thank does this mean that it will wait 300s before trying to connect and this 5 times? over a total duration of 5X300?
yep thats correct - should give your router plenty time to get back up and running

mkrasselt1, 2023-02-16:
> couldn't find such a thing in the library. any ideas? :)
If there is no why to trigger such a portal opening in you case, the only option would be to keep it open all the time.

tablatronix, 2023-02-17:
You have to use the beta version