tzapu/WiFiManager #1516: OTA Error: Flash Read Failed

zdzisius, 2022-10-31:
Hello,
I tried to use this great library to upload a simple text file to flash by update firmware.
Unfortunatelly I got error message: "OTA Error: Flash Read Failed".
I have checked if I am able to write to flash in code using SPIFFS.h library.. and this way works fine..
So my question is if it is a possible to write to flash by using option "update firmware"?

Thanks in advance,
regards

tablatronix, 2022-10-31:
What esp?

zdzisius, 2022-10-31:
esp32-wroom-1

tablatronix, 2022-10-31:
hmm could be wrong flash size set , I use this and it works. I will check again

zdzisius, 2022-10-31:
I am using Partition scheme with: Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS)
Is it wrong?

tablatronix, 2022-10-31:
I think so, try the next or large app, you need equal size of your code to ota, or you need to ota a small sketch first as an intermediate. what does the compiler say about your sketch size?

zdzisius, 2022-10-31:
These are options for Partition scheme:

[image]

As you can see there is only one option with OTA.

And here are informations from compiler:

`Sketch uses 1184173 bytes (60%) of program storage space. Maximum is 1966080 bytes.
Global variables use 53808 bytes (16%) of dynamic memory, leaving 273872 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.2.1
Serial port COM3
Connecting......
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00132fff...
Compressed 18912 bytes to 13025...
Writing at 0x00001000... (100 %)
Wrote 18912 bytes (13025 compressed) at 0x00001000 in 0.5 seconds (effective 287.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 129...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (129 compressed) at 0x00008000 in 0.1 seconds (effective 299.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.2 seconds (effective 410.2 kbit/s)...
Hash of data verified.
Compressed 1189952 bytes to 765489...
Writing at 0x00010000... (2 %)
Writing at 0x00016aac... (4 %)
Writing at 0x0001bd47... (6 %)
Writing at 0x00025b57... (8 %)
Writing at 0x00030e3e... (10 %)
Writing at 0x0003b60c... (12 %)
Writing at 0x00047e32... (14 %)
Writing at 0x00053f6d... (17 %)
Writing at 0x000591b2... (19 %)
Writing at 0x0005e41d... (21 %)
Writing at 0x000631db... (23 %)
Writing at 0x00067fd5... (25 %)
Writing at 0x0006d71b... (27 %)
Writing at 0x00074b09... (29 %)
Writing at 0x0007ac1c... (31 %)
Writing at 0x00080b35... (34 %)
Writing at 0x000869c6... (36 %)
Writing at 0x0008c48c... (38 %)
Writing at 0x000922b9... (40 %)
Writing at 0x00097844... (42 %)
Writing at 0x0009cb15... (44 %)
Writing at 0x000a1e61... (46 %)
Writing at 0x000a703a... (48 %)
Writing at 0x000ac374... (51 %)
Writing at 0x000b13ae... (53 %)
Writing at 0x000b66de... (55 %)
Writing at 0x000bb806... (57 %)
Writing at 0x000c0925... (59 %)
Writing at 0x000c6443... (61 %)
Writing at 0x000cbae0... (63 %)
Writing at 0x000d1081... (65 %)
Writing at 0x000d63e3... (68 %)
Writing at 0x000db76f... (70 %)
Writing at 0x000e0b24... (72 %)
Writing at 0x000e5d46... (74 %)
Writing at 0x000eb3bd... (76 %)
Writing at 0x000f0af1... (78 %)
Writing at 0x000f67d5... (80 %)
Writing at 0x000fc2a1... (82 %)
Writing at 0x00101963... (85 %)
Writing at 0x00109eaf... (87 %)
Writing at 0x001126c3... (89 %)
Writing at 0x0011805f... (91 %)
Writing at 0x0011dde2... (93 %)
Writing at 0x001238ea... (95 %)
Writing at 0x0012902f... (97 %)
Writing at 0x0012e51c... (100 %)
Wrote 1189952 bytes (765489 compressed) at 0x00010000 in 12.2 seconds (effective 777.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

tablatronix, 2022-10-31:
You need 50% for ota, try huge app

zdzisius, 2022-10-31:
Something has changed.. instead of showing error esp32 is restarting while updating.. I don't know if it is a normal behavior.. but later on I tried to check if any new file is stored in flash with this code:

`File root = SPIFFS.open("/");
File file = root.openNextFile();

        //list of files on flash
        while (file) {
          Serial.print("FILE: ");
          Serial.println(file.name());
          file = root.openNextFile();
        }`  

but unfortunately, nothing is listed...

zdzisius, 2022-11-03:
Maybe there is a misunderstanding here..
My question was if it is possible to upload a simple file with text inside via update firmware option (e.g. test.txt) that could be read in code?

closed 2022-11-11 by zdzisius

tablatronix, 2022-11-11:
Did you figure it out, would probably make a good addition to the exmaples

zdzisius, 2023-02-03:
Hi, sorry.. I've just noticed that you asked.. and yes, I resolved the issue by changing flash mode in the ide: tools -> flash mode -> DIO