tzapu/WiFiManager #1520: Error when compiling my project with WifiManager

anhellillo, 2022-11-15:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master/2.0.14-beta

Esp8266/Esp32:  ESP32

Hardware:			ESP32-2432S028R / ILI9341

Core Version:      	2.4.0, staging

### Description

Hello, I have entered the instructions to use WifiManager in my project and when compiling only it gives me this error.
You can help me. Thanks.

### Settings in IDE

Module:  			ESP32 dev module

Additional libraries: 
#include <WiFi.h>       
#include <WebServer.h>
#include <HTTPClient.h>
#include <ArduinoJson.hpp>
#include <ArduinoJson.h>
#include <NTPClient.h>

#include "time.h"

#include <lvgl.h>
#include <TFT_eSPI.h>
#include <TFT_Touch.h>

#include <WiFiManager.h>
#include <DNSServer.h>
#include <FS.h>
#include <SD.h>
#include <SPIFFS.h>
### Sketch

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
c:\Users\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::handleRequest()':
c:\Users\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:1318:35: error: 'HTTPAuthMethod' has not been declared
     server->requestAuthentication(HTTPAuthMethod::BASIC_AUTH); // DIGEST_AUTH
```

tablatronix, 2022-11-15:
hmm can you give anymore info about your environment? what version of esp are you using?

tablatronix, 2022-11-15:
Yes is this in arduino? It should be in compile log. check that you do not have a third parry webserver library installed

anhellillo, 2022-11-15:
ok, you were right. I had changed my original Webserver library and now it compiles without problems, but when starting it gives this error:
E (21) SPIFFS: mount failed, -10025
E (22) SPIFFS: mount failed, -10025
Do you know what can happen now?

tablatronix, 2022-11-16:
Might have the flash size or partitions set wrong

closed 2022-11-28 by tablatronix

anhellillo, 2022-12-27:
Hello,
I am writing to you because I have problems with the Wifimanager library,
since I upgraded it to my application, the space occupied by my Sketch goes
from 71% to 104% and it does not compile.
I've tried going back to the previous version and the same thing still
happens.
Can you help me and give me some clue what can happen.
PDT: The only other thing I've done is update the firmware on my Arduino
AVR Boards from version 1.8.5 to 1.8.6.
Thank you.
Greetings.

tablatronix, 2022-12-27:
ESP library is also signifigantly larger now

anhellillo, 2022-12-27:
he probado a hacer un downgrade de esp library y me sigue ocurriendo lo mismo.
si, incluyo estas instrucciones el compilador me indica que se ha revasado el 100%. Si las comento, compila y me indica que ha utilzado el 71% del espacio util.
    WiFiManager WifiM;
    bool sw1 = WifiM.autoConnect( "eControl_AP", "12345678" );
    if ( !sw1 ) ESP.restart();
¿Hay alguna forma de no tener que cargar toda la libreria? Gracias.

anhellillo, 2022-12-27:
I have tried to downgrade esp library and the same thing keeps happening.
If I include these instructions, the compiler tells me that 100% has been exceeded. If I comment them, it compiles and tells me that it has used 71% of the useful space.
     WiFiManagerWifiM;
     bool sw1 = WifiM.autoConnect( "eControl_AP", "12345678" );
     if ( !sw1 ) ESP.restart();
Is there a way to avoid having to load the whole library? Thank you.

tablatronix, 2022-12-27:
Maybe you changes the partition size?

anhellillo, 2022-12-27:
Thank you, for the moment I have changed the partition and I have been able to compile but, wouldn't there be a possibility if I am only going to use the instructions that I have attached to you, download the library and make it only for those instructions? I understand that we all like our tools to be as complete and diverse as possible, but having to work with these microchips, maybe a reduced version could be useful for many people. Thank you anyway for your effort and dedication.
Greetings.