tzapu/WiFiManager #1522: I always see *WM: Using last saved values, should be faster

netdudeuk, 2022-11-20:
Hardware:			ESP8266 NodeMcu ESP-12E Development Board

Core Version:      	From - PlatformIO tzapu/WiFiManager@^0.16.0

PlatformIO won't let me select any of the much later beta versions.

My device always shows the below serial output and has never let me connect to the AP using my phone.  It always just connects to the WLAN AP that I was previously using before I added in WIFIMANAGER to my existing code.

I would have expected that the resetSetttings() ought have forced the AP pop up every time.

Module:  			ESP8266 NodeMcu ESP-12E Development Board

### Sketch

void setup()

 WiFiManager wifiManager;
 wifiManager.resetSettings();
 wifiManager.autoConnect("AutoConnectAP", "Nixie");

#END
```

### Debug Messages

*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
*WM:
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0      
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3    
*WM: IP Address:
*WM: 192.168.1.194

WiFi connected
IP address: 192.168.1.194

netdudeuk, 2022-11-26:
Managed to get it working thanks.

closed 2022-11-26 by netdudeuk

tablatronix, 2022-11-26:
Yeah you have to add a custom lib dep in pio its annoying

netdudeuk, 2022-11-26:
> Yeah you have to add a custom lib dep in pio its annoying

I didn't actually do that.  Are you referring the need to add a beta version ?  I've just got 	tzapu/WiFiManager@^0.16.0 listed in mine.  Please do you have a better lib dep entry to replace that one ?  Thanks.

tablatronix, 2022-11-26:
Yeah github master