tzapu/WiFiManager #1523: Esp8266 wroom-02 overflow

ehammock, 2022-11-22:
## PLEASE TRY Latest Master BRANCH before submitting bugs, in case they were already fixed. ##

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP8266 WROOM-02 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			"WiFi Battery" marked wemos but unlikely

Additional libraries: WiFi manager only, basic example. 

### Sketch

Bundled wifimanager basic example. 

```cpp
#BEGIN
#include <Arduino.h>

void setup() {

}

void loop() {

}
#END
```

### Debug Messages

```
messages here

The serial port overloads when the module is switched on. This works fine on a standard esp8266 but it seems to have overload issues with the wroom-02. I am flashing with a 2mb file size including a small file system. 
```

tablatronix, 2022-11-22:
what do you mean overloads?

ehammock, 2022-11-23:
Hi, many thanks for coming back to me. When I look at the serial monitor it continually scrolls with hex data. It doesn't do that with and Esp01 only the wroom-02.

ehammock, 2022-11-23:
Hi, the builtin BASIC example in arduino under library wifimanager.

tablatronix, 2022-11-24:
Ill check but thats odd sounds like its crashing or a boot loop?

ehammock, 2022-11-24:
Yes, it seems to be on boot.

I'll take another look this evening. My gut feel is that it is related to
the flash size which on the wroom-02 I think is smaller than the standard
esp8266. I would like to get it working though, as WiFi manager perfectly
suits my need, also the wroom-02 'WiFi battery' is a perfect physical
solution for my need.

Thanks.

tablatronix, 2022-11-24:
What is a wroom 02?

tablatronix, 2022-11-24:
oh yeah these are pretty old, I think i have one

ehammock, 2022-11-25:
Many thanks.

I'm happy to contribute to your project (buy a coffee, patrion, fiverr etc)
if we can get this working.

Thanks, Edward.

tablatronix, 2022-11-28:
Can you maybe add a delay on boot so you can see if there is a crashdump or something

Andysoft2000, 2022-12-05:
Basic Infos
Hardware:
ESP8285 ((Lolin)D1 Mini Lite)

WiFimanager Branch/Release: Master

Esp8266/Esp32: ESP8285 / D1 Mini Lite

Hardware: ESP-12e, esp01, esp25

Core Version: 3.0.2
system_get_sdk_version():  2.2.2-dev(38a443e)
system_get_boot_version(): 31
getFreeHeap():             47344

Problem description
Settings in IDE

Additional libraries: WiFi manager only, OnDemandConfigPortai example..
Sketch
Bundled wifimanager example.

Decoding stack results
0x401006cc: pvPortZalloc(size_t, char const*, int) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\umm_malloc/umm_heap_select.h line 85
0x4020ab44: ESP8266WiFiGenericClass::mode(WiFiMode) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 427
0x4020abdd: ESP8266WiFiGenericClass::enableSTA(bool) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.cpp line 479
0x4020af53: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 368
0x4020af5c: ESP8266WiFiSTAClass::disconnect(bool) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi\src\ESP8266WiFiSTA.cpp line 372
0x40216ae8: __esp_yield() at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 116
0x40217205: __delay(unsigned long) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 55
0x402171fa: __delay(unsigned long) at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_wiring.cpp line 54
0x40204c53: WiFiManager::resetSettings() at C:\Users\andys\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 2602
0x402014af: setup() at C:\Users\andys\AppData\Local\Temp\arduino_modified_sketch_86454/OnDemandConfigPortal.ino line 121
0x40216c00: loop_wrapper() at C:\Users\andys\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266\core_esp8266_main.cpp line 198

If I use a D1 Mini (ESP8266 inside) everything works fine.  
I hope this helps a little..