tzapu/WiFiManager #1525: ESP8266 Issue with using WM and painlessMesh together

wok1909, 2022-11-27:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: Esp8266

Hardware: NodeMCU

Core Version:  Don't know

### Description

Problem description

I am currently trying to use WM and painlessMesh together.
The reason is, I want to make one NodeMCU to connect to both WiFi and Mesh and the rest of the NodeMCUs to be connected only in Mesh. The NodeMCU that is connected to both WiFi and Mesh will collect all the message from other NodeMCUs and send MQTT message through WiFi.

However, I am currently having a big trouble doing this. I have seen the previous issues with WM and painlessMesh using together and I also have checked that example of using both could be considered (https://github.com/tzapu/WiFiManager/issues/850#issuecomment-638220325). What I have tried didn't work and want to ask some help for solving this.

This is my code.

```
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include "WiFiManager.h"          //https://github.com/tzapu/WiFiManager
#include "EspMQTTClient.h"
#include "painlessMesh.h"

#define   MESH_PREFIX     "my_mesh"
#define   MESH_PASSWORD   "my_passwd"
#define   MESH_PORT       5555

// MQTT Client setting params
#define mqtt_broker "my_ip"
#define mqtt_id "my_id"
#define mqtt_pswd "my_passwd"
#define mqtt_user "mesh_test"

// MQTT Topic
#define TOPIC "my_mesh/iphone"

// MESH
Scheduler userScheduler; // to control your personal task
painlessMesh  mesh;

// MQTT Client params
EspMQTTClient *client = NULL;
String ssid;
String pswd;

// Needed for painless library
void receivedCallback( uint32_t from, String &msg ) {
  Serial.printf("Received from %u msg=%s\n", from, msg.c_str());
}

void newConnectionCallback(uint32_t nodeId) {
    Serial.printf("--> startHere: New Connection, nodeId = %u\n", nodeId);
}

void changedConnectionCallback() {
  Serial.printf("Changed connections\n");
}

void nodeTimeAdjustedCallback(int32_t offset) {
    Serial.printf("Adjusted time %u. Offset = %d\n", mesh.getNodeTime(),offset);
}

// Setting Up MQTT Client
void setup_mqttclient() {
  if(client != NULL) {
    delete client;
    client = NULL;
  }
    
  client = new EspMQTTClient(
          ssid.c_str(),
          pswd.c_str(),
          mqtt_broker,  // MQTT Broker server ip
          mqtt_id,   // Can be omitted if not needed
          mqtt_pswd,   // Can be omitted if not needed
          mqtt_user,     // Client name that uniquely identify your device
          1883              // The MQTT port, default to 1883. this line can be omitted
        );
}

// Wi-Fi Dynamically Connecting 
void wifiConnect(bool isReset) {
  WiFi.begin(ssid.c_str(), pswd.c_str(), 0, 0, true);
  WiFiManager wifiManager;

  if(isReset)
    wifiManager.resetSettings();

  bool res;
  res = wifiManager.autoConnect("esp8266 AutoConnectAP");  
 
  if(!res) {
      Serial.println("Failed to connect");
      // ESP.restart();
  } 
  else {
    // Wi-Fi Connected
    Serial.println(F("WIFIManager connected!"));
    
    ssid = wifiManager.getWiFiSSID();
    pswd = wifiManager.getWiFiPass();

    Serial.println(ssid);
    Serial.println(pswd); 

    setup_mqttclient();
  }
}

void onConnectionEstablished() {
  client->subscribe("iot/21600437", [](const String & payload) {
     Serial.println(payload);
  });

  client->publish("iot/21600437", "Connection with ESP Success!");
  Serial.println("\nConnection with ESP Success!");
}

void setup()
{
  Serial.begin(115200);
  
  

  mesh.setDebugMsgTypes( ERROR | STARTUP );  // set before init() so that you can see startup messages
  mesh.init( MESH_PREFIX, MESH_PASSWORD, &userScheduler, MESH_PORT );

  mesh.onReceive(&receivedCallback);
  mesh.onNewConnection(&newConnectionCallback);
  mesh.onChangedConnections(&changedConnectionCallback);
  mesh.onNodeTimeAdjusted(&nodeTimeAdjustedCallback);

  
  wifiConnect(true);
}

void loop()
{
    
}
```

This is a very simple test I wrote for checking if both WM and painlessMesh works. 

The result I get after connecting to a WiFi using WifiManager is like this.
Here is my Serial Monitor result.

```
03:52:10.762 -> *wm:[2] NUM CLIENTS:  0
03:53:01.652 -> {l l⸮⸮| ⸮d⸮|⸮d⸮c<ǃ⸮⸮{⸮c⸮b⸮⸮og⸮l'o⸮⸮⸮#p⸮⸮${l{$p⸮'⸮*wm:[1] resetSettings 
03:53:03.096 -> *wm:[1] SETTINGS ERASED 
03:53:03.096 -> *wm:[1] AutoConnect 
03:53:03.096 -> *wm:[2] Connecting as wifi client... 
03:53:03.096 -> *wm:[2] setSTAConfig static ip not set, skipping 
03:53:03.096 -> *wm:[1] No wifi saved, skipping 
03:53:03.096 -> *wm:[2] Connection result: WL_NO_SSID_AVAIL
03:53:03.134 -> *wm:[1] AutoConnect: FAILED 
03:53:03.134 -> *wm:[2] Starting Config Portal 
03:53:03.134 -> *wm:[2] Disabling STA 
03:53:03.134 -> *wm:[2] Enabling AP 
03:53:03.134 -> *wm:[1] StartAP with SSID:  esp8266 AutoConnectAP
03:53:03.605 -> *wm:[2] AP has anonymous access! 
03:53:04.114 -> *wm:[1] AP IP address: 10.21.199.1
03:53:04.114 -> *wm:[1] Starting Web Portal 
03:53:04.114 -> *wm:[2] HTTP server started 
03:53:04.149 -> *wm:[2] WiFi Scan ASYNC started 
03:53:04.149 -> *wm:[2] Config Portal Running, blocking, waiting for clients... 
03:53:06.298 -> *wm:[2] WiFi Scan ASYNC completed in 2187 ms
03:53:06.331 -> *wm:[2] WiFi Scan ASYNC found: 25
03:53:24.910 -> *wm:[2] <- Request redirected to captive portal 
03:53:25.022 -> *wm:[2] <- HTTP Root 
03:53:25.385 -> *wm:[2] Scan is cached 19059 ms ago
03:53:25.745 -> *wm:[2] <- Request redirected to captive portal 
03:53:25.783 -> *wm:[2] <- HTTP Root 
03:53:25.817 -> *wm:[2] Scan is cached 19497 ms ago
03:53:26.168 -> *wm:[2] <- Request redirected to captive portal 
03:53:26.204 -> *wm:[2] <- HTTP Root 
03:53:26.242 -> *wm:[2] Scan is cached 19899 ms ago
03:53:27.295 -> *wm:[2] <- HTTP Wifi 
03:53:27.295 -> *wm:[2] Scan is cached 20971 ms ago
03:53:27.295 -> *wm:[1] 25 networks found
03:53:27.329 -> *wm:[2] AP:  -56 HAIL221-2G
03:53:27.329 -> *wm:[2] AP:  -64 DIRECT-riC48x Series
03:53:27.329 -> *wm:[2] AP:  -67 Hailers221
03:53:27.329 -> *wm:[2] AP:  -81 Policylaw_center
03:53:27.329 -> *wm:[2] AP:  -82 iptime
03:53:27.329 -> *wm:[2] AP:  -83 DIRECT-x0C460 Series
03:53:27.367 -> *wm:[2] AP:  -86 fuzzer-wifi
03:53:27.367 -> *wm:[2] AP:  -86 kkim2.4G
03:53:27.367 -> *wm:[2] AP:  -91 RepetierServer
03:53:27.477 -> *wm:[2] <- Request redirected to captive portal 
03:53:27.477 -> *wm:[2] <- HTTP Root 
03:53:27.534 -> *wm:[2] Scan is cached 21188 ms ago
03:53:31.516 -> *wm:[2] NUM CLIENTS:  1
03:53:36.225 -> *wm:[2] <- HTTP WiFi save  
03:53:36.474 -> *wm:[2] processing save 
03:53:38.253 -> *wm:[2] Connecting as wifi client... 
03:53:38.253 -> *wm:[2] setSTAConfig static ip not set, skipping 
03:53:38.253 -> *wm:[1] Connecting to NEW AP: DL-Lab 2G
03:53:38.360 -> *wm:[1] connectTimeout not set, ESP waitForConnectResult... 
03:53:42.029 -> *wm:[2] Connection result: WL_CONNECTED
03:53:42.029 -> *wm:[1] Connect to new AP [SUCCESS] 
03:53:42.029 -> *wm:[1] Got IP Address: 
03:53:42.029 -> *wm:[1] 192.168.0.2 
03:53:42.029 -> *wm:[2] shutdownConfigPortal 
03:53:42.029 -> *wm:[2] restoring usermode STA
03:53:43.030 -> *wm:[2] WiFi Reconnect, was idle 
03:53:43.030 -> *wm:[2] wifi status: WL_IDLE_STATUS
03:53:43.030 -> *wm:[2] wifi mode: STA
03:53:43.030 -> *wm:[2] configportal closed 
03:53:43.030 -> *wm:[1] config portal exiting 
03:53:43.030 -> WIFIManager connected!
03:53:43.030 -> SSID
03:53:43.030 -> PASSWD
```

What I expect to see is after connecting to the WiFi, it should reach to function onConnectionEstablished() and print out "Connection with MQTT Broker Success!" on my Serial monitor but nothing comes out even after minutes passed. Usually it should come up right away when painlessMesh is not implemented.

Is there anyway that I can use WM and painlessMesh together?

wok1909, 2022-11-28:
One solution that I have thought of is using WM only for getting the WiFi SSID and Password from user. 
If WM can open HTML page to show user which WiFi is available to connect and read WiFi SSID and Password from user,  I think I can just pass this info to MQTTClient and connect WiFi right through without using WM.

Will this be possible for only using WM to get WiFi SSID and Password?

tablatronix, 2022-11-28:
I would assume since you are managing wifi client yourself you should NOT be using a constructor that also does wifi negotiation.

closed 2022-11-28 by tablatronix