tzapu/WiFiManager #1527: Persistent custom values

pablopeu, 2022-11-28:
Is there a way to make the defined custom values to be persistant?

`  WiFiManagerParameter custom_User_ID("User_ID", "Used ID", "", 30);
  WiFiManagerParameter custom_Bot_Token("Bot_Token", "BOT token", "", 46);
  wm.addParameter(&custom_User_ID);
  wm.addParameter(&custom_Bot_Token);`

This is my example of the values I want to keep across reboots/resets, does the library allows for this, or I need to write/retrieve them by myself from say EEPROM?

Thanks

tablatronix, 2022-11-28:
you need to save them to fs yourself

closed 2022-11-28 by tablatronix

pablopeu, 2022-11-29:
solved it by using the Effortless SPIFFS library (https://github.com/thebigpotatoe/Effortless-SPIFFS)

` 
String usid = custom_User_ID.getValue();
String toke = custom_Bot_Token.getValue();

  if (usid !=""){
    fileSystem.saveToFile("/usid.txt", usid);
  }
  if (usid ==""){
    fileSystem.openFromFile("/usid.txt", usid);
    userid = usid.toInt();  //userid is global
  }

   if (toke !=""){
    fileSystem.saveToFile("/toke.txt", toke);
  }
  if (toke ==""){
    fileSystem.openFromFile("/toke.txt", toke);
    toke.toCharArray(buf, 80); // char token[80] is global
  }`

tablatronix, 2022-11-29:
Consider using littlefs, spiffs has been deprecated