tzapu/WiFiManager #1528: A bug in the init() method of the WiFiManagerParameter class

Mollayo, 2022-12-01:
I believe I found a bug in the init() method of the WiFiManagerParameter class.

In this init() method, the value of _length is set to 1 by default. See here:
https://github.com/tzapu/WiFiManager/blob/9f5981e2fd40db86780299d5ba3a3840b891a998/WiFiManager.cpp#L61
The _value variable is also initialised to nullptr.

This init() method is then using the setValue() method to allocate the pointer for _value. However, the allocation is not made in case the length parameter in the init() function is set to 1 because of the condition "if(_length != length)" not being satisfied. See here: 
https://github.com/tzapu/WiFiManager/blob/9f5981e2fd40db86780299d5ba3a3840b891a998/WiFiManager.cpp#L91

In consequence, the _value variable is still equal to nullptr after calling the init() method, making WifiManager to crash.

One solution would be to write " _value  = new char[_length + 1]; " instead of "_value          = nullptr;" at this line:
https://github.com/tzapu/WiFiManager/blob/fe9774fe0f231767f3fc59de1a03a9c44f06adc3/WiFiManager.cpp#L62

tablatronix, 2022-12-01:
Yeah there are 2 issues on this, I have not had a chance to look at it

#1504

tablatronix, 2022-12-01:
or 
`  if(_length != length || _value == nullptr){` perhaps?

tablatronix, 2022-12-01:
I forget why I used length =1 
I think 0 made it not possible to detect initialized params or out of scope objs

commit 118ca1ec (2022-12-08): fixes #1528
closed 2022-12-08 by tablatronix