tzapu/WiFiManager #1530: Incompatibility with ESP32 ESPAsyncWebServer

joba-1, 2022-12-13:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP32 Wroom32

Core Version:      	tested latest stable and staging

### Description

When migrating firmware from using standard WebServer to ESPAsyncWebserver there are conflicts and compile fails

Would be nice if I could still use WiFiManager to provision WLAN and not cook my own!

### Settings in IDE

Module:  			mhetesp32minikit

[env:mhetesp32minikit]
platform = espressif32
; platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
board = mhetesp32devkit
framework = arduino

Additional libraries: 

lib_deps = 
    https://github.com/tzapu/WiFiManager.git
    ESP Async WebServer

### Sketch

```cpp
#BEGIN
#include <Arduino.h>

#include <ESPAsyncWebServer.h>
#include <WiFiManager.h>

AsyncWebServer web_server(80);

void setup() {
    WiFi.mode(WIFI_STA);
    WiFi.hostname("AsyncWiFiManager");

    WiFiManager wm;
    if (!wm.autoConnect(WiFi.getHostname(), WiFi.getHostname())) {
        ESP.restart();
        while (true)
            ;
    }

    web_server.on("/", []( AsyncWebServerRequest *request ) { 
        request->send(200, "text/html", "<html><head><title>Nooo!<title></head><body>WTF :(</body><html>");
    });
    web_server.begin();
}

void loop() {}
#END
```

### Debug Messages

```
In file included from /home/joachim/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from /home/joachim/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
[... 93 lines cut ...]
/home/joachim/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from src/main.cpp:3:
.pio/libdeps/mhetesp32devkit/ESP Async WebServer/src/ESPAsyncWebServer.h:62:3: note: previous declaration 'WebRequestMethod HTTP_OPTIONS'
   HTTP_OPTIONS = 0b01000000,
   ^~~~~~~~~~~~
In file included from /home/joachim/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/HTTP_Method.h:4,
                 from /home/joachim/.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/WebServer.h:30,
                 from .pio/libdeps/mhetesp32devkit/WiFiManager/WiFiManager.h:94,
                 from src/main.cpp:4:
/home/joachim/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:128:6: error: 'HTTP_PATCH' conflicts with a previous declaration
   XX(28, PATCH,       PATCH)        \
      ^~
/home/joachim/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:138:45: note: in definition of macro 'XX'
 #define XX(num, name, string) HTTP_##name = num,
                                             ^~~
/home/joachim/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/nghttp/port/include/http_parser.h:139:3: note: in expansion of macro 'HTTP_METHOD_MAP'
   HTTP_METHOD_MAP(XX)
   ^~~~~~~~~~~~~~~
In file included from src/main.cpp:3:
.pio/libdeps/mhetesp32devkit/ESP Async WebServer/src/ESPAsyncWebServer.h:60:3: note: previous declaration 'WebRequestMethod HTTP_PATCH'
   HTTP_PATCH   = 0b00010000,
   ^~~~~~~~~~
```

joba-1, 2022-12-13:
I worked a bit on this myself and found the errors are "only" compile time errors.
If I split code of WiFiManager and ESPAsyncWebserver in two compilation units, so that #include <ESPAsyncWebServer.h> and #include <WiFiManager.h> are not in the same file, then it basically works.

Only gripe left: if the WiFiManager actually opens the portal webserver, it seems the listen port is not released:

```
*wm:config portal exiting 
Webserver
[246649][E][AsyncTCP.cpp:1268] begin(): bind error: -8
```

After a reboot (without wifi setup) the async webserver binds and works fine

Maybe you could consider using this before bind (not tested):

```
int flag = 1;
setsockopt(listen_sock, SOL_SOCKET, SO_REUSEADDR, &flag, sizeof(flag));
```

joba-1, 2022-12-13:
Found yet another "solution":

include WiFiManager first. ESPAsyncWebServer is already prepared to not redefine the conflicting symbols if the standard webserver is already included.

the bind error remains though...

joba-1, 2022-12-14:
see https://github.com/tzapu/WiFiManager/tree/feature_asyncwebserver
I assume it is compatible ;)

closed 2022-12-14 by joba-1

tablatronix, 2022-12-14:
Yeah port issues are a problem no idea what the cause is. Never figured it out

joba-1, 2022-12-15:
Well, I still would like to get rid of it.
Hope you don't mind if I note here what I found so far. 
Feel free to comment, even if it is just "been there, done that" again :)

What I found out so far:
* Port in use is caused mostly by connections in TIME_WAIT (seen them)
* TIME_WAIT on the server side is caused by server trying to close a connection without recv'ing until len==0
* HTTP servers and clients usually do this right, but HTTP defaults to let connections open after requests for efficiency
* Connections not kept open, if client or server request close explicitly in the HTML header

Sooo, 
* A rather simple check would be to add this close request in your HTTP_HEADER to always use it
    *  if there are still TIME_WAIT connections, the servers (async and simple) have a bug (unlikely)
* Later optimize to use close header only on requests that may lead to a web server shutdown

will try that later...

tablatronix, 2022-12-15:
It shouldn't have anything to do with it, the webserver obj is closed and deleted, the ESP lib is not allowing the port to rebind for some reason.

tablatronix, 2022-12-16:
As soon as I get 2.0 released I will work on reworking the async branch so it can be optional in main.

joba-1, 2022-12-18:
I gave up on ESPAsyncWebServer restarts.
I reduced my test firmware to just serve a /restart page to initiate a delayed webserver restart (and added prints to all relevant constructors and destructors). 
Still the port sometimes cannot be reallocated although all constructor/destructor pairs are visible before the actual restart. And sometimes it just works.
I'll just use ESP restarts if your WiFiManager portal was active. Fine for my usecases.

tablatronix, 2022-12-18:
Yeah its really an esp bug

joba-1, 2022-12-24:
couldn't let it rest :D

so far I never got "port in use" when using this code (called in loop()):

```c++
// Need to wait for restart request to finish before unbinding listening port, else browser receives error
// Need to wait for open requests to finish before restarting webserver, else port is not free for reuse
void handle_restart() {
    const unsigned long delay_unbind = 1000;  // let restart request finish (response, redirect, ...) before unbinding listener
    const unsigned long delay_restart = 1000;   // let open requests finish after unbinding listener port
    static unsigned long unbind_ms = 0;
    static unsigned long restart_ms = 0;
    unsigned long now = millis();

    if (restart_ms) {
        if (now - restart_ms > delay_restart) {
            printf("%lu: Webserver setup\n", now);
            restart_ms = 0;
            setup_webserver();
        }
    }
    else if (unbind_ms) {
        if (now - unbind_ms > delay_unbind) {
            printf("%lu: Webserver stops listening\n", now);
            web_server->end(); // no new requests
            unbind_ms = 0;
            restart_ms = now;
            if (!restart_ms) 
                restart_ms--;
        }
    }
    else if (shouldRestart) {
        printf("%lu: Webserver restart pending\n", now);
        shouldRestart = false;
        unbind_ms = now;
        if (!unbind_ms) 
            unbind_ms--;
    }
}

void setup_webserver() {
    if (web_server) {
        printf("%lu: Stopping Webserver\n", millis());
        delete(web_server);
    }

    printf("%lu: Starting Webserver\n", millis());
    web_server = new AsyncWebServer(WEBSERVER_PORT);

    web_server->on(...
```

don`t know if it matters, but I use ottowinter/ESPAsyncWebServer-esphome

ear9mrn, 2024-01-14:
I just came across the same issue. It sort of works if you call/run wifimanager first then espasyncwebserver second although I have found this still has issues. The way that the http call methods are defined as noted in above errors are different in their respective header files.

webserver.h
`
/* Request Methods */
#define HTTP_METHOD_MAP(XX)         \
  XX(0,  DELETE,      DELETE)       \
  XX(1,  GET,         GET)          \
  XX(2,  HEAD,        HEAD)         \
  XX(3,  POST,        POST)         \
  XX(4,  PUT,         PUT)          \
`

ESPAsynWebServer.h

`
typedef enum {
  HTTP_GET     = 0b00000001,
  HTTP_POST    = 0b00000010,
  HTTP_DELETE  = 0b00000100,
  HTTP_PUT     = 0b00001000,
`

GET is fine as both = 1, but any other method and the values do not correspond. As such, I was getting onNotFound for anything other than GET when using wifimanager followed by ESPAsyncWebServer. My solution was to manually enter the values that correspond to the ESPAsyncWebServer methods rather than use the defined values, like HTTP_POST. For example:

`
if (request->method() == 64) {} //for HTTP_OPTIONS
`
due to HTTP_OPTIONS = 64 as defined by ESPAsyncWebServer 
`
 HTTP_OPTIONS = 0b01000000,
`
and it is 6 with Webserver.h:

`
(XX(6,  OPTIONS,     OPTIONS)      \):
` 
Any suggested remedy for this dilemma other than manually defining my own method values?

Thanks,

Pete.

ear9mrn, 2024-01-14:
Just to be clear as to my solution:

`
//redefine HTTP methods as per ESPAsyncWebserver.h
 typedef enum {
        MY_HTTP_GET          = 0b00000001,
        MY_HTTP_POST        = 0b00000010,
        MY_HTTP_DELETE     = 0b00000100,
        MY_HTTP_PUT          = 0b00001000,
        MY_HTTP_PATCH      = 0b00010000,
        MY_HTTP_HEAD       = 0b00100000,
        MY_HTTP_OPTIONS = 0b01000000,
        MY_HTTP_ANY         = 0b01111111,
        } my_http_methods;
`

a POST response would look like this:

`
server.on("/getinfo", MY_HTTP_POST, [] (AsyncWebServerRequest *request){
        request->send(200, "text/plain", "OK");
    });
`
and onNotFound:
`
//needed for CORS
if (request->method() == MY_HTTP_OPTIONS ) {
          request->send(200, "plain/text", "OK");
    } else {
          request->send(404, "plain/text", "not found");
    }
`
Hope this helps others.

qinst64, 2024-01-21:
I also got the same errors, not so complete for above answers, there is my summary 
### Reason
1. `ESPAsyncWebServer` conflicts with `ESP8266WebServer` (required by `WiFiManager`) on `HTTP_XXX`
2.  `ESPAsyncWebServer` handles by checks `#ifndef WEBSERVER_H`, however  `ESP8266WebServer` does `#define ESP8266WEBSERVER_H`. So simply changing the importing order will not work.

### Method 1
Just write like this
```C++
#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#define WEBSERVER_H "fix confict"
#include <WiFiManager.h>
#include <ESPAsyncWebServer.h>
```
note: need to include `WiFiManager.h` first, and define `WEBSERVER_H` as anthing before `ESPAsyncWebServer`
### Method 2
Go to `arduino/libraries/ESPAsyncWebServer`, replace all `HTTP_` with `MY_HTTP_`. There are more than 50+ occurrences. This can be easily done by IDE like VSCode. 
Then no conficts any more, however in your code should also use `MY_HTTP_XXX` instead of `HTTP_XXX`
### Conclusion
Method 1 is suggested

tablatronix, 2024-01-21:
You are trying to run these togather?

Have you tried the async branch ?

qinst64, 2024-01-22:
> You are trying to run these togather?
> 
> Have you tried the async branch ?

nope. `ESPAsync_WiFiManager` is archived,  `ESPAsyncWiFiManager` is not actively maintained(2 yr ago).

tablatronix, 2024-01-22:
Those are forks

grantnorwood, 2024-05-31:
Another solution folks may encounter is when using ESPAsyncWebServer and WiFiManager in separate class files, and those class files aren't ordered correctly.  Especially if your IDE likes to re-order includes alphabetically on its own 🤦 

In this case, just add `#include <WiFiManager.h>` to your `main.cpp` and ensure it's ordered before your class files.  This works great for me using VS Code and PlatformIO!

gpena208777, 2024-08-31:
> Found yet another "solution":
> 
> include WiFiManager first. ESPAsyncWebServer is already prepared to not redefine the conflicting symbols if the standard webserver is already included.
> 
> the bind error remains though...

This worked!

Thanks so much