tzapu/WiFiManager #1535: Many times the list of available wifi is not displayed in "Configure WiFi" menu.

DonatelloX, 2022-12-21:
#### Hardware
ESP32 Dev Module

#### Software
Arduino IDE 2.0.3
Arduino core ESP32 2.0.5

### Description

With the latest versions of the WiFiManager library, it is only sometimes possible to view the list of available WiFis in the "Configure WiFi" menu. Many times the list of available wifi is not displayed. Tried on multiple devices. 
It is not a signal strength problem because trying several times then all the nearby wifi are displayed.

tablatronix, 2022-12-21:
hmm, let me see what might cause that. I know there was some esp bugs in 2.0.4 but nobody else brought this up.

Can you possibly obtain serial logs and bump up debugging?

DonatelloX, 2022-12-21:
Sure, thanks for reply!

This is what happens by pressing the "Configure WiFi" button. SSID names of WiFi networks are not displayed:
```
15:22:55.816 -> *wm:Starting Web Portal 
15:23:25.027 -> *wm:[ERROR] scan waiting 
15:23:26.309 -> *wm:. 
15:23:26.434 -> *wm:. 
15:23:26.527 -> *wm:. 
15:23:26.621 -> *wm:. 
15:23:26.714 -> *wm:. 
15:23:26.809 -> *wm:. 
15:23:26.935 -> *wm:. 
15:23:27.028 -> *wm:. 
15:23:27.121 -> *wm:. 
15:23:27.214 -> *wm:. 
15:23:27.338 -> *wm:. 
15:23:27.430 -> *wm:. 
15:23:27.523 -> *wm:. 
15:23:27.601 -> *wm:. 
15:23:27.740 -> *wm:. 
15:23:27.834 -> *wm:. 
15:23:27.913 -> *wm:-2 networks found
```

With debug information below:
```
15:33:45.738 -> *wm:Starting Web Portal 
15:33:45.738 -> [  7842][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
15:33:47.440 -> [  9585][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
15:33:55.352 -> [ 17469][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
15:33:55.525 -> [ 17657][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
15:33:55.802 -> [ 17932][W][WebServer.cpp:436] send(): content length is zero
15:33:58.261 -> [ 20410][W][WebServer.cpp:436] send(): content length is zero
15:33:58.432 -> [ 20557][E][WebServer.cpp:649] _handleRequest(): request handler not found
15:33:59.394 -> [ 21536][W][WebServer.cpp:436] send(): content length is zero
15:33:59.439 -> *wm:[ERROR] scan waiting 
15:34:03.113 -> *wm:. 
15:34:03.205 -> *wm:. 
15:34:03.299 -> *wm:. 
15:34:03.437 -> *wm:. 
15:34:03.530 -> *wm:. 
15:34:03.624 -> *wm:. 
15:34:03.717 -> *wm:. 
15:34:03.825 -> *wm:. 
15:34:03.918 -> *wm:. 
15:34:04.010 -> *wm:. 
15:34:04.102 -> *wm:. 
15:34:04.228 -> *wm:. 
15:34:04.321 -> *wm:-2 networks found
15:34:04.321 -> [ 26454][W][WebServer.cpp:436] send(): content length is zero
15:34:04.570 -> [ 26672][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
15:34:05.636 -> [ 27754][W][WebServer.cpp:436] send(): content length is zero
15:34:05.713 -> [ 27836][W][WebServer.cpp:436] send(): content length is zero
```

After several tries, it works (the WiFi name is displayed):
```
15:32:17.559 -> [ 17781][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
15:32:18.241 -> [ 18479][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
15:32:18.518 -> [ 18725][W][WebServer.cpp:436] send(): content length is zero
15:32:20.926 -> [ 21147][W][WebServer.cpp:436] send(): content length is zero
15:32:21.182 -> [ 21436][E][WebServer.cpp:649] _handleRequest(): request handler not found
15:32:22.197 -> [ 22424][W][WebServer.cpp:436] send(): content length is zero
15:32:22.197 -> [ 22431][W][WebServer.cpp:436] send(): content length is zero
15:32:27.223 -> [ 27429][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
15:32:37.641 -> [ 37852][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
15:32:37.641 -> *wm:1 networks found
15:32:38.800 -> [ 39035][W][WebServer.cpp:436] send(): content length is zero
15:32:39.003 -> [ 39233][W][WebServer.cpp:436] send(): content length is zero
```

DonatelloX, 2022-12-21:
With verbose message:
When it doesn't work:
```
16:02:56.238 -> [ 31635][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
16:02:56.238 -> [ 31636][V][Parsing.cpp:122] _parseRequest(): method: GET url: /wifi search: 
16:02:56.238 -> [ 31639][V][Parsing.cpp:226] _parseRequest(): headerName: Host
16:02:56.238 -> [ 31644][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
16:02:56.238 -> [ 31651][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
16:02:56.238 -> [ 31657][V][Parsing.cpp:227] _parseRequest(): headerValue: keep-alive
16:02:56.238 -> [ 31663][V][Parsing.cpp:226] _parseRequest(): headerName: x-wap-profile
16:02:56.238 -> [ 31669][V][Parsing.cpp:227] _parseRequest(): headerValue: http://218.249.47.94/Xianghe/MTK_Phone_KK_UAprofile.xml
16:02:56.238 -> [ 31680][V][Parsing.cpp:226] _parseRequest(): headerName: Upgrade-Insecure-Requests
16:02:56.238 -> [ 31687][V][Parsing.cpp:227] _parseRequest(): headerValue: 1
16:02:56.238 -> [ 31693][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
16:02:56.238 -> [ 31698][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (Linux; Android 7.0; thor Build/NRD90M; wv) AppleWebKit/537.36 (KHTML, like Gecko) Version/4.0 Chrome/51.0.2704.91 Mobile Safari/537.36
16:02:56.238 -> [ 31717][V][Parsing.cpp:226] _parseRequest(): headerName: Accept
16:02:56.238 -> [ 31722][V][Parsing.cpp:227] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/webp,*/*;q=0.8
16:02:56.238 -> [ 31734][V][Parsing.cpp:226] _parseRequest(): headerName: Referer
16:02:56.238 -> [ 31739][V][Parsing.cpp:227] _parseRequest(): headerValue: http://192.168.4.1/
16:02:56.238 -> [ 31747][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
16:02:56.238 -> [ 31753][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip, deflate
16:02:56.238 -> [ 31760][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Language
16:02:56.238 -> [ 31766][V][Parsing.cpp:227] _parseRequest(): headerValue: it-IT,en-US;q=0.8
16:02:56.238 -> [ 31773][V][Parsing.cpp:226] _parseRequest(): headerName: X-Requested-With
16:02:56.238 -> [ 31779][V][Parsing.cpp:227] _parseRequest(): headerValue: com.android.captiveportallogin
16:02:56.238 -> [ 31787][V][Parsing.cpp:254] _parseArguments(): args: 
16:02:56.238 -> [ 31792][V][Parsing.cpp:237] _parseRequest(): Request: /wifi
16:02:56.238 -> [ 31797][V][Parsing.cpp:238] _parseRequest():  Arguments: 
16:02:56.238 -> *wm:[ERROR] scan waiting 
16:02:56.238 -> *wm:. 
16:02:56.347 -> *wm:. 
16:02:56.440 -> *wm:. 
16:02:56.533 -> *wm:. 
16:02:56.643 -> *wm:. 
16:02:56.736 -> *wm:. 
16:02:56.829 -> *wm:. 
16:02:56.921 -> *wm:. 
16:02:57.048 -> *wm:. 
16:02:57.141 -> *wm:. 
16:02:57.233 -> *wm:. 
16:02:57.327 -> *wm:. 
16:02:57.549 -> *wm:-2 networks found
16:02:57.549 -> [ 33021][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
16:02:57.549 -> [ 33022][V][Parsing.cpp:122] _parseRequest(): method: GET url: /generate_204 search: 
16:02:57.549 -> [ 33026][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
16:02:57.549 -> [ 33031][V][Parsing.cpp:227] _parseRequest(): headerValue: Dalvik/2.1.0 (Linux; U; Android 7.0; thor Build/NRD90M)
16:02:57.549 -> [ 33042][V][Parsing.cpp:226] _parseRequest(): headerName: Host
16:02:57.549 -> [ 33047][V][Parsing.cpp:227] _parseRequest(): headerValue: captive.apple.com
16:02:57.549 -> [ 33054][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
16:02:57.549 -> [ 33060][V][Parsing.cpp:227] _parseRequest(): headerValue: Keep-Alive
16:02:57.549 -> [ 33066][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
16:02:57.549 -> [ 33072][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip
16:02:57.549 -> [ 33078][V][Parsing.cpp:254] _parseArguments(): args: 
16:02:57.549 -> [ 33083][V][Parsing.cpp:237] _parseRequest(): Request: /generate_204
16:02:57.549 -> [ 33089][V][Parsing.cpp:238] _parseRequest():  Arguments: 
16:02:57.549 -> [ 33094][E][WebServer.cpp:649] _handleRequest(): request handler not found
16:02:57.549 -> [ 33101][W][WebServer.cpp:436] send(): content length is zero
16:02:57.549 -> [ 33109][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
16:02:57.595 -> [ 33183][V][WiFiGeneric.cpp:381] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 1
16:02:57.627 -> [ 33184][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
16:03:02.526 -> [ 38116][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
16:03:02.650 -> [ 38117][V][Parsing.cpp:122] _parseRequest(): method: GET url: /generate_204 search: 
16:03:02.650 -> [ 38121][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
16:03:02.650 -> [ 38126][V][Parsing.cpp:227] _parseRequest(): headerValue: Dalvik/2.1.0 (Linux; U; Android 7.0; thor Build/NRD90M)
16:03:02.650 -> [ 38137][V][Parsing.cpp:226] _parseRequest(): headerName: Host
16:03:02.650 -> [ 38142][V][Parsing.cpp:227] _parseRequest(): headerValue: captive.apple.com
16:03:02.650 -> [ 38149][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
16:03:02.650 -> [ 38155][V][Parsing.cpp:227] _parseRequest(): headerValue: Keep-Alive
16:03:02.650 -> [ 38161][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
16:03:02.650 -> [ 38168][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip
[... 46 lines cut ...]
[... cut ...]
15:55:10.145 -> [198278][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
15:55:10.145 -> [198285][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip
15:55:10.145 -> [198290][V][Parsing.cpp:254] _parseArguments(): args: 
15:55:10.145 -> [198295][V][Parsing.cpp:237] _parseRequest(): Request: /generate_204
15:55:10.145 -> [198301][V][Parsing.cpp:238] _parseRequest():  Arguments: 
15:55:10.145 -> [198307][E][WebServer.cpp:649] _handleRequest(): request handler not found
15:55:10.145 -> [198314][W][WebServer.cpp:436] send(): content length is zero
```

tablatronix, 2022-12-21:
Can you turn on ESP debugging ? 
-2 networks ???? 
interesting

DonatelloX, 2022-12-22:
In my test, I used `wifiManager.setDebugOutput(true);`, but I don't understand how to increase the debug level.

tablatronix, 2022-12-22:
in arduino? its in the menu

DonatelloX, 2022-12-22:
If you see the reports I posted above, I used DEBUG and VERBOSE ("core debug level" option in menu of the Arduino IDE).
I noticed that when library works (it has successfully searched for wifi networks) the message appears:
`[ 37852][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE`
While this message is missing when scan fails.

tablatronix, 2022-12-22:
you need wifi debug level, ah maybe there is just nothing useful then.
This looks like an esp library issue, is there anything special about your envirmoment, how many aps? Clean power supply? no other wifi close by

DonatelloX, 2022-12-22:
At the moment I have only 1 WiFi AP, there are no other APs nearby. But even if I move to an area with dozens of APs WiFi, many times they are not scanned, only sometimes there is a list of all available APs.
I powered the ESP32 with a stabilized power supply, if I try with an ESP32 development kit connected to the PC (therefore power supply from PC) I get the same results.

tablatronix, 2022-12-22:
And you are definitely on 2.0.5? I will see if I can try to reproduce or at least add some workaround.

Can you add a wifi mode() to the start of your sketch to make sure wifi is initializing early enough and see if that makes a diffeence?

tablatronix, 2022-12-22:
It looks like when it fails scan wait is timing out because scan is taking forever, it seems to eventually finish, but its taking awhile..

16:02:57.627 -> [ 33184][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE

DonatelloX, 2022-12-27:
I used the version 2.0.5 of the ESP32 Core Arduino and then the version 2.0.6 but the problem remains.
Tried to use different wifi modes but the problem is not solved.

DonatelloX, 2023-01-16:
I tested with ESP32-S3 and I have the same problem: the list of WiFi networks is not always visible.

georgevbsantiago, 2023-08-29:
I have the same problem.
I also identified that the ESP32 (devkit with Platformio) is taking time to map the Routers signal. And it doesn't always work. Sometimes I have to click the Refresh button for it to work.

```
platform = espressif32@6.3.2
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
	WiFiManager@2.0.16-rc.2
```

tablatronix, 2023-09-08:
There has not been any updates to this, I will have to check if it is still causing issues.