tzapu/WiFiManager #1536: Parameters aren't showing on the portal. - *wm:[ERROR] WiFiManagerParameter is out of scope

bentodman, 2022-12-22:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.0.3

### Description

Everything seems to compile and upload fine. 
When i navigate to the portal i'm unable to see any parameters / input forms. 
In the serial monitor I'm getting an error *wm:[ERROR] WiFiManagerParameter is out of scope.
I've tried stripping this back out and i'm struggling to get it to work, sorry i can't be anymore detailed but happy to test etc as needed!

### Settings in IDE

Module:  			Wemos D1 Mini (Clone)

Additional libraries: 

### Sketch

```cpp
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager 2.0.14 Beta
#include <ESP8266WiFi.h>
#include <PubSubClient.h>

#ifdef ESP32
  #include <SPIFFS.h>
#endif

#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson 6.19.4

//define your default values here, if there are different values in config.json, they are overwritten.
char mqtt_server[40] = "connect.mybroker.com";
char mqtt_port[6] = "1883";
char api_token[34] = "YOUR_API_TOKEN";

//flag for saving data
bool shouldSaveConfig = false;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

const char* mqtt_username = "USERNAME";
const char* mqtt_password = "PASSWORD";
const char* deviceId = "0001";
String command;

WiFiManager wifiManager;
WiFiClient espClient;
PubSubClient client(espClient);

void setup() {
  Serial.begin(115200);
  Serial.println();
  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  Serial.println("mounting FS...");

  if (SPIFFS.begin()) {
    Serial.println("mounted file system");
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      Serial.println("reading config file");
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        Serial.println("opened config file");
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);

 #if defined(ARDUINOJSON_VERSION_MAJOR) && ARDUINOJSON_VERSION_MAJOR >= 6
        DynamicJsonDocument json(1024);
        auto deserializeError = deserializeJson(json, buf.get());
        serializeJson(json, Serial);
        if ( ! deserializeError ) {
#else
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        json.printTo(Serial);
        if (json.success()) {
#endif
          Serial.println("\nparsed json");
          strcpy(mqtt_server, json["mqtt_server"]);
          strcpy(mqtt_port, json["mqtt_port"]);
          strcpy(api_token, json["api_token"]);
        } else {
          Serial.println("failed to load json config");
        }
        configFile.close();
      }
    }
  } else {
    Serial.println("failed to mount FS");
  }
  //end read

  // The extra parameters to be configured (can be either global or just in the setup)
  // After connecting, parameter.getValue() will get you the configured value
  // id/name placeholder/prompt default length
  
  WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40);
  WiFiManagerParameter custom_mqtt_port("port", "mqtt port", mqtt_port, 6);
  WiFiManagerParameter custom_api_token("apikey", "API token", api_token, 32);

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);

  //set static ip
  //wifiManager.setSTAStaticIPConfig(IPAddress(10, 0, 1, 99), IPAddress(10, 0, 1, 1), IPAddress(255, 255, 255, 0));

  //add all your parameters here
  wifiManager.addParameter(&custom_mqtt_server);
  wifiManager.addParameter(&custom_mqtt_port);
  wifiManager.addParameter(&custom_api_token);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //set minimu quality of signal so it ignores AP's under that quality
  //defaults to 8%
  //wifiManager.setMinimumSignalQuality();

  //sets timeout until configuration portal gets turned off
  //useful to make it all retry or go to sleep
  //in seconds
  //wifiManager.setTimeout(120);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  //read updated parameters
  strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(mqtt_port, custom_mqtt_port.getValue());
  strcpy(api_token, custom_api_token.getValue());
  Serial.println("The values in the file are: ");
  Serial.println("\tmqtt_server : " + String(mqtt_server));
  Serial.println("\tmqtt_port : " + String(mqtt_port));
  Serial.println("\tapi_token : " + String(api_token));

  //save the custom parameters to FS
  if (shouldSaveConfig) {
    Serial.println("saving config");
 #if defined(ARDUINOJSON_VERSION_MAJOR) && ARDUINOJSON_VERSION_MAJOR >= 6
    DynamicJsonDocument json(1024);
#else
    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
#endif
    json["mqtt_server"] = mqtt_server;
    json["mqtt_port"] = mqtt_port;
    json["api_token"] = api_token;

    File configFile = SPIFFS.open("/config.json", "w");
    if (!configFile) {
      Serial.println("failed to open config file for writing");
    }

#if defined(ARDUINOJSON_VERSION_MAJOR) && ARDUINOJSON_VERSION_MAJOR >= 6
    serializeJson(json, Serial);
    serializeJson(json, configFile);
#else
    json.printTo(Serial);
    json.printTo(configFile);
#endif
    configFile.close();
    //end save
  }

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
 // IO Setup
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
 //MQTT Server
  client.setServer(mqtt_server, atoi(mqtt_port));
  client.setCallback(mqttMSGcallback);
}

// Run when we receive a msg
void mqttMSGcallback(char* topic, byte* payload, unsigned int length) {
 // Add data to variable
  String message;
  for (int i = 0; i < length; i++) {
    message += (char)payload[i];
  }

  if (message == "1") {
  addCredit();
  }

}

// Adding a credit
void addCredit() {
Serial.print("Credit Added \n");  
  digitalWrite(LED_BUILTIN, LOW);
  delay(500);
  digitalWrite(LED_BUILTIN, HIGH);
}

// Handing MQTT connection and status. 
void reconnect() {
  while (!client.connected()) {
    Serial.print("Attempting Cloud Connection...");
    if (client.connect(deviceId, mqtt_username, mqtt_password)) {
      Serial.println("Connected To Cloud");
      client.subscribe(deviceId);
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      delay(5000);
    }
  }
}

void loop() { 
// Check if we have MQTT connection.
 if (!client.connected()) {
    reconnect();
  }
  client.loop();

// Backdoor commands via serial.
 if(Serial.available()){
        command = Serial.readStringUntil('\n');
         
        if(command.equals("settings")){
         wifiManager.startConfigPortal(deviceId);
        }
    }

}

```

### Debug Messages

```
messages here
```

tablatronix, 2022-12-22:
You are setting up params in setup, but starting the portal in loop, so they ARE out of scope

bentodman, 2022-12-22:
Doh! That makes sense, Thank you!

closed 2022-12-22 by bentodman

MEHUL95, 2023-01-01:
@bentodman did you found out a solution for this issue? I tried separating wifimanager configuration in a different function and calling that function from void.setup() from void.loop() but it didn't work, setup tab seems blank, I cannot fill any custom parameters.

Bruce17, 2023-01-02:
@MEHUL95 I had the same issue even with the example code [OnDemandConfigPortal.ino](https://github.com/tzapu/WiFiManager/blob/v2.0.15-rc.1/examples/Super/OnDemandConfigPortal/OnDemandConfigPortal.ino)

I moved the `WiFiManagerParameter` into the global scope f.e.:
```cpp
WiFiManagerParameter *custom_mqtt_server;
WiFiManagerParameter *custom_mqtt_port;
```

... and changed the code inside the setup to something like:
```cpp
custom_mqtt_server = new WiFiManagerParameter("server", "mqtt server", "", 40);
custom_mqtt_port = new WiFiManagerParameter("port", "mqtt port", "", 6);

// ...
wm.addParameter(custom_mqtt_server);
wm.addParameter(custom_mqtt_port);
```