tzapu/WiFiManager #1540: Erroe in setConnectTimeout()

rritzi, 2023-01-02:
even if i set _connectTimeout ,my programme keeps handing in the loop and doesnot comeout of it.

danielminson, 2023-03-22:
Please give a more comprehensive answer. When I try setting  wifimanager.setConnectTimeout(30), for example, it simple does not set the value. The debugging says that it is not set.
[image]

tablatronix, 2023-03-23:
sorry you said comeout and I thought you wanted the other timeout to abort the loop.

hmm, that should be working for wifi timeout..

Is your esp getting stuck here for 60seconds? This is a new ESP bug we are looking into.
But connecttimeout should be working, what version you using ?

closed 2023-03-23 by tablatronix

tablatronix, 2023-03-23:
It looks like https://github.com/tzapu/WiFiManager/commit/701d58fd0751f27008a3851ae5bd73d79ed50a17 introduced this, but this bug is old, but now the connection code is using NEW all the time as a workaround for esp32 wifiinit #1439

danielminson, 2023-03-23:
This is on autoconnect and it is getting stuck for 60 seconds before saying that the new WiFi connection failed (because of wrong password).

tablatronix, 2023-03-23:
I already fixed it

danielminson, 2023-03-23:
You fixed it for esp32? What logs do you need?

tablatronix, 2023-03-23:
Sorry, Are you using 8266 or esp32? Any serial logs will be useful.

I need to know the esp logs during the 60seconds of retry, specifically the arduino wifi event error

danielminson, 2023-03-23:
I am using esp32. How should I get this log? Is there any line of code you can provide me?

tablatronix, 2023-03-23:
umm if you are using arduino you can enable debugging in the menu I think

danielminson, 2023-03-23:
I am using platformio.

tablatronix, 2023-03-23:
ah you can add to your ini/env

`build_flags = -D WM_DEBUG_LEVEL=4 -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DCORE_DEBUG_LEVEL=5`

and add `Serial.debugOutput(true);` to code

tablatronix, 2023-03-27:
oh so you dont have this issue with correct password ever?

danielminson, 2023-03-27:
Sometimes it doesn’t connect with correct password, but it think it’s due to lower signal.

tablatronix, 2023-03-28:
I mean there is a workaround in wm for the bug, and I connecttimeout is also fixed

danielminson, 2023-03-28:
How can I get a version without this issues I am facing then? I am using the latest release…

tablatronix, 2023-03-28:
you are using git?

danielminson, 2023-03-28:
Yes, but I am downloading the library from the release page. I think you should make a new release if there are fixes in the git that are not in the latest public release. It would also be nice for updating the library on arduino board manager... 
[image]

ramanraja, 2024-03-01:
I still face this issue as of February 2024. I downloaded the latest release from Git. I am using ESP8266.
First I enter the correct password, and connect to the Wifi network successfully. Then I just switch off my wireless modem and restart the ESP. When it is unable to connect to the AP, it goes into a blocking WiFi Manager portal, and never times out.
```
    WiFiManager wm;
    wm.setTimeout(120);   
    // I also tried:
    //wm.setConfigPortalTimeout(120);  
    bool result = wm.autoConnect("MYAP");
    if (!result) {
      Serial.println("Failed to connect to WiFi network");
    }
 
```
It never reaches the line` if(!result)..`
I see on the serial console
```
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  9229 ms
```

ramanraja, 2024-03-02:
A follow up of the time out issue: I tried non-blocking mode, (appropriately making the wm object global and calling wm.process() in the main loop). There was  a different issue this time:
```
    wm.setConfigPortalBlocking(false);
    wm.setConfigPortalTimeout(180);
    if(!wm.autoConnect("AutoConnectAP")){
       Serial.println("Could not connect to the network");
    } else {
      Serial.println("Connected.");
    }
```

The portal starts all right, but it does not wait for the 3 minute timeout.
Before I could connect to the AP, `wm.autoConnect()` returns immediately with a 'false' return value.

```
*wm:AutoConnect 
*wm:Connecting to SAVED AP: INTOF
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  9228 ms
*wm:StartAP with SSID:  AutoConnectAP
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal 
Could not connect to the network.
```

tablatronix, 2024-03-03:
This is a bug in esp library. Adding your own connecttimeout will override it

timf34, 2024-03-05:
I also have this same issue. AutoConnect gets stuck in a blocking mode if the wrong ap and password are entered.

@tablatronix how would you recommend adding my own setConnectTimeout? I'm not sure how to get passed the fact that its blocking. 

Thanks

commit 4611c75d (2023-03-23): fix #1540