tzapu/WiFiManager #1541: Image (base64 in HTTP_ROOT_MAIN) above 70dpi is not displayed in HTML page

georgevbsantiago, 2023-01-04:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master
Esp8266 
Hardware: NodeMCU
Platformio  6.1.5
Core Version: 2.0.15-rc.1

### Description

I have identified that the default image example (base64) in HTTP_ROOT_MAIN[] has stopped working.
I have identified that if the image is above 70dpi (base64), it is not rendered in the HTML.
And if the image is above 100 dpi or larger than 10kb (or 20kb, I don't remember exactly), it causes a reset on esp8266.

I had success with a 70dpi color image (70 Width x 38 Height) with 4kbs.

I think this bug is causing a stutter when opening the homepage. Sometimes when trying “192.168.4.1” the page is blank. But if you try “192.168.4.1/wifi”, the AP SSID and Password registration page is displayed on the screen.

Example of the standard image (base64) of HTTP_ROOT_MAIN[] available in the Wifimanager code. Stopped working
`const char HTTP_ROOT_MAIN[]        PROGMEM = "<img title=' alt=' src='data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAADAAAAAwCAYAAABXAvmHAAADQElEQVRoQ+2YjW0VQQyE7Q6gAkgFkAogFUAqgFQAVACpAKiAUAFQAaECQgWECggVGH1PPrRvn3dv9/YkFOksoUhhfzwz9ngvKrc89JbnLxuA/63gpsCmwCADWwkNEji8fVNgotDM7osI/x777x5l9F6JyB8R4eeVql4P0y8yNsjM7KGIPBORp558T04A+CwiH1UVUItiUQmZ2XMReSEiAFgjAPBeVS96D+sCYGaUx4cFbLfmhSpnqnrZuqEJgJnd8cQplVLciAgX//Cf0ToIeOB9wpmloLQAwpnVmAXgdf6pwjpJIz+XNoeZQQZlODV9vhc1Tuf6owrAk/8qIhFbJH7eI3eEzsvydQEICqBEkZwiALfF70HyHPpqScPV5HFjeFu476SkRA0AzOfy4hYwstj2ZkDgaphE7m6XqnoS7Q0BOPs/sw0kDROzjdXcCMFCNwzIy0EcRcOvBACfh4k0wgOmBX4xjfmk4DKTS31hgNWIKBCI8gdzogTgjYjQWFMw+o9LzJoZ63GUmjWm2wGDc7EvDDOj/1IVMIyD9SUAL0WEhpriRlXv5je5S+U1i2N88zdPuoVkeB+ls4SyxCoP3kVm9jsjpEsBLoOBNC5U9SwpGdakFkviuFP1keblATkTENTYcxkzgxTKOI3jyDxqLkQT87pMA++H3XvJBYtsNbBN6vuXq5S737WqHkW1VgMQNXJ0RshMqbbT33sJ5kpHWymzcJjNTeJIymJZtSQd9NHQHS1vodoFoTMkfbJzpRnLzB2vi6BZAJxWaCr+62BC+jzAxVJb3dmmiLzLwZhZNPE5e880Suo2AZgB8e8idxherqUPnT3brBDTlPxO3Z66rVwIwySXugdNd+5ejhqp/+NmgIwGX3Py3QBmlEi54KlwmjkOytQ+iJrLJj23S4GkOeecg8G091no737qvRRdzE+HLALQoMTBbJgBsCj5RSWUlUVJiZ4SOljb05eLFWgoJ5oY6yTyJp62D39jDANoKKcSocPJD5dQYzlFAFZJflUArgTPZKZwLXAnHmerfJquUkKZEgyzqOb5TuDt1P3nwxobqwPocZA11m4A1mBx5IxNgRH21ti7KbAGiyNn3HoF/gJ0w05A8xclpwAAAABJRU5ErkJggg==' /><h1>{v}</h1><h3>WiFiManager</h3>";`

tablatronix, 2023-01-06:
Definitely a memory issue. How much heap do you have?

georgevbsantiago, 2023-08-29:
I identified the same issue with ESP32 (devkit with Platformio).
The default library image is not showing in the html page.

closed 2023-09-08 by tablatronix