tzapu/WiFiManager #1543: wifi manager seems to redirect to 0.0.0.0 for the CaptivePortal

EtoTen, 2023-01-07:
### Basic Infos

#### Hardware

CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/az-delivery-devkit-v4.html
PLATFORM: Espressif 32 (2022.12.2) > AZ-Delivery ESP-32 Dev Kit C V4
HARDWARE: ESP32 240MHz, 520KB RAM, 4MB Flash

WiFimanager Branch/Release:  Master (latest from github not release)

### Description

Problem description:

Everything seems to work, except I get redirected to 0.0.0.0 instead of the Captive Portal IP. I think the issue may have started after I upgraded to the 2022.12.2 release of the Espressif platform.

I identified the issue being at WifiManager.cpp, line 2403
'''
  String serverLoc = toStringIp(server->client().localIP());
  if(_httpPort != 80) serverLoc += ":" + (String)_httpPort; // add port if not default
  bool doredirect = serverLoc != server->hostHeader(); // redirect if hostheader not server ip, prevent redirect loops
'''
As a quick hack, this worked as a work arround:
'''
  String serverLoc =  "192.168.4.1"
  if(_httpPort != 80) serverLoc += ":" + (String)_httpPort; // add port if not default
  bool doredirect = serverLoc != server->hostHeader(); // redirect if hostheader not server ip, prevent redirect loops
'''

### Settings in IDE

[env:node32s]
platform = espressif32
board = az-delivery-devkit-v4
board_build.mcu = esp32
board_build.f_cpu = 240000000L
framework = arduino
monitor_speed = 115200
upload_protocol = esptool
upload_flags = 
	--connect-attempts= 200
lib_deps = 
	paulstoffregen/Time@^1.6.1
	jchristensen/Timezone@^1.2.4
	plerup/EspSoftwareSerial@^6.15.2
	sparkfun/SparkFun u-blox GNSS Arduino Library@^2.2.0
	lennarthennigs/Button2@^2.2.2
	rpolitex/ArduinoNvs@^2.5
	earlephilhower/ESP8266Audio@^1.9.7
	fabianoriccardi/Melody Player@^2.4.0
	https://github.com/tzapu/WiFiManager.git

### Sketch

```cpp
#BEGIN
#include <Arduino.h>
#include <driver/dac.h>

#include <WiFi.h>
#include <SPI.h>
#include <TimeLib.h>
#include <Time.h>
#include <Timezone.h>
#include <DNSServer.h>
#include <WebServer.h>

#include <WiFiManager.h> //https://github.com/tzapu/WiFiManager

#include "LEDControl.h"
#include "SPI_Nixie_Interface.h"
#include "NTP_time.h"
#include "GPS_time.h"
#include "main.h"

#include <stdio.h>
#include <stdlib.h>

#include "ArduinoNvs.h"
#include "Button2.h"

void setup() {
  wifiManager.setHostname("redacted");
  wifiManager.setCountry("US");
  wifiManager.setDebugOutput(true);
  wifiManager.setConfigPortalBlocking(false);
  WiFi.setSleep(false);
  WiFi.setTxPower(WIFI_POWER_19_5dBm); // Set WiFi RF power output to highest level
  wifiManager.startConfigPortal(AP_NAME);
}

void loop() {
     wifiManager.process();
}
#END
```

### Debug Messages

```
*wm:[2] <- Request redirected to captive portal 
```

PaddyCube, 2023-01-09:
It's the same on my end. Recently started using Wifi Manager and gets redirected to 0.0.0.0
Your workaround fixed it for me for now.

EtoTen, 2023-01-12:
I messed with it more, and here is the fix that works for both cases:

 ```
String serverLoc;
  if ((WiFi.status()) != WL_CONNECTED)
    serverLoc = toStringIp(WiFi.softAPIP());
  else
    serverLoc = toStringIp(WiFi.localIP());

  if (_httpPort != 80)
    serverLoc += ":" + (String)_httpPort;              // add port if not default
  bool doredirect = serverLoc != server->hostHeader(); // redirect if hostheader not server ip, prevent redirect loops
  // doredirect = !isIp(server->hostHeader()) // old check
```

I will try and create a fork / pull request with the fix.

EtoTen, 2023-01-12:
Pull request created https://github.com/tzapu/WiFiManager/pull/1547

tablatronix, 2023-01-13:
I am not seeing an issue...
do you have more info about your env?

Works fine for me, this seems related to your dns issue, and have you tried full erase?

EtoTen, 2023-01-13:
I don't have a DNS problem, I think you miss-read as I replied to someone else's issue offering a solution for DNS.

I am using this for the platform:
https://github.com/espressif/arduino-esp32/releases/tag/2.0.6
(ESP32 Arduino 2.0.6 based on ESP-IDF 4.4.3)

and am using VS.code / platformIO for IDE
 
What about your self? What sdk versions are you building off?

Per your advise, I just did a full erase without positive results.

Any other items that I could give you?

tablatronix, 2023-01-13:
Same env
```
[env:esp32_dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_extra_dirs = /Users/shawn/projects/microcontrollers/dev/libraries
build_flags = -DWM_DEBUG_PORT=Serial  -DWM_DEBUG_LEVEL=5 -DDEBUG_ESP_PORT=Serial  -DDEBUG_ESP_CORE -DDEBUG_ESP_WIFI -DDEBUG_ESP_HTTP_CLIENT -DDEBUG_ESP_HTTP_UPDATE -DDEBUG_ESP_HTTP_SERVER -DDEBUG_ESP_UPDATER -DDEBUG_ESP_OTA -DDEBUG_TLS_MEM
```

tablatronix, 2023-01-13:
heres what debug logs should contain

```php
*wm:[1] Starting Web Portal 
[   892][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
[   893][V][WebServer.cpp:87] WebServer(): WebServer::Webserver(port=80)
*wm:[2] HTTP server started 
*wm:[2] WiFi Scan ASYNC started 
[   917][V][WiFiGeneric.cpp:340] _arduino_event_cb(): STA Started
[   919][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 2 - STA_START
*wm:[3] setupDNSD 
*wm:[3] dns server started with ip:  192.168.4.1
*wm:[2] Config Portal Running, non blocking (processing) 
[  7568][V][WiFiGeneric.cpp:383] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 5
[  7569][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[ 10960][V][WiFiGeneric.cpp:407] _arduino_event_cb(): AP Station Connected: MAC: 14:98:77:55:c1:e3, AID: 1
[ 10961][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 12 - AP_STACONNECTED
[ 11015][V][WiFiGeneric.cpp:421] _arduino_event_cb(): AP Station IP Assigned:192.168.4.2
[ 11016][D][WiFiGeneric.cpp:931] _eventCallback(): Arduino Event: 14 - AP_STAIPASSIGNED
[ 11503][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[ 11504][V][Parsing.cpp:122] _parseRequest(): method: GET url: /hotspot-detect.html search: 
[ 11509][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[ 11514][V][Parsing.cpp:227] _parseRequest(): headerValue: captive.apple.com
[ 11521][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[ 11527][V][Parsing.cpp:227] _parseRequest(): headerValue: close
[ 11533][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[ 11539][V][Parsing.cpp:227] _parseRequest(): headerValue: CaptiveNetworkSupport-428.120.3 wispr
[ 11547][V][Parsing.cpp:254] _parseArguments(): args: 
[ 11552][V][Parsing.cpp:237] _parseRequest(): Request: /hotspot-detect.html
[ 11559][V][Parsing.cpp:238] _parseRequest():  Arguments: 
[ 11564][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:[2] captivePortal check 
*wm:[4] -> captive.apple.com 
*wm:[2] <- Request redirected to captive portal 
[ 11593][W][WebServer.cpp:436] send(): content length is zero
[ 11639][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=192.168.4.1
[ 11640][V][Parsing.cpp:122] _parseRequest(): method: GET url: / search: 
[ 11642][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[ 11648][V][Parsing.cpp:227] _parseRequest(): headerValue: 192.168.4.1
[ 11654][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[ 11660][V][Parsing.cpp:227] _parseRequest(): headerValue: close
[ 11666][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[ 11672][V][Parsing.cpp:227] _parseRequest(): headerValue: CaptiveNetworkSupport-428.120.3 wispr
[ 11680][V][Parsing.cpp:254] _parseArguments(): args: 
[ 11685][V][Parsing.cpp:237] _parseRequest(): Request: /
[ 11690][V][Parsing.cpp:238] _parseRequest():  Arguments: 
*wm:[2] <- HTTP Root 
*wm:[4] -> 192.168.4.1 
*wm:[3] [WIFI] reportStatus prev: WL_IDLE_STATUS
```

EtoTen, 2023-01-13:
Stand-by, let me build a skeleton project that just has the WifiManager code to test with on my end.

EtoTen, 2023-01-13:
```
Starting...
nvsTimeOffset: 492
time_offset: 4294967288
*wm:[1] resetSettings
*wm:[3] WiFi_enableSTA enable
*wm:[1] SETTINGS ERASED 
Starting Wifi AP
*wm:[2] Starting Config Portal
*wm:[3] WiFi station disconnect
*wm:[3] WiFi_enableSTA enable
*wm:[2] Disabling STA
*wm:[2] Enabling AP
*wm:[1] StartAP with SSID:  NixieClock
*wm:[2] AP has anonymous access!
*wm:[1] SoftAP Configuration
*wm:[1] -------------------- 
*wm:[1] ssid:             NixieClock
*wm:[1] password:
*wm:[1] ssid_len:         10
*wm:[1] channel:          1
*wm:[1] authmode:
*wm:[1] ssid_hidden:     
*wm:[1] max_connection:   4
*wm:[1] country:          US ␁␋␔
*wm:[1] beacon_interval:  100(ms)
*wm:[1] --------------------
*wm:[1] AP IP address: 192.168.4.1
*wm:[2] setting softAP Hostname: nixie-clock
*wm:[3] hostname: AP:  nixie-clock
*wm:[2] WiFiSetCountry to US
*wm:[2] [OK] esp_wifi_set_country:  US
*wm:[3] setupConfigPortal
*wm:[1] Starting Web Portal
*wm:[2] HTTP server started 
*wm:[2] WiFi Scan ASYNC started
*wm:[3] setupDNSD
*wm:[3] dns server started with ip:  192.168.4.1
*wm:[2] Config Portal Running, non blocking (processing)
*wm:[1] [SYS] WM version:  v2.0.15-rc.1
*wm:[1] [SYS] Arduino version:  2.0.6
*wm:[1] [SYS] ESP SDK version:  v4.4.3.20221227
*wm:[1] [SYS] Free heap:        267928
*wm:[1] [SYS] Chip ID: 64496192
*wm:[1] [SYS] Chip Model: ESP32-D0WDQ6
*wm:[1] [SYS] Chip Cores: 2
*wm:[1] [SYS] Chip Rev: 1
Ready!
*wm:[2] NUM CLIENTS:  0
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1
*wm:[2] <- HTTP Root
*wm:[2] <- Request redirected to captive portal
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] <- Request redirected to captive portal 
*wm:[2] NUM CLIENTS:  1

```

Whole skeleton package I put here: [(https://github.com/EtoTen/WifiManager_issue_1543_test](https://github.com/EtoTen/WifiManager_issue_1543_test)

tablatronix, 2023-01-14:
No esp debug there

EtoTen, 2023-01-15:
Ahh sorry I was going crazy trying to enable full debuging, then I just copied your build_flags and it worked.

Here is an example of when it redirects to 0.0.0.0. 

```
Starting...
HV5222 initialized SPI
RTC singleton constructor.
[   832][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=22 scl=21 freq=100000
[   835][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Current RTC Time (unixtime): 1673771664
nvsTimeOffset: 492
time_offset: -8
[  5859][W][esp32-hal-timer.c:226] timerAttachInterrupt(): EDGE timer interrupt is not supported! Setting to LEVEL...
[  5879][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[  5989][W][WiFiGeneric.cpp:1299] setTxPower(): Neither AP or STA has been started
Starting Wifi AP[  5989][V][WiFiGeneric.cpp:338]
 _arduino_event_cb(): STA Started
[  5993][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
[MEM] free: 271672 | max: 45044 | frag:  84% 
*wm:[2] Starting Config Portal
[  6013][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  6013][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
[MEM] free: 267704 | max: 45044 | frag:  84%
*wm:[3] WiFi station disconnect
[MEM] free: 267896 | max: 45044 | frag:  84%
*wm:[3] WiFi_enableSTA enable
[  6042][V][WiFiGeneric.cpp:341] _arduino_event_cb(): STA Stopped
[  6042][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 3 - STA_STOP
[MEM] free: 269500 | max: 45044 | frag:  84% 
*wm:[2] Disabling STA
[MEM] free: 269500 | max: 45044 | frag:  84%
*wm:[2] Enabling AP
[MEM] free: 269500 | max: 45044 | frag:  84%
*wm:[1] StartAP with SSID:  NixieClock
[MEM] free: 269500 | max: 45044 | frag:  84%
*wm:[2] AP has anonymous access!
[  6079][V][WiFiGeneric.cpp:393] _arduino_event_cb(): AP Stopped
[  6079][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 11 - AP_STOP
[MEM] free: 269616 | max: 4504[  6087][V][WiFiGeneric.cpp:390] _arduino_event_cb(): AP Started
[  6093][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 10 - AP_START
4 | frag:  84%
*wm:[1] SoftAP Configuration
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] --------------------
[MEM] free: 269292 | max: 45044 | frag:  84% 
*wm:[1] ssid:             NixieClock
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] password:
[MEM] free: 269292 | max: 45044 | frag:  84% 
*wm:[1] ssid_len:         10
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] channel:          1
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] authmode:        
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] ssid_hidden:
[MEM] free: 269292 | max: 45044 | frag:  84%
*wm:[1] max_connection:   4
[MEM] free: 269292 | max: 45044 | frag:  84% 
*wm:[1] country:          US ␁␋␔
[MEM] free: 269484 | max: 45044 | frag:  84%
*wm:[1] beacon_interval:  100(ms)
[MEM] free: 269484 | max: 45044 | frag:  84% 
*wm:[1] --------------------
[MEM] free: 268820 | max: 45044 | frag:  84% 
*wm:[1] AP IP address: 192.168.4.1
[MEM] free: 268780 | max: 45044 | frag:  84%
*wm:[2] setting softAP Hostname: nixie-clock
[MEM] free: 268780 | max: 45044 | frag:  84%
[... 247 lines cut ...]
[MEM] free: 252164 | max: 45044 | frag:  83% 
*wm:[2] <- Request redirected to captive portal
[ 27638][W][WebServer.cpp:436] send(): content length is zero
[ 27645][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=0.0.0.0
[ 27646][V][Parsing.cpp:122] _parseRequest(): method: GET url: /generate_204 search:
[ 27654][V][Parsing.cpp:226] _parseRequest(): headerName: Connection
[ 27659][V][Parsing.cpp:227] _parseRequest(): headerValue: close
[ 27666][V][Parsing.cpp:226] _parseRequest(): headerName: User-Agent
[ 27671][V][Parsing.cpp:227] _parseRequest(): headerValue: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/60.0.3112.32 Safari/537.36
[ 27686][V][Parsing.cpp:226] _parseRequest(): headerName: Host
[ 27691][V][Parsing.cpp:227] _parseRequest(): headerValue: www.google.us
[ 27698][V][Parsing.cpp:226] _parseRequest(): headerName: Accept-Encoding
[ 27704][V][Parsing.cpp:227] _parseRequest(): headerValue: gzip
[ 27710][V][Parsing.cpp:254] _parseArguments(): args:
[ 27715][V][Parsing.cpp:237] _parseRequest(): Request: /generate_204
[ 27721][V][Parsing.cpp:238] _parseRequest():  Arguments:
[ 27726][E][WebServer.cpp:649] _handleRequest(): request handler not found
[MEM] free: 253636 | max: 45044 | frag:  83% 
*wm:[4] -> www.google.us
[MEM] free: 253780 | max: 45044 | frag:  83%
*wm:[2] <- Request redirected to captive portal
[ 27754][W][WebServer.cpp:436] send(): content length is zero
[MEM] free: 257864 | max: 45044 | frag:  83% 
*wm:[2] NUM CLIENTS:  1
```

tablatronix, 2023-01-15:
`[196564][V][WebServer.cpp:296] handleClient(): New client: client.localIP()=0.0.0.0`

thats is soo bizarre, hmm

tablatronix, 2023-01-15:
aha

`[  2266][E][WiFiServer.cpp:91] begin(): ---------------- IPv6`

tablatronix, 2023-01-15:
This is not in my library, you are using some alternative libraries?

tablatronix, 2023-01-15:
https://github.com/espressif/arduino-esp32/pull/7520

you are using experimental PRs?

EtoTen, 2023-01-15:
I don't think I am on using alternative libraries or experimental PRs (not sure what those are actually). 
Can you tell based on the sample project I put up?

EtoTen, 2023-01-16:
Okay I know what happened now. I had this installed: https://github.com/tasmota/platform-espressif32/releases/tag/2022.12.2 
"Tasmota Core 2.0.6 with IPv6 support" because I was testing single core mode and, did not realize it was defaulting to it vs the normal framework core. Reverting to the normal framework made redirects work ok. I also confirmed manually disabling ipv6 for the connection fixed it. 
I guess if they do merge that patch for ipv6 into the normal espressif core it would also create the same issue, so the issue / patch may still be valid.

aFunkyBass, 2023-08-21:
> Reverting to the normal framework made redirects work ok. I also confirmed manually disabling ipv6 for the connection fixed it.

Hello @EtoTen I had exactly the same situation: installed Tasmota-32 on my ESP and now reinstalling my app which uses WiFiManager, it redirects on 0.0.0.0

Sorry for my newbie question how did you get it back to the normal behaviour?

tablatronix, 2023-08-21:
There is a workaround, but I have not added it to wm, because it seems like an edge case bug, not sure tho.
I thought this was fixed in esp core

EtoTen, 2023-08-21:
> 

How I fixed it was to not use "Tasmota Core". Have they since merged those changes to the normal core?

tablatronix, 2023-08-22:
afaik tasmota is constantly updating from esp core/idf but not sure as it does use a custom fork

aFunkyBass, 2023-09-04:
I noticed that on PIO Homepage -> Platforms I had 2 Espressif 32 installed, one was probably installed by the Tasmota source.

My solution was to uninstall both of them and reinstall just the official one.

tablatronix, 2023-09-18:
https://github.com/espressif/arduino-esp32/pull/7520

Adding workaround proposed above, but wrapped it in a ip check.

Ideally I can add an ipv6 check and handle it special that way. but I am not sure the ideal workaround without rereading all that. 

- [ ] Try compiling IDF without IPv6
- [ ] add IPv6 check and reinterpret the localip from safest source.
- [ ] Rework the apssid/sta ip logic for workaround for getting server ip, can check headers etc.

mineshaftgap, 2023-11-28:
I am experiencing this and do not have Tasmota installed, hard coding the "192.168.4.1" as mentioned in https://github.com/tzapu/WiFiManager/issues/1543#issue-1523775677 fixed it.

tablatronix, 2023-11-28:
This is fixed in master

closed 2023-11-28 by tablatronix

mineshaftgap, 2023-12-02:
@tablatronix thanks!

For anyone who is developing with PlatformIO, I explicitly used the master branch in my platformio.ini:
```ini
lib_deps = 
	https://github.com/tzapu/WiFiManager.git
```

tablatronix, 2023-12-03:
Ill start pushing releases for pio since they don support pre releases