tzapu/WiFiManager #1544: dns does not look to be set properly after connecting to wifi

BouchardClaude, 2023-01-08:
### Basic Infos
I appear that the dns is not set properly when using DHCP STA , preventing to communicate with services using dns name.

### Description
When using autoconnect, the ip, mask and gateway are set properly, but it look like the dns is never set. I'm not sure if its a desired behaviour or if its linked to WifiManager itself or a sub library.

Additional libraries: 

### Sketch
Exemple of code that produce the ouput 
```
   if (wm.autoConnect(hostname, pass)) {
        debugPrintf("WiFi connected - IP = %i.%i.%i.%i\n", WiFi.localIP()[0],
                    WiFi.localIP()[1], WiFi.localIP()[2], WiFi.localIP()[3]);
        uint8_t dns_no;
        WiFi.dnsIP(dns_no);
        debugPrintf("DNS-ADDRESS: %d\n", dns_no);
```
will output the following:
### Debug Messages

```
*wm:[2] Connection result: WL_CONNECTED
*wm:[1] AutoConnect: SUCCESS
*wm:[2] Connected in 4519 ms
*wm:[1] STA IP Address: 192.168.5.43
[00:00:04] WiFi connected - IP = 192.168.5.43
[00:00:04] DNS-ADDRESS: 0
```
And ill be able to use anything not using DNS. 
Is theyre a way to hard-set the DNS without using static ip?

EtoTen, 2023-01-13:
Try 
`cpp Serial.println( "DNS:" + WiFi.dnsIP().toString());`

also try 

`cpp WiFi.config(WiFi.localIP(),` WiFi.gatewayIP(), WiFi.subnetMask(), IPAddress(8,8,8,8)); 
( from https://github.com/espressif/arduino-esp32/issues/2990 )

tablatronix, 2023-01-13:
What esp lib version?

tablatronix, 2023-01-13:
Works for me

```

[INFORMATION] TEST
*wm:[SYS] WM version:  v2.0.15-rc.1
*wm:[SYS] Arduino version:  2.0.6
*wm:[SYS] ESP SDK version:  v4.4.3
*wm:[SYS] Free heap:        288364
*wm:[SYS] Chip ID: 3753011116
*wm:[SYS] Chip Model: ESP32-D0WDQ5
*wm:[SYS] Chip Cores: 2
*wm:[SYS] Chip Rev: 1

....
connected...yeey :)
[WIFI] WIFI INFO DEBUG
[WIFI] SAVED: YES
[WIFI] SSID: leela
[WIFI] PASS: Turdtr0n
[WIFI] HOSTNAME: WM_ESP32_DFB267AC
[WIFI] DNS:192.168.20.190
[WIFI] DNS:192.168.20.1
```

BouchardClaude, 2023-01-13:
Thanks for the help :)

Okay so first I completly misunderstood the dnsIP() usage in c++. @EtoTen thanks to your snippet i was able to confirm that I'm receiving the DNS  ip correcly.

To test it, I try to sync the time. 

```        
debugPrintf("DNS - IP = %i.%i.%i.%i\n", WiFi.dnsIP()[0],WiFi.dnsIP()[1], WiFi.dnsIP()[2], WiFi.dnsIP()[3]);
debugPrintf("GW - IP = %i.%i.%i.%i\n", WiFi.gatewayIP()[0],WiFi.gatewayIP()[1], WiFi.gatewayIP()[2], WiFi.gatewayIP()[3]);
debugPrintf("GW - IP = %i.%i.%i.%i\n", WiFi.subnetMask()[0], WiFi.subnetMask()[1], WiFi.subnetMask()[2], WiFi.subnetMask()[3]);

 timeSync("GMT-5", "time-a-g.nist.gov", "utcnist2.colorado.edu");
````

and it gave me this ( everything look look ip wise, but it fail to set the time ( or to do any dns query that I tested)
```
[00:00:05] WiFi connected - IP = 192.168.5.35
[00:00:05] MAC-ADDRESS: 3616196xxxxx
[00:00:05] DNS - IP = 192.168.4.1
[00:00:05] GW - IP = 192.168.4.1
[00:00:05] GW - IP = 255.255.252.0
Syncing time........................................
Synchronized time: Thu Jan  1 05:00:25 1970  ( time failed to sync)

```
but if I dont use dns, then its working ( so packets are going in and out and are forwareded properly by the router.
example:
 
```[00:00:05] WiFi connected - IP = 192.168.5.35
[00:00:05] MAC-ADDRESS: 3616196358217
[00:00:05] DNS - IP = 192.168.4.1
[00:00:05] GW - IP = 192.168.4.1
[00:00:05] GW - IP = 255.255.252.0
Syncing time.
Synchronized time: Fri Jan 13 18:09:35 2023````( time is good)

````

The funny thing is I tried with a different network and it worked just fine ( exemple using my cell as wifi AP and using dns to sync time)

`        timeSync("GMT-5", "time-a-g.nist.gov", "utcnist2.colorado.edu"); `
will work perfetly:

```
[00:01:19] WiFi connected - IP = 192.168.54.171
[00:01:19] MAC-ADDRESS: 3616196358xxx
[00:01:19] DNS - IP = 192.168.54.123
[00:01:19] GW - IP = 192.168.54.123
[00:01:19] GW - IP = 255.255.255.0
Syncing time...............................
Synchronized time: Fri Jan 13 19:17:30 2023

```

Can it be that the issue is with the subnet I use internally confliting with the one used when in portal mode?  
I cannot exclude something fishy with the router too, but every device I have use this with no issue.

tablatronix, 2023-01-13:
Yeah or it could be subnet mask not allowing that ip

BouchardClaude, 2023-01-13:
thats why im suspecting a bug, cause 192.168.4.1 255.255.252.0 is a valid subnet and 192.168.5.35 is a valid ip in this subnet
If I take the same <non working> router and change the mask to be a /24 then everything is working good
[22:27:41] WiFi connected - IP = 192.168.0.26
[22:27:41] MAC-ADDRESS: 3616196358xxx
[22:27:41] DNS - IP = 192.168.0.1
[22:27:41] GW - IP = 192.168.0.1
[22:27:41] GW - IP = 255.255.255.0
Syncing time
Synchronized time: Sat Jan 14 03:27:41 2023

So I think the network mask is unproperly process OR the 192.168.4.x range is conflicting with something, but thats unlikely since at that point its in STA mode

tablatronix, 2023-01-14:
There is a way to change the default server ip but I forget at the moment how. Would be interesting to test a different class. Ill try to test this in my lab one day.

PatrickV2, 2023-01-15:
It seems to me that I have the same problem. I added the WiFiManager to one of my programs and without it my MQTT broker (mqttbroker) is resolved normally but with WiFiManager I can only use the IP-number directy.

BouchardClaude, 2023-01-16:
That's interesting cause its the exact same issue for me (MQTT broker) , but your mask is /24 so I was probably wrong thinking its related to mask.
Funny part is I tried many time, and sometimes it work, sometimes no with the same setup  .. so so far its not easy to point a finger on whats not working.  I'm working on trying to be able to reproduce the issue every time so we can see what element is causing the issue

Eisbaeeer, 2025-01-23:
I think it´s same issue on my side. If I connect first time to a new Wifi, DNS is not working. If I reboot the device after connecting to Wifi, the DNS can resolve.