tzapu/WiFiManager #1546: Autocorrect enabled for wifi "Password" field on WiFi setup page

pomplesiegel, 2023-01-11:
Currently a browser will try to autocorrect the value a user enters for the wifi password, because of HTML settings. This is because there are no fields `autocorrect="off"` and `autocapitalize="none"` for passwords. 

Current HTML for password field:
```<input id="p" name="p" maxlength="64" type="password" placeholder="">```

In contrast, the SSID field has both of these in place correctly: 
```<input id="s" name="s" maxlength="32" autocorrect="off" autocapitalize="none" placeholder="">```

Is there an easy way to add this into the password's HTML using the current API? I imagine `autocorrect="off"` and `autocapitalize="none"` should always be in place, since passwords are often not proper english words. 

Thank you,
Michael

tablatronix, 2023-01-13:
shouldn't this be the default for browsers on input type=password?

pomplesiegel, 2023-01-13:
Interesting! For whatever reason at least through the "safari" browser on Mac OS desktop it still attempted to create multiple capitalized words out of my password input. Are you seeing a different behavior on your browser?

tablatronix, 2023-01-13:
I can check, but we can add that just to be sure I guess, it just takes memory