tzapu/WiFiManager #1549: ESP32 RECONNECT

PabloOyarzo, 2023-01-14:
Hi is it possible to provide a reconnect sample code?

As autoreconnect doesn't seem to be working. 

 Saw @tablatronix said in another issue that you check wifi status and do wifi begin again

tablatronix, 2023-01-14:
I think it is supposed to be working now, what are you seeing?

PabloOyarzo, 2023-01-14:
I Haven't turned on the debug verbose, but it doesn't reconnect automatically. I'm using an ESP32, it's using the WiFi Library from arduino esp32 the latest (which is supposed to fix all reconnect issues) and wifi managers latest too. I'm testing with a hostpot right next to the esp32.

Tried it with and without autoreconnect(true).

PabloOyarzo, 2023-01-17:
It works now, still would like to see that check and reconnect little snippet of code that you use please!

yfrit83, 2024-02-06:
Hi, yes it work now, but only one time!

To reprosuce the bug :
1. Boot  ESP32-S3 device
2. Cut the wifi
3. ESP32-S3 device disconnect.
4. Start the wifi
5. ESP32-S3 device reconect automaticly
6. cut the wifi again
7. ESP32-S3 device disconect
8. start the wifi again
9. ESP32-S3 device dosn't reconnect. 

Note : The device need to be rebooted to reconect.

parameter :
  WiFi.mode(WIFI_STA);
  wm.setConfigPortalBlocking(false);
  wm.setConfigPortalTimeout(60);
  wm.setBreakAfterConfig(true);
  wm.setWiFiAutoReconnect(true);

- vm.process(); in loop

tablatronix, 2024-02-11:
Whats logs say?