tzapu/WiFiManager #1553: Cant get the runtime web portal to work - startWebPortal()

jonasbjurel, 2023-01-22:
### Basic Infos
I can't get the run-time webportal to work.
 

#### Hardware
WiFimanager Branch/Release:  2.0.15-rc-1

Esp8266/Esp32: 

Hardware:			ESP wrover

Core Version:      	2.0.0

### Description

Problem description
I want the webportal to run all the time such that I can perform an OTA at any time, the config portal works okay. After the config portal has concluded after a config save and connect , I try to start a runtime web-portal with no success, cannot connect to it with the IP-address the AP has assigned to my ESP32.
Simplified code snippet:
    WiFiManager wifiManager;
    wifiManager.setAPStaticIPConfig(IPAddress(WIFI_MGR_AP_IP),
                                    IPAddress(WIFI_MGR_AP_GW_IP),
                                    IPAddress(255,255,255,0));
    wifiManager.setAPCallback(configModeCb);
    wifiManager.setPreSaveConfigCallback(preSaveConfigCb);
    wifiManager.setSaveConfigCallback(saveConfigCb);
    wifiManager.setTitle(WIFI_MGR_HTML_TITLE);
    wifiManager.setConnectTimeout(WIFI_MGR_STA_CONNECT_TIMEOUT_S);
    wifiManager.setConfigPortalTimeout(WIFI_MGR_AP_CONFIG_TIMEOUT_S);
    wifiManager.setShowStaticFields(true);
    wifiManager.setShowDnsFields(true);
    wifiManager.setShowInfoErase(true);
    wifiManager.setShowInfoUpdate(true);
    wifiManager.setBreakAfterConfig(true);
    wifiManager.setShowPassword(false);
    wifiManager.setCaptivePortalEnable(true);
    wifiManager.setConfigResetCallback(resetCb);
    wifiManager.setConfigPortalTimeoutCallback(configurePortalConnectTimeoutCb);
    hostNameConfigParam = new WiFiManagerParameter("HostName", "HostName",  networkConfig.hostName, 31);
    wifiManager.addParameter(hostNameConfigParam);
    mqttServerUriConfigParam = new WiFiManagerParameter("MQTTserverURI", "MQTTserverURI", networkConfig.mqttUri, 100);
    wifiManager.addParameter(mqttServerUriConfigParam);
    char tmpMqttPort[6];
    mqttServerPortConfigParam = new WiFiManagerParameter("MQTTserverPort", "MQTTserverPort",
                                                             itoa(networkConfig.mqttPort,
                                                             tmpMqttPort, 10), 5);
   wifiManager.addParameter(mqttServerPortConfigParam);
   char apSsid[50];
   sprintf(apSsid, "%s_%s", WIFI_MGR_AP_NAME_PREFIX, getMac());
   wifiManager.startConfigPortal(apSsid);
   wifiManager.stopConfigPortal();
   wifiManager.startWebPortal();                                                   // Does not work

For full code see: https://github.com/jonasbjurel/GenericJMRIdecoder/blob/main/client/Arduino/genJMRIDecoderClient/networking.cpp

### Settings in IDE

Module:  			-

Additional libraries: 
See: https://github.com/jonasbjurel/GenericJMRIdecoder/blob/main/client/Arduino/genJMRIDecoderClient/networking.cpp

### Sketch
See: https://github.com/jonasbjurel/GenericJMRIdecoder/blob/main/client/Arduino/genJMRIDecoderClient/networking.cpp

### Debug Messages
....
*wm:config portal exiting 
*wm:Starting Web Portal 
....

tablatronix, 2023-01-23:
Bump your wn debugging up. Could be conflict with mqtt client? So it just isnt listening?

tablatronix, 2023-01-25:
Yeah not sure I will be able to reproduce, hoping extra logging shows something obvious

jonasbjurel, 2023-01-27:
Hi, I tried to do this to get more debug output, but I didn't get any more debug info:
#define WM_DEBUG_LEVEL 4
#include <WiFiManager.h>

Any suggestions.
BR/Jonas

tablatronix, 2023-01-28:
you know I don't think i added a method to change logging level real time.. You have to do it via a build flag or global env..
or change it in the .h file.

jonasbjurel, 2023-01-28:
Isnt that what I just did with:
#define WM_DEBUG_LEVEL 4
#include <WiFiManager.h>
?
BR/Jonas

tablatronix, 2023-01-28:
you cannot define in a sketch it doesn't work

tablatronix, 2023-02-09:
Any luck ?

jonasbjurel, 2023-02-13:
Got full logging, troubleshooting as we speak.
Hope to provide feed-back soon.
BR/Jonas

jonasbjurel, 2023-02-14:
My bad, missunderstanding of the function.
I thought that the wifimanager would start an own task for the async operation, so I didn't periodically process() the HTTP requests.
Thanks for the support.

closed 2023-02-14 by jonasbjurel