tzapu/WiFiManager #1558: Entering wrong password to your AP = never ending lock

Jagakatt, 2023-01-31:
### Wrong password in portal and you are stuck.
My first post, and I will probably piss some one off because I missed something. My deepest apologies in advance!

#### Hardware
WiFimanager Branch/Release:  Master(?)   2.0.15-rc.1

Esp8266

Hardware:			ESP-12e, esp01, esp25

Core Version:      	No clue...  Arduindo IDE 2.0.3..?

### Description

Problem description
// When starting up in "portal mode" and you enter the portal, click on "Configure WiFi", select your SSID, and enter THE WRONG PASSWORD.
// Now you are stuck. A reset (button press) does not help. You have to connect to another network, then close it down do get another try.
// Not possible to recover without connecting to another "valid network"

### Settings in IDE, STANDARD

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

// Test, is it a bug if wrong WiFi-AP-password is entered?
// When starting up in "portal mode" and you enter the portal, click on "Configure WiFi", select your SSID, and enter THE WRONG PASSWORD.
// Now you are stuck. A reset (button press) does not help. You have to connect to another network, then close it down do get another try.
// Not possible to recover without connecting to another "valid network"

// Based on: https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/AutoConnectNonBlocking/AutoConnectNonBlocking.ino
// WiFimanager Tzapu 2.0.15-rc.1
// Arduino IDE 2.0
// WEMOS D1 Mini pro ESP8266

#include <WiFiManager.h>  // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once:
  Serial.begin(115200);

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  wm.setConfigPortalBlocking(false);
  wm.setConfigPortalTimeout(60);
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("AutoConnectAP", "wwwwwwww")) {
    Serial.println("connected...yeey :)");
  } else {
    Serial.println("Configportal running");
  }
}

String WiFiString;
String previousWiFiString;

void loop() {
  wm.process();
  // put your main code here, to run repeatedly:

  if (WiFi.status() == WL_IDLE_STATUS) {  // when not connected to a network, but powered on
    WiFiString = "WL_IDLE_STATUS 0";
  } else if (WiFi.status() == WL_NO_SSID_AVAIL) {
    WiFiString = "WL_NO_SSID_AVAIL 1 ";
  } else if (WiFi.status() == WL_SCAN_COMPLETED) {
    WiFiString = "WL_SCAN_COMPLETED 2";
  } else if (WiFi.status() == WL_CONNECTED) {
    WiFiString = "WL_CONNECTED 3";
  } else if (WiFi.status() == WL_CONNECT_FAILED) {
    WiFiString = "WL_CONNECT_FAILED 4";
  } else if (WiFi.status() == WL_CONNECTION_LOST) {
    WiFiString = "WL_CONNECTION_LOST 5 ";
  } else if (WiFi.status() == WL_DISCONNECTED) {
    WiFiString = "WL_DISCONNECTED 6 ";
  }

  if (WiFiString != previousWiFiString) {
    Serial.print("WiFi.status                      : ");
    Serial.println(WiFiString);
    previousWiFiString = WiFiString;
  }
}

```

### Debug Messages

```
/*21:29:22.684 -> {ll��|�l�<�$�c|ǃ��s�#�c��g'�l'o���cp�lrl{lx�n��lĜb'�|���c��gn�dćd`�ng$`'{���o#d�dp�g�;������co�|�c��'o�l`�g'l`o;Ǜ�gc��`s��'c��`�$���d`��g�d*wm:AutoConnect 
21:29:22.877 -> *wm:Connecting to SAVED AP: turbo2
21:29:23.358 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
21:29:26.629 -> *wm:AutoConnect: FAILED 
21:29:26.629 -> *wm:StartAP with SSID:  AutoConnectAP
21:29:28.620 -> *wm:AP IP address: 192.168.4.1
21:29:28.620 -> *wm:Starting Web Portal 
21:29:28.716 -> Configportal running
21:29:28.716 -> WiFi.status                      : WL_IDLE_STATUS 0
21:29:57.195 -> *wm:3 networks found
21:30:41.042 -> *wm:4 networks found
21:30:45.309 -> *wm:4 networks found
21:30:56.658 -> *wm:Connecting to NEW AP: turbo2
21:30:56.787 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
21:31:03.046 -> *wm:[ERROR] Connect to new AP Failed 
21:31:19.638 -> *wm:3 networks found

// Entered the INCORRECT password and saved, and waited...... And nothing. Resetted the device and it comes up in portal mode again. Grrr!

21:32:02.534 -> sd$��|�l�|�d�b|����;�c�c��og�lgn���cp��dsd;lx�o��l��c'�|���c��'o�d��d`�gnl`n{���obd�dp�g�;������co�<�c��o'�$`�no$`gs�ۓo#��`{��o#��`�$Ǐ�$`��g�l*wm:AutoConnect 
21:32:02.728 -> *wm:Connecting to SAVED AP: turbo2
21:32:03.304 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
21:32:09.270 -> *wm:AutoConnect: FAILED 
21:32:09.270 -> *wm:StartAP with SSID:  AutoConnectAP
21:32:11.261 -> *wm:AP IP address: 192.168.4.1
21:32:11.261 -> *wm:Starting Web Portal 
21:32:11.389 -> Configportal running
21:32:11.389 -> WiFi.status                      : WL_IDLE_STATUS 0
21:32:27.628 -> *wm:3 networks found

// Entered the CORRECT password and saved, and waited...... And nothing. Resetted the device and it comes up in portal mode again. Grrr!

21:33:11.574 -> ;ld��|�$�|�d�b<����s�b�#��no�$g'���cp��ds$sdp�g��l��c'�|���b��ng�$��l �no$`'{���ocd�lx�n�{�ܜ���co�<�c��no�d`�no$`'{���gcĒ s��'c��`�$���$`��g�d*wm:AutoConnect 
21:33:11.767 -> *wm:Connecting to SAVED AP: turbo2
21:33:12.246 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
21:33:18.245 -> *wm:AutoConnect: FAILED 
21:33:18.245 -> *wm:StartAP with SSID:  AutoConnectAP
21:33:20.235 -> *wm:AP IP address: 192.168.4.1
21:33:20.235 -> *wm:Starting Web Portal 
21:33:20.363 -> Configportal running
21:33:20.363 -> WiFi.status                      : WL_IDLE_STATUS 0
*/
```

tablatronix, 2023-01-31:
are you SURE you are using that version ? this looks like a very old bug

Jagakatt, 2023-01-31:
I just installed thru the library manager in Arduino IDE.

`
pepe@turbo2:~/Arduino/libraries/WiFiManager$ cat library.properties 
name=WiFiManager
version=2.0.15-rc.1
author=tzapu
maintainer=tablatronix
sentence=WiFi Configuration manager with web configuration portal for Espressif ESPx boards, by tzapu
paragraph=Library for configuring ESP8266/ESP32 modules WiFi credentials and custom parameters at runtime with captive portal.
category=Communication
url=https://github.com/tzapu/WiFiManager.git
architectures=esp8266,esp32
pepe@turbo2:~/Arduino/libraries/WiFiManager$ 

`

Jagakatt, 2023-01-31:
I just installed thru the library manager in Arduino IDE.

pepe@turbo2:~/Arduino/libraries/WiFiManager$ cat library.properties 
name=WiFiManager
version=2.0.15-rc.1
author=tzapu
maintainer=tablatronix
sentence=WiFi Configuration manager with web configuration portal for Espressif ESPx boards, by tzapu
paragraph=Library for configuring ESP8266/ESP32 modules WiFi credentials and custom parameters at runtime with captive portal.
category=Communication
url=https://github.com/tzapu/WiFiManager.git
architectures=esp8266,esp32
pepe@turbo2:~/Arduino/libraries/WiFiManager$

Jagakatt, 2023-02-01:
Seam to be working as it should. Was fooled by the automatic pre-filled-in SSID in the portal. Looked like it was only the password missing to do the connection. Much headache for nothing but really happy it's working. Sorry for my post.

[image]
[image]
[image]

tablatronix, 2023-02-01:
What browser are you using? the placeholder should also be the value and submitted still, thats odd

Jagakatt, 2023-02-01:
Looks like it's the ESP that outputs the SSID in grey if you have a previous successful connection (and now SSID is unavailable).
Or if you have a previous unsuccessful connection attempt with incorrect password.

Well it's easy to go around it by tapping on the correct network in the list when you know it. But at least it fooled me when it looked like it was already in place. ;- )

Do not know what browser-components launched in the phone when accessing "Config portal" from the WiFi-setting.
But it's the same behavior on the following phones:

Xiaomi Redmi Note 8 Pro
Kernel 4.14.141-g95d4269
Model M1906g7g
Android 10 QP1A.190711.020
MIUI Global 12.0.8
Chrome: 109.0.5414.117

Doogee S96Pro
Doogee-s96pro-EEA-Android10.0-20211116
Chrome: 109.0.5414.86

Samsung Galaxy A5
Android 8.0.0

Doogee s98
Android 12

tablatronix, 2023-02-01:
Let me check a few things, if its in grey it should be resbumitted on save, but perhaps not always working, or not being prefilled correctly sometimes?

Project-Exhale, 2023-05-06:
I just encountered the same bug, SSID looks prefilled and when saved, the missing SSID results in "Connect to new AP Failed" and ESP32 is locked until restart.

prochazkapeter, 2023-05-11:
Hello, I have similar issue. When I enter a bad password in the Configuration Portal it goes into endless loop of (xxxxx) wifi:Set status to INIT. When I reset the ESP32, it goes back to this loop. I have  to call resetSettings() to fix this.

Example here:
17:41:41.460 -> *wm:AP IP address: 192.168.4.1
17:41:41.460 -> *wm:Starting Web Portal 
17:41:50.006 -> *wm:20 networks found
17:41:57.518 -> *wm:Connecting to NEW AP: WB12-2G
17:41:57.611 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
17:42:04.740 -> E (31317) wifi:Set status to INIT
17:42:04.817 -> E (31376) wifi:Set status to INIT
17:42:04.817 -> E (31402) wifi:Set status to INIT
x
x
17:42:05.998 -> *wm:Connect Wifi, ATTEMPT # 2 of 33
17:42:05.998 -> *wm:Connecting to NEW AP: WB12-2G
17:42:05.998 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
17:42:05.998 -> E (32601) wifi:Set status to INIT
17:42:06.075 -> E (32636) wifi:Set status to INIT
17:42:06.122 -> E (32704) wifi:Set status to INIT

I would imagine, that the connection should fail due to the bad password and create to configuration portal again.

Thank you :-)

tablatronix, 2023-05-11:
Thanks!
i have no idea what 
17:42:05.998 -> E (32601) wifi:Set status to INIT is...
hmm

chriskinal, 2023-05-11:
I have seen that error as well when I turn off the AP the ESP32 is connected to. I was simulating losing connection to the AP while testing my other issue. It seemed to happen randomly with connection losses > 10 seconds.

I think it is coming from the program that is in the Setup() versus WiFiManager. The errors I get from WiFiManager are all prefixed with "wm:' since that is the name of the WiFiManager instance.

tablatronix, 2023-05-11:
#1540

tablatronix, 2023-05-11:
I pushed a release, made it release. It might have a fix for this already

tablatronix, 2023-05-31:
Has anyone retested this?

pho3nix06, 2023-11-13:
I am having the same issue, in AP mode when a wrong password is entered, the ESP is stuck, and showing "ESP waitForConnectResult...". And after one minute it is showing "[ERROR] Connect to new AP Failed ". 

[image]

Is there any way that I can print the "wrong password" in serial terminal while the config portal keeping open?

tablatronix, 2023-11-15:
Thanks I will test this and see whats going on, sounds like a bug detecting the save as a change.

tablatronix, 2023-11-16:
Not sure how to reproduce this bug, for one, I cannot safe the wrong password, as wm does not save on fail by default..
So thats odd.

If I do use the wrong password, I can still just save the new one and no stuck ??

```php
[WIFI] WIFI_INFO DEBUG
[WIFI] SAVED: YES
[WIFI] SSID: leela
[WIFI] PASS: wrongpassword
*wm:AutoConnect 
*wm:Connecting to SAVED AP: leela
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:AutoConnect: FAILED for  13611 ms
*wm:StartAP with SSID:  WM_AutoConnectAP
*wm:AP IP address: 192.168.4.1
[CALLBACK] configModeCallback fired
*wm:Starting Web Portal 
*wm:5 networks found
*wm:Connecting to NEW AP: leela (RIGHT PASSWORD)
*wm:connectTimeout not set, ESP waitForConnectResult... 
*wm:Connect to new AP [SUCCESS] 
*wm:Got IP Address: 
*wm:192.168.20.254 
[CALLBACK] saveCallback fired
*wm:config portal exiting 
connected...yeey :)
```

tablatronix, 2023-11-16:
You have to enter the SSID, is that what you are talking about? 
A placeholder does not get submitted in the form, you have to fill out the SSID, you cannot just change the password.

pho3nix06, 2023-11-16:
No, the SSID is automatically getting filled by clicking the SSIDs that's been showing after scan, there is no issue about that.
Let me elaborate a bit,

- Lets say the portal is being opened for 5 minutes (**_setConfigPortalTimeout(300))_** and the connect timeout is set for 10 seconds _**(setConnectTimeout(10))**_.

- Now the portal will be opened for 5 minute, and will close either on successful configuration or after timeout (which is 5 minute in this case), and after entering the password, ESP will try to connect for 10 seconds.

- During these 5 minutes if I enter a wrong password in the webpage, the ESP will try to connect for 10 seconds and will fail (ERROR Log message will be printed), so for successful connection I have to close the webpage and open it again.

Now my question is, how can I get an error callback(for entering wrong WiFi credentials) after spending 10 seconds of failed attempt while the config portal is still open (while the stack is still running)? 
In the documentation, there is function _**setAPCallback()**_ which will return on failed WiFi connection attempt. But it is returning after 5 minutes **_(setConfigPortalTimeout())_**.

tablatronix, 2023-11-16:
Thats is not the same issue here, also not sure what you mean close and open the browser again, I suggest you open another issue with this question. If you want to close the portal on failures, you can but you have to set breakafterconfig=true, or use a callback. If it fails and the portal is still open, you can still go back to the previous page

tablatronix, 2023-11-16:
Also what is 'ERROR scan waiting?'

RFDarter, 2023-11-17:
@tablatronix 
Your recent "fix" in WiFIManagaer.cpp line 1817
if(_ssid **=** "" && _pass != ""){

made thing a lot worse. I took me a while to finde this mistake.

I guess you meant to write
if(_ssid **==** "" && _pass != ""){

tablatronix, 2023-11-17:
geez, thats the 2nd time, my bad, been working on 3 projects in 5 different languages

pho3nix06, 2023-11-18:
@tablatronix 
Sure I will open another issue.
Regarding the  **_ERROR scan waiting_** , not sure, I thought this is also from wm. 
Let me look into this and will update if I find anything.

tablatronix, 2023-11-18:
Yeah I definitely have a goal to get the saving flow better worked out but there are so many variables with accounting for captiveportals and ap stability when channel scanning for connects. So getting your info will help me figure out specific use cases

but lets make sure you are actually using this library