tzapu/WiFiManager #1559: A more complete example-file of "AutoConnectNonBlocking"

Jagakatt, 2023-02-01:
### Basic Infos
Hi!
If it would be possible for someone with knowledge (which I do not have) to create a better AutoConnectNonBlocking-example-file ?
It wold be really appreciated and helpful with an example where some code is executed during the "Portal Mode" and also shown how to know when exiting the portal.

I know there are some info here:
https://github.com/tzapu/WiFiManager#save-settings
But I do not manage to get it working. Example files are the best!

Would be really nice to reach the goal to urge someone to connect to WiFi: x with Password: y with a simple message on a display. And Finally a message "Device connected!" or something similar..
[image]

[image]

Searched the web for hours, and all I can find is others who not succeed either. So no code to "borrow" from someone. :-( 
And the "NonBlocking" is needed to be able to toggle between messages if I got it right.

I supply my sketchy sketch which almost gets me there. And no, I do not consider me as a programmer. Haha!.
It must be a smarter way?

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: 

Hardware:			ESP-12e, esp01, esp25

Core Version:      	2.4.0, staging

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp
// WiFiManager in non blocking mode to toggle between messages on display. (Not made by a programmer.....)
// Original sketch: https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/AutoConnectNonBlocking/AutoConnectNonBlocking.ino

#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);  // I2C address 0x27, 16 column and 2 rows connected to SCL --> D1, SDA -> D2
#include <WiFiManager.h>             // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run once:
  Serial.begin(115200);
  lcd.init();  // initialize the lcd
  lcd.backlight();

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  wm.setConfigPortalBlocking(false);
  std::vector<const char *> menu = { "wifi", "restart", "exit" };  // Added by me...
  wm.setMenu(menu);
  wm.setConfigPortalTimeout(60);
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("MyPortal", "password")) {
    Serial.println("connected...yeey)");
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("Connected!!!");
    // } else {
    //   Serial.println("Configportal running");    // Not possible to see when
    //   runningConfigPortal = true;
  }
}

unsigned int currentMillis;
unsigned int startMillis;
unsigned int WiFiErrorCount;
bool portalMode;
bool startingUp = true;

void loop() {
  wm.process();
  // put your main code here, to run repeatedly:

  currentMillis = millis();
  if (currentMillis - startMillis > 2000) {  //test whether the period has elapsed
    String checkMode;
    checkMode = WiFi.softAPIP().toString();  // See if this returns 192.168.4.1 = Portal Mode. Did not manage to find any other way to get notice
    Serial.println(checkMode);               // of when exiting from "PortalMod".
    if (checkMode == "192.168.4.1") {
      Serial.println("###################################### IN PORTAL MODE ################### ");
      portalMode = true;
      startingUp = true;
    } else {
      portalMode = false;
    }
    Serial.print("Every second - WiFi.status is : ");
    Serial.println(WiFi.status());
    if (WiFi.status() == WL_CONNECTED) {  //
      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print("Connected!!!");
      WiFiErrorCount = 0;
      startingUp = false;
    } else if ((WiFi.status() != WL_CONNECTED) && (portalMode == false)) {
      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print("NOT Connected!!!");
      lcd.setCursor(0, 1);
      lcd.print("Restarting.");
      lcd.print(WiFiErrorCount);
      WiFiErrorCount++;
      if ((WiFiErrorCount > 100) && (startingUp == false)) {  // Restart device after about 200 sec if no WiFi.
        ESP.restart();
      }
    }
    startMillis = currentMillis;
  }

  if ((WiFi.status() != WL_CONNECTED) && (portalMode == true)) {  // Show configuration messages on display if no WiFi and in PortalMode.
    portalDisplayMessages();
  }

  if (WiFi.status() == WL_CONNECTED) {
    // futureCrappyCode();      // Code dependent on an active WiFi connection.
  }
}

void portalDisplayMessages() {
  int currentMillis;
  static unsigned int startMillis;
  static unsigned int messageCount = 0;

  currentMillis = millis();
  if (currentMillis - startMillis > 10000) {  //test whether the period has elapsed
    Serial.println("Running portal,  10 seconds message switch");

    switch (messageCount) {
      case 0:
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print("AccessPoint:");
        lcd.setCursor(0, 1);
        lcd.print("MyPortal");
        messageCount++;
        break;

      case 1:
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print("Password:");
        lcd.setCursor(0, 1);
        lcd.print("gggggggg");
        messageCount = 0;
    }
    startMillis = currentMillis;
  }
}

```

### Debug Messages

```
messages here
```

Jagakatt, 2023-02-01:
Thanks for asking! Do almost get a little Christmas feeling when I can which for something! :-)
It would be nice for normal non nerdy people to complete the "ConfigPortal" which I have realized watching normal people in action can be a bit confusing. 
So be able get some trigger to react upon when:

* No previous/known network available (less important)
* Portal Started (important)
* Wrong WiFi-password  (would be really good)
* Portal exited (Important)
* Successfully connected (important) but probably the same as (WiFi.status() == WL_CONNECTED)

To be able to get some trigger of the above listed situation would at least enable me to create some nice assistance for the enduser. And I assume the NonLocking mode is the way be able to run code to toggle between messages on small displays when in portal-mode. 
Thanks!

tablatronix, 2023-02-01:
I assume you know about the existingones already ?

```
//callbacks
  // called after AP mode and config portal has started
  //  setAPCallback( std::function<void(WiFiManager*)> func );
  //  
  // called after webserver has started
  //  setWebServerCallback( std::function<void()> func );
  //  
  // called when settings reset have been triggered
  //  setConfigResetCallback( std::function<void()> func );
  //  
  // called when wifi settings have been changed and connection was successful ( or setBreakAfterConfig(true) )
  //  setSaveConfigCallback( std::function<void()> func );
  //  
  // called when saving either params-in-wifi or params page
  //  setSaveParamsCallback( std::function<void()> func );
  //  
  // called when saving params-in-wifi or params before anything else happens (eg wifi)
  //  setPreSaveConfigCallback( std::function<void()> func );
  //  
  // called just before doing OTA update
  //  setPreOtaUpdateCallback( std::function<void()> func );
```

tablatronix, 2023-02-02:
heres a crappy demo, do not use this code, just an example.
in reality you should be setting states in these callbacks and NOT actually doing stuff.

Then do stuff based on those states

```c++
// Liquidcrystal demo
// WiFiManager in non blocking mode to toggle between messages on display. (Not made by a programmer.....)
// Original sketch: https://github.com/tzapu/WiFiManager/blob/master/examples/NonBlocking/AutoConnectNonBlocking/AutoConnectNonBlocking.ino

#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x25, 16, 2);  // I2C address 0x27, 16 column and 2 rows connected to SCL --> D1, SDA -> D2
#include <WiFiManager.h>             // https://github.com/tzapu/WiFiManager
WiFiManager wm;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run oQnce:
  Serial.begin(115200);
  Serial.setDebugOutput(false);

  Wire.setClock(400000L);  // set i2c speed 400khz
  Wire.begin();  
  lcd.init();  // initialize the lcd
  lcd.backlight();

  lcd.clear();
  lcd.setCursor(0, 0);
  lcd.print("Connecting to wifi .....");

  //reset settings - wipe credentials for testing
  wm.resetSettings();

  wm.setAPCallback(configModeCallback);
  // wm.setWebServerCallback(bindServerCallback);
  wm.setSaveConfigCallback(saveWifiCallback);
  // wm.setSaveParamsCallback(saveParamCallback);
  // wm.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);

  wm.setConfigPortalBlocking(false);
  std::vector<const char *> menu = { "wifi", "restart", "exit" };  // Added by me...
  wm.setMenu(menu);
  wm.setConfigPortalTimeout(60);
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("MyPortal", "password")) {
    Serial.println("connected...yeey)");
    lcd.clear();
    lcd.setCursor(0, 0);
    lcd.print("Connected!!!");
    // } else {
    //   Serial.println("Configportal running");    // Not possible to see when
    //   runningConfigPortal = true;
  }
}

void saveWifiCallback(){
  Serial.println("[CALLBACK] saveCallback fired");
      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print("Saving new wifi....");
      delay(2000);
}

//gets called when WiFiManager enters configuration mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("[CALLBACK] configModeCallback fired");
  // myWiFiManager->setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); 
  // Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  // Serial.println(myWiFiManager->getConfigPortalSSID());
  // 
  // esp_wifi_set_bandwidth(WIFI_IF_AP, WIFI_BW_HT20);
  lcd.clear();
  lcd.setCursor(0, 0); 
  lcd.print("NO WiFi, Connect to AP: " + wm.getConfigPortalSSID());
  delay(1000);
}

void checkPortalState(){
    bool cp = wm.getConfigPortalActive();
    bool wp = wm.getWebPortalActive();
      lcd.clear();
      lcd.setCursor(0, 0);      
    if(cp){
      lcd.print("Portal Running...");
      delay(1000);
    } else {
      lcd.print("Portal Closed!");
      delay(1000);
    }
}

unsigned int currentMillis;
unsigned int startMillis;
unsigned int WiFiErrorCount;
bool portalMode;
bool startingUp = true;

void loop() {
  wm.process();
  // put your main code here, to run repeatedly:

  currentMillis = millis();
  if (currentMillis - startMillis > 2000) {  //test whether the period has elapsed
    String checkMode;
    checkMode = WiFi.softAPIP().toString();  // See if this returns 192.168.4.1 = Portal Mode. Did not manage to find any other way to get notice
    Serial.println(checkMode);               // of when exiting from "PortalMod".
    if (checkMode == "192.168.4.1") {
      Serial.println("###################################### IN PORTAL MODE ################### ");
      portalMode = true;
      startingUp = true;
    } else {
      portalMode = false;
    }
    Serial.print("Every second - WiFi.status is : ");
    Serial.println(WiFi.status());
    if (WiFi.status() == WL_CONNECTED) {  //
      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print("Connected!!!");
      WiFiErrorCount = 0;
      startingUp = false;
    } else if ((WiFi.status() != WL_CONNECTED) && (portalMode == false)) {
      lcd.clear();
      lcd.setCursor(0, 0);
      lcd.print("NOT Connected!!!");
      lcd.setCursor(0, 1);
      lcd.print("Restarting.");
      lcd.print(WiFiErrorCount);
      WiFiErrorCount++;
      if ((WiFiErrorCount > 100) && (startingUp == false)) {  // Restart device after about 200 sec if no WiFi.
        ESP.restart();
      }
    }
    startMillis = currentMillis;
  }

  checkPortalState();

  if ((WiFi.status() != WL_CONNECTED) && (portalMode == true)) {  // Show configuration messages on display if no WiFi and in PortalMode.
    portalDisplayMessages();
  }

  if (WiFi.status() == WL_CONNECTED) {
    // futureCrappyCode();      // Code dependent on an active WiFi connection.
  }
  delay(1000);
}

void portalDisplayMessages() {
  int currentMillis;
  static unsigned int startMillis;
  static unsigned int messageCount = 0;

  currentMillis = millis();
  if (currentMillis - startMillis > 10000) {  //test whether the period has elapsed
    Serial.println("Running portal,  10 seconds message switch");

    switch (messageCount) {
      case 0:
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print("AccessPoint:");
        lcd.setCursor(0, 1);
        lcd.print("MyPortal");
        messageCount++;
        break;

      case 1:
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print("Password:");
        lcd.setCursor(0, 1);
        lcd.print("gggggggg");
        messageCount = 0;
    }
    startMillis = currentMillis;
  }
}
```

tablatronix, 2023-02-02:
[image]

Jagakatt, 2023-02-02:
> I assume you know about the existingones already ?
>
> ```
> //callbacks
>   // called after AP mode and config portal has started
>   //  setAPCallback( std::function<void(WiFiManager*)> func );
>   //
>   // [...]

Sorry, I did not know of the existing ones. And even less how to put them in action. It's just a lot of magic lines of code that unfortunately makes no sense to me.
Played around earlier with what's found under "Callbacks" on the WiFiManager main page here, but without success.
I have only programmed for a few months so I am still at the level where I get totally chocked every time the compilation goes through without a few pages of red text...

But I definitely think I will manage to progress further thanks to the pieces of code you added to my sketch!!
I will upload a cleaner version here in a few days.
THANK YOU SO MUCH!!!

Jagakatt, 2023-02-22:
// A sketchy sketch to show messages on a LCD1602 when WiFiManager is in "non blocking mode"
// Messages about network status and WiFimanager configuration portal.
// And nope, I do not consider myself as a skilled programmer, but after scanning the Internet I realized I am not the only one not managing to execute code when the portal was active.
// So hopefully it can help someone.
// At least what's found below fulfilled my requirements, even though there are probably a zillion ways to do it in a better way, so feel free to modify! :-)
// Fredrik

// To be added somehow: A message when wrong network credentials were entered.
// Looks like "saveWifiCallback" only is executed when proper network credentials were entered.
// But no big deal. :-)

// Setup LCD1602 with I2C-interface
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);  // I2C address 0x27, 16 column and 2 rows connected to SCL --> D1, SDA -> D2
#include <WiFiManager.h>             // https://github.com/tzapu/WiFiManager

WiFiManager wm;

bool starting;
bool newWIFIsuccess = false;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run oQnce:
  Serial.begin(115200);
  Serial.setDebugOutput(true);  //
  lcd.init();                   // initialize the lcd
  lcd.backlight();

  starting = true;
  displayWiFimanagerMessages();  // Just to show the initial message.

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  wm.setAPCallback(configModeCallback);
  // wm.setWebServerCallback(bindServerCallback);
  wm.setSaveConfigCallback(saveWifiCallback);
  // wm.setSaveParamsCallback(saveParamCallback);
  // wm.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);

  wm.setConfigPortalBlocking(false);                               
  std::vector<const char *> menu = { "wifi", "restart", "exit" };  // Limit the portal buttuns to these three.
  wm.setMenu(menu);                                                
  wm.setConfigPortalTimeout(120);                                 
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("MyPortal", "MyPassword")) {
    Serial.println("connected...yeey)");
    // lcd.clear();
    // lcd.setCursor(0, 0);
    // lcd.print("Connected!!!");
     } else {

  }
}

void loop() {
  wm.process();  // WiFimanager to be executed frequently.
  displayWiFimanagerMessages();
}

void displayWiFimanagerMessages() {
  static unsigned int messageCount = 0;
  static unsigned long previousMillis = 0;  // To run directly (== 0) when called from "Setup"
  if (millis() - previousMillis > 3000 || previousMillis == 0) {   // Toggle between messages every 3s.
    previousMillis = millis();

    //Serial.println(wm.getWiFiPass());  //  <-- Will get you the stored Passwd
    //Serial.println(wm.getWiFiSSID());  //  <-- Will get you the stored SSID
[... 112 lines cut ...]

void saveWifiCallback() {                                                            // Executed when saving new network credentials.
  Serial.println("[CALLBACK] saveCallback fired. Saving new network credentials.");  /// Only seam to be execuded when saving valid network credentials.

  newWIFIsuccess = true;  // To trigger message when proper network credentials have been entered.

  // lcd.clear();
  // lcd.setCursor(0, 0);
  // lcd.print("Saving new wifi....");
  // delay(2000);
}

void checkPortalState() {
  // bool cp = wm.getConfigPortalActive();
  // bool wp = wm.getWebPortalActive();
  // lcd.clear();
  // lcd.setCursor(0, 0);
  // if (cp) {
  //   lcd.print("Portal Running...");
  //   delay(1000);
  // } else {
  //   lcd.print("Portal Closed!");
  //   delay(1000);
  // }
}

Jagakatt, 2023-02-22:
And no, I have no clue of why not everything below is seen as "code" :-(

`
// A sketchy sketch to show messages on a LCD1602 when WiFiManager is in "non blocking mode"
// Messages about network status and WiFimanager configuration portal.
// And nope, I do not consider myself as a skilled programmer, but after scanning the Internet I realized I am not the only one not managing to execute code when the portal was active.
// So hopefully it can help someone.
// At least what's found below fulfilled my requirements, even though there are probably a zillion ways to do it in a better way, so feel free to modify! :-)
// Fredrik

// To be added somehow: A message when wrong network credentials were entered.
// Looks like "saveWifiCallback" only is executed when proper network credentials were entered.
// But no big deal. :-)

// Setup LCD1602 with I2C-interface
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);  // I2C address 0x27, 16 column and 2 rows connected to SCL --> D1, SDA -> D2
#include <WiFiManager.h>             // https://github.com/tzapu/WiFiManager

WiFiManager wm;

bool starting;
bool newWIFIsuccess = false;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run oQnce:
  Serial.begin(115200);
  Serial.setDebugOutput(true);  //
  lcd.init();                   // initialize the lcd
  lcd.backlight();

  starting = true;
  displayWiFimanagerMessages();  // Just to show the initial message.

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  wm.setAPCallback(configModeCallback);
  // wm.setWebServerCallback(bindServerCallback);
  wm.setSaveConfigCallback(saveWifiCallback);
  // wm.setSaveParamsCallback(saveParamCallback);
  // wm.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);

  wm.setConfigPortalBlocking(false);
  std::vector<const char *> menu = { "wifi", "restart", "exit" };  // Limit the portal buttuns to these three.
  wm.setMenu(menu);
  wm.setConfigPortalTimeout(120);
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("MyPortal", "MyPassword")) {
    Serial.println("connected...yeey)");
    // lcd.clear();
    // lcd.setCursor(0, 0);
    // lcd.print("Connected!!!");
  } else {
  }
}

void loop() {
  wm.process();  // WiFimanager to be executed frequently.
  displayWiFimanagerMessages();
}

void displayWiFimanagerMessages() {
  static unsigned int messageCount = 0;
  static unsigned long previousMillis = 0;                        // To run directly (== 0) when called from "Setup"
  if (millis() - previousMillis > 3000 || previousMillis == 0) {  // Toggle between messages every 3s.
    previousMillis = millis();

[... 115 lines cut ...]
void saveWifiCallback() {                                                            // Executed when saving new network credentials.
  Serial.println("[CALLBACK] saveCallback fired. Saving new network credentials.");  /// Only seam to be execuded when saving valid network credentials.

  newWIFIsuccess = true;  // To trigger message when proper network credentials have been entered.

  // lcd.clear();
  // lcd.setCursor(0, 0);
  // lcd.print("Saving new wifi....");
  // delay(2000);
}

void checkPortalState() {
  // bool cp = wm.getConfigPortalActive();
  // bool wp = wm.getWebPortalActive();
  // lcd.clear();
  // lcd.setCursor(0, 0);
  // if (cp) {
  //   lcd.print("Portal Running...");
  //   delay(1000);
  // } else {
  //   lcd.print("Portal Closed!");
  //   delay(1000);
  // }
}
`

Jagakatt, 2023-02-22:
```

// A sketchy sketch to show messages on a LCD1602 when WiFiManager is in "non blocking mode"
// Messages about network status and WiFimanager configuration portal.
// And nope, I do not consider myself as a skilled programmer, but after scanning the Internet I realized I am not the only one not managing to execute code when the portal was active.
// So hopefully it can help someone.
// At least what's found below fulfilled my requirements, even though there are probably a zillion ways to do it in a better way, so feel free to modify! :-)
// Fredrik

// To be added somehow: A message when wrong network credentials were entered.
// Looks like "saveWifiCallback" only is executed when proper network credentials were entered.
// But no big deal. :-)

// Setup LCD1602 with I2C-interface
#include <LiquidCrystal_I2C.h>
LiquidCrystal_I2C lcd(0x27, 16, 2);  // I2C address 0x27, 16 column and 2 rows connected to SCL --> D1, SDA -> D2
#include <WiFiManager.h>             // https://github.com/tzapu/WiFiManager

WiFiManager wm;

bool starting;
bool newWIFIsuccess = false;

void setup() {
  WiFi.mode(WIFI_STA);  // explicitly set mode, esp defaults to STA+AP
  // put your setup code here, to run oQnce:
  Serial.begin(115200);
  Serial.setDebugOutput(true);  //
  lcd.init();                   // initialize the lcd
  lcd.backlight();

  starting = true;
  displayWiFimanagerMessages();  // Just to show the initial message.

  //reset settings - wipe credentials for testing
  //wm.resetSettings();

  wm.setAPCallback(configModeCallback);
  // wm.setWebServerCallback(bindServerCallback);
  wm.setSaveConfigCallback(saveWifiCallback);
  // wm.setSaveParamsCallback(saveParamCallback);
  // wm.setPreOtaUpdateCallback(handlePreOtaUpdateCallback);

  wm.setConfigPortalBlocking(false);
  std::vector<const char *> menu = { "wifi", "restart", "exit" };  // Limit the portal buttuns to these three.
  wm.setMenu(menu);
  wm.setConfigPortalTimeout(120);
  //automatically connect using saved credentials if they exist
  //If connection fails it starts an access point with the specified name
  if (wm.autoConnect("MyPortal", "MyPassword")) {
    Serial.println("connected...yeey)");
    // lcd.clear();
    // lcd.setCursor(0, 0);
    // lcd.print("Connected!!!");
  } else {
  }
}

void loop() {
  wm.process();  // WiFimanager to be executed frequently.
  displayWiFimanagerMessages();
}

void displayWiFimanagerMessages() {
  static unsigned int messageCount = 0;
  static unsigned long previousMillis = 0;                        // To run directly (== 0) when called from "Setup"
  if (millis() - previousMillis > 3000 || previousMillis == 0) {  // Toggle between messages every 3s.
    previousMillis = millis();

    //Serial.println(wm.getWiFiPass());  //  <-- Will get you the stored Passwd
[... 114 lines cut ...]
void saveWifiCallback() {                                                            // Executed when saving new network credentials.
  Serial.println("[CALLBACK] saveCallback fired. Saving new network credentials.");  /// Only seam to be execuded when saving valid network credentials.

  newWIFIsuccess = true;  // To trigger message when proper network credentials have been entered.

  // lcd.clear();
  // lcd.setCursor(0, 0);
  // lcd.print("Saving new wifi....");
  // delay(2000);
}

void checkPortalState() {
  // bool cp = wm.getConfigPortalActive();
  // bool wp = wm.getWebPortalActive();
  // lcd.clear();
  // lcd.setCursor(0, 0);
  // if (cp) {
  //   lcd.print("Portal Running...");
  //   delay(1000);
  // } else {
  //   lcd.print("Portal Closed!");
  //   delay(1000);
  // }
}
```

tablatronix, 2023-02-23:
Is this a working update?

Jagakatt, 2023-02-23:
Yes, it does work. But keep in mind that I'm new to coding.

tzapu, 2023-02-23:
maybe it helps, here s a sketch i ve used long ago that uses a tiny oled and displays connection prompt/statuses, ota statuses and so on. it illustrates callback usage. credits to all the people that made the original
```
/**The MIT License (MIT)

Copyright (c) 2016 by Daniel Eichhorn

Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to deal
in the Software without restriction, including without limitation the rights
to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
copies of the Software, and to permit persons to whom the Software is
furnished to do so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.

See more at http://blog.squix.ch
*/

/* Customizations by Neptune (NeptuneEng on Twitter, Neptune2 on Github)
 *  
 *  Added Wifi Splash screen and credit to Squix78
 *  Modified progress bar to a thicker and symmetrical shape
 *  Replaced TimeClient with built-in lwip sntp client (no need for external ntp client library)
 *  Added Daylight Saving Time Auto adjuster with DST rules using simpleDSTadjust library
 *  https://github.com/neptune2/simpleDSTadjust
 *  Added Setting examples for Boston, Zurich and Sydney
  *  Selectable NTP servers for each locale
  *  DST rules and timezone settings customizable for each locale
   *  See https://www.timeanddate.com/time/change/ for DST rules
  *  Added AM/PM or 24-hour option for each locale
 *  Changed to 7-segment Clock font from http://www.keshikan.net/fonts-e.html
 *  Added Forecast screen for days 4-6 (requires 1.1.3 or later version of esp8266_Weather_Station library)
 *  Added support for DHT22, DHT21 and DHT11 Indoor Temperature and Humidity Sensors
 *  Fixed bug preventing display.flipScreenVertically() from working
 *  Slight adjustment to overlay
 */

#include <ESP8266WiFi.h>
#include <Ticker.h>
#include "settings.h"
#include <JsonListener.h>
#include <ArduinoOTA.h>
#include <ESP8266mDNS.h>
#include <time.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BMP280.h>
#include <PubSubClient.h>        //https://github.com/Imroy/pubsubclient

#include "WundergroundClient.h"
#include "WeatherStationFonts.h"
#include "WeatherStationImages.h"
#include "DSEG7Classic-BoldFont.h"
#include "ThingspeakClient.h"

Adafruit_BMP280 bme;

WiFiClient wclient;
PubSubClient mqttClient(wclient);
[... 419 lines cut ...]
}
*/

// converts the dBm to a range between 0 and 100%
int8_t getWifiQuality() {
  int32_t dbm = WiFi.RSSI();
  if(dbm <= -100) {
      return 0;
  } else if(dbm >= -50) {
      return 100;
  } else {
      return 2 * (dbm + 100);
  }
}

void setReadyForWeatherUpdate() {
  Serial.println("Setting readyForUpdate to true");
  readyForWeatherUpdate = true;
}

void setReadyForIndoorUpdate() {
  Serial.println("Setting setReadyForIndoorUpdate to true");
  readyForIndoorUpdate = true;
}
```