tzapu/WiFiManager #1563: Getting stuck in an infinite loop with wrong password/SSID

eaabk, 2023-02-15:
WiFimanager Branch/Release:  Master

Esp8266/Esp32:  ESP32

Hardware:			espressif32

### Code

`void setup() {

WiFi.mode(WIFI_STA);
WiFiManager wm;
bool res;

wm.setConnectRetries(3);
wm.setConnectTimeout(120);
wm.setConfigPortalTimeout(180);

res = wm.autoConnect("xxx");

  if (!res)
  {
    Serial.println("Failed to connect, Restarting");
    delay(1000);
    wm.resetSettings();
    ESP.restart();

    delay(3000);
  }
  else
  {
    // if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

}`

### Debug Messages
`*wm:AutoConnect
*wm:Connect Wifi, ATTEMPT # 1 of 3
*wm:No wifi saved, skipping
*wm:Connect Wifi, ATTEMPT # 2 of 3
*wm:No wifi saved, skipping
*wm:Connect Wifi, ATTEMPT # 3 of 3
*wm:No wifi saved, skipping
*wm:AutoConnect: FAILED
*wm:StartAP with SSID:  xxxxx
*wm:AP IP address: 192.168.4.1
*wm:Starting Web Portal
[ 22957][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 24108][E][WebServer.cpp:649] _handleRequest(): request handler not found
[ 24187][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:24 networks found
[ 25885][E][WebServer.cpp:649] _handleRequest(): request handler not found
*wm:Connect Wifi, ATTEMPT # 1 of 3
*wm:Connecting to NEW AP: ABC
*wm:connectTimeout not set, ESP waitForConnectResult... 
E (44042) wifi:Association refused temporarily, comeback time 649 mSec
E (47785) wifi:Association refused temporarily, comeback time 649 mSec
E (51533) wifi:Association refused temporarily, comeback time 649 mSec
E (90596) wifi:Association refused temporarily, comeback time 649 mSec
*wm:Connect Wifi, ATTEMPT # 4 of 43
*wm:Connecting to NEW AP: ABC
*wm:connectTimeout not set, ESP waitForConnectResult...
E (102357) wifi:Association refused temporarily, comeback time 649 mSec
E (106122) wifi:Association refused temporarily, comeback time 649 mSec
E (109874) wifi:Association refused temporarily, comeback time 649 mSec
*wm:Connect Wifi, ATTEMPT # 5 of 55
*wm:Connecting to NEW AP: ABC
*wm:connectTimeout not set, ESP waitForConnectResult... 
E (121572) wifi:Association refused temporarily, comeback time 649 mSec
E (125318) wifi:Association refused temporarily, comeback time 649 mSec
E (129054) wifi:Association refused temporarily, comeback time 649 mSec
E (132792) wifi:Association refused temporarily, comeback time 649 mSec
E (136574) wifi:Association refused temporarily, comeback time 649 mSec
E (140291) wifi:Association refused temporarily, comeback time 649 mSec
E (144004) wifi:Association refused temporarily, comeback time 649 mSec
E (147746) wifi:Association refused temporarily, comeback time 649 mSec
E (151477) wifi:Association refused temporarily, comeback time 649 mSec
E (155194) wifi:Association refused temporarily, comeback time 649 mSec
E (155854) wifi:Association refused temporarily, comeback time 649 mSec
E (159586) wifi:Association refused temporarily, comeback time 649 mSec
E (163299) wifi:Association refused temporarily, comeback time 649 mSec
E (167015) wifi:Association refused temporarily, comeback time 649 mSec
E (167676) wifi:Association refused temporarily, comeback time 649 mSec
E (171420) wifi:Association refused temporarily, comeback time 649 mSec
E (172078) wifi:Association refused temporarily, comeback time 649 mSec
E (175808) wifi:Association refused temporarily, comeback time 649 mSec`
```
### Notes
If i enter the SSID or password, I get stuck in an infinitte loop. My understanding is that the  wm.setConnectTimeout(120); should timeout the autoconnect after 120 seconds making it return a false which with the if condition in my code would reset the setting and reset the board but never does happen. 
I have checked the rest of the code for delays. 

Appreciate the help. 

```

tablatronix, 2023-02-17:
I just started seeing this in esp 2.0.6 this is not in wm, it will eventually timeout. It is stuck in autoreconnect loop because of the unchecked failure mode, same happens with wrong password and handshake failures no reason why some routers do this I think its a fundamental flaw in some of them

tablatronix, 2023-02-17:
[image]