tzapu/WiFiManager #1565: ESP32-S3 custom board not storing the WiFi credentials after restart

Sireevenkat, 2023-02-20:
Hi,
I am using ESP32-S3 custom board with Arduino IDE. In my project I am storing the Wi-Fi credentials in the ESP32-S3 custom board but after restarting the board Board not giving stored credentials. I checked same with ESP32-S3 development board and Its storing the given credentials after restarting the board. Can you suggest the solution for this. Due to what issue it will happen? What I have to change in the Software to make this work.

I am using WiFiManager library to give the WiFi credentials to the board.

Thanks

sabas1080, 2023-02-28:
check that it is the same version of esp32-s3 as the one on the development board, I am using credential storage in S3 and it works

Amtho, 2023-06-15:
Im getting thesame issue with this module:
https://www.digikey.de/de/products/detail/espressif-systems/ESP32-S3-WROOM-2-N32R8V/15970964

tablatronix, 2023-06-15:
Got any logs? Is it actually saving credentials?

tablatronix, 2023-06-15:
Works for me
[image]
[image]
[image]