tzapu/WiFiManager #1575: Does not switch between wifi and ap automatically

hackintoshlover, 2023-03-23:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master

Esp8266/Esp32: ESP32

Hardware: ESP32-Wroom

Core Version:  2.0.6

### Description

I am using the example sketch Auto-Connect Non Blocking

### Settings in IDE

Additional libraries: None

Problem:
After uploading the code, the esp starts in ap mode and i can access the web portal to connect it to wifi router.
But once the wifi router goes off, the device does not automatically enter into ap mode or connect to other router whose credentials are stored in it.
How do i resolve the issue?

tablatronix, 2023-03-23:
ok well esp should automatically do autoreconnect, but it seems to not work very well.
Also we can only store 1 set of credentials, otherwise you have to use multiwifi which wm does not support at this time.

hackintoshlover, 2023-03-23:
Would be really helpful if you could implement it.
There's a

closed 2023-03-23 by hackintoshlover

tablatronix, 2023-03-24:
So you want a secondary wifi ap to fallback to? This is not really an issue as much now that mesh is a thing..

Esp supports the basic wifi creds easily but adding another is a fair amount of code. I will look into it though