tzapu/WiFiManager #1578: Question: Saving Static IP

lmcbmai, 2023-03-28:
Hi, 
I am very new to WiFiManager and having troubles understanding whether static IP/Gateway/Netmask/DNS values are saved automatically after exiting the portal, or I need to write a code to make them survive power cycle or restart of ESP8266. I tried many code examples published here and I am seeing that SSID and Password values are saved but IPs are not. If additional steps are required to save IPs, I'd appreciate if somebody shares a code snippet or a link to an example showing how to do it.
Thanks

Bighoneypot, 2023-03-29:
Do you want to configure your device without dhcp?
That is, do you want to manually configure the IP address etc etc?

Bighoneypot, 2023-03-29:
You need similar this?
[image]

lmcbmai, 2023-03-29:
I want to configure device with static IP. My observation is that unlike SSID and Password, the values of Static IP and other 3 related fields are not saved; they are lost after power cycling or resetting the microcontroller. 
Yesterday I wrote a code saving these 4 values in a file in LittleFS and loading them on power up. It is working fine now, and my only remaining question is - do I really need to write my own code to save static IP and related fields, or WiFiManager can do it and I simply do not know how to enable/call this feature.
Thanks

lmcbmai, 2023-03-29:
@delugeia Thanks for your suggestions. That's the right (professional) way of doing things and I am all for it. Unfortunately in some cases that is not possible. For example I have 5G based wireless Internet and 5G modem I got from my ISP is integrated with WiFi router that has extremely limited list of configurable options. I cannot even set the local subnet of my choice, leave alone assigning static IPs to particular clients. Since there is no way to change the netmask, I reduced DHCP pool and using upper IPs outside pool range as static. Adding another WiFi router would be an option, but I don't want to overcomplicate things if not absolutely necessary.

tablatronix, 2023-03-30:
Wm does not save anything to any part of flash by design

closed 2023-03-30 by tablatronix

lmcbmai, 2023-04-01:
@tablatronix  Thanks for clarifying, but than do you know by any chance which other library saves SSID and Password? They somehow survive restarts nevertheless I am not doing anything to save and restore them.

tablatronix, 2023-04-01:
Credentials are saved by the esp library wm doesnt have to handle that

LewiLim, 2023-06-13:
> You need similar this? [image]

Hi, is there an examples like this

tablatronix, 2023-06-13:
I believe there is one in the examples with spiffs, there is a litttlefs example but it does not include params

debbyagestira, 2024-04-03:
@Bighoneypot, @tablatronix Hello, how do I make those fields blank?

I have tried to delete the hardcoded values like this:
```char static_ip[16];
char static_gw[16];
char static_sn[16];```
or
```char static_ip[16] = "';
char static_gw[16] = "';
char static_sn[16] = "";```
but the fields/forms is gone..
>before
![IMG_20240403_090854](https://github.com/tzapu/WiFiManager/assets/73832877/4a4cdab2-5335-40bc-a6e9-fdcd678b7ea5)
>after
![IMG_20240403_090906](https://github.com/tzapu/WiFiManager/assets/73832877/037dd201-f111-4219-b636-f95c61572f7d)

> You need similar this? ![image](https://user-images.githubusercontent.com/35570068/228508520-be62349e-f7ea-4954-9e8c-c17fbb09c85d.png)