tzapu/WiFiManager #1579: Web Server doesn't starts after Wifi connected through WiFiManager

patelabhi23, 2023-03-29:
I am using ESP32 Wroom DA module. When I connect to the WiFi Manager AP & then save new WiFi credentials then the Web Server doesn't starts. It gets the IP & Internet Connection but when I open WebServer HTTTP404 error. 

Currently, if I do hard reset of the ESP then the WebServer starts working, but instead of Hard Reset on ESP I would like to do with the code itself.

Bighoneypot, 2023-03-29:
@patelabhi23 can you paste your full code and write your the result you expected?

patelabhi23, 2023-03-29:
@Bighoneypot yes here is the code:

```
#include <WiFiManager.h>
WiFiManager wm;
#include "ESPAsyncWebServer.h"

AsyncWebServer server ( 80 );

void setup() {
  Serial.begin(9600);

  wm.resetSettings();

  bool res;
  res = wm.autoConnect("Abc", "123"); 

  if (!res) {
    Serial.println("Failed to connect");
  }
  else {
    Serial.println("connected...yeey :)");
    ESP.restart();
  }

}

void beginServer()
{
  server.reset();

  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(200, "text/html", webPage());
    delay(1000);
    ESP.restart();
  });
  server.begin();
  Serial.println ( "HTTP server started" );
}

void loop() {
  Serial.println(WiFi.localIP());
  beginServer();
  delay(5000);
}

String webPage() {
  String ptr = "<!DOCTYPE html> <html>\n";
  ptr += "<body>\n";
  ptr += "<p>Hello World</p>";
  ptr += "</body>\n";
  ptr += "</html>\n";
  return ptr;
}
```

Once the Wifi is configured then it prints the IP address but when I open the webpage using that IP it shows that "This site can't be reached"

[image]

tablatronix, 2023-03-29:
There is a problem in esp with reopening a port for example opening another webserver even after one is closed. There is an issue we never figured it out

tablatronix, 2023-03-29:
Yeah you cannot run 2 webservers even if you close one the port is not free have to reatart

patelabhi23, 2023-03-29:
So you mean to say that wifimanager library doesn't supports two webservers?

Bighoneypot, 2023-03-29:
> Yeah you cannot run 2 webservers even if you close one the port is not free have to reatart

This is the problem, you can't run 2 webservers @patelabhi23

patelabhi23, 2023-03-29:
@Bighoneypot I am closing the same port & again reopening it then too why it isn't working. Also you have the option of multiple ports in ESP32 & I am running multiple ports by saving the Wifi credentials in the code, but once I integrate wifimanager everything stops out

tablatronix, 2023-03-29:
Its an ESP BUG! ports are not released, run it on a different port I bet it works

patelabhi23, 2023-03-29:
@tablatronix 
1. could you suggest any solution on how to solve that esp bug or even how to debug that?
2. I have already added the esp restart in the snippet above but then too it isn't working, so any alternatives for that?

Kinuseka, 2023-03-30:
The logical way is to not make WiFiManager run on port 80, but on other uncommon ports.

tablatronix, 2023-03-31:
Its deep inside the wifiserver library, I am not very familiar with it but I never figured out or even got anywhere.

There might be a conflict between async and regular webserver libraries also , have you tried running both without wm?
I am working on the async branch to get it switchable

patelabhi23, 2023-03-31:
Yes when I run without wm both of them works well but on integrating with wm it causes issue.

Also the esp.restart isn't working in that case so any alternatives for that?

patelabhi23, 2023-05-01:
@tablatronix as per this were you able to make the async branch switchable?

> Its deep inside the wifiserver library, I am not very familiar with it but I never figured out or even got anywhere.
> 
> There might be a conflict between async and regular webserver libraries also , have you tried running both without wm? I am working on the async branch to get it switchable

tablatronix, 2023-05-01:
No but I will try to get some traction on it this week and try to get some releases out

SVLoneStar, 2023-05-08:
Any news on this?
Alternatively...is there a way to start WifiManager listening on a port other than 80? Thx!

patelabhi23, 2023-05-10:
No not any yet as the asyncwebserver trys to reopen the same port but it isn't being able to do so.

> Any news on this? Alternatively...is there a way to start WifiManager listening on a port other than 80? Thx!

patelabhi23, 2023-05-10:
Hey @tablatronix  any updates?

> No but I will try to get some traction on it this week and try to get some releases out

patelabhi23, 2023-05-10:
@tablatronix  any updates?

> No but I will try to get some traction on it this week and try to get some releases out

tablatronix, 2023-05-11:
#1531

closed 2023-05-11 by tablatronix

kjyv, 2023-05-29:
Why was this closed? The bug is still there and the linked ticket seems related, but is not a duplicate or superset of this one

tablatronix, 2023-05-29:
Because its a bug i ESP not wm

khoint2101, 2023-06-12:
> Bất kỳ tin tức về điều này? Ngoài ra... có cách nào để bắt đầu nghe WifiManager trên cổng khác 80 không? Thx!

> Any news on this? Alternatively...is there a way to start WifiManager listening on a port other than 80? Thx!

You can open custom port for WM lib. You can use to 
`WiFiManager wm;
 wm.setHttpPort(8080); `
 This is open port 8080 for wifimanager web config, but when you login to wifi, you must type: 192.168.4.1:8080 to open web.
 When using the other port, webserver will start after success config wifi. DON'T NEED Hard Reset :>